OpenIPC/firmware #13: Image Quality 3516EV200 + SC2315E (SC4239)

oldies, 2021-04-25:
I've successfully flashed a XM-Based CAM with Hi3516EV200 Chip. But there is a Problem with the image quality (see picture attached):

[image]

Maybe someone of you already know this problem or could help to debug and identify the problem. 

ipctool output:
```
root@ipcam:~# ipctool
---
chip:
  vendor: HiSilicon
  model: 3516EV200
  id: 021833861e0038d9b370030a0362f69327874620018629e3
ethernet:
  mac: "**:**:**:**:**:**"
  u-mdio-phyaddr: 1
  phy-id: 0x20669903
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 6d76aff1
        contains:
          - name: uboot-env
            offset: 0x30000
      - name: env
        size: 0x10000
        sha1: 1adc95be
      - name: kernel
        size: 0x200000
        sha1: 3ec36165
      - name: rootfs
        size: 0x500000
        sha1: dad261bc
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
    addrMode: 3-byte
ram:
  total: 64M
  media: 32M
firmware:
  u-boot: "2016.11 (Oct 29 2018 - 16:06:38)"
  kernel: "4.9.37 (Thu Apr 22 13:59:19 UTC 2021)"
  toolchain: gcc version 7.5.0 (Buildroot 2020.02-g1f8ffa6) 
  sdk: "Hi3516EV200_MPP_V1.0.1.2 B030 Release (Oct 18 2019, 18:21:00)"
sensors:
- vendor: SmartSens
  model: SC2315E
  control:
    bus: 0
    type: i2c
    addr: 0x60
  viState: down
```

majestic log:
```
10:22:15 [    main] main@143                     Majestic version master+e2e5d65, built on 2021-04-20
10:22:15 [app_conf] load_config@152              Using /etc/majestic.yaml as main configuration
10:22:15 [app_conf] parse_app_config@239         SENSOR=sc2315e_i2c
10:22:15 [app_conf] find_sensor_config@115       matched sensor config: sc2315e_i2c_1080p_line.ini
10:22:15 [app_conf] find_sensor_config@125       Using /etc/sensors/sc2315e_i2c_1080p_line.ini as sensor configuration
10:22:15 [  config] parse_int64@416              Can't parse param 'volume' value 'auto'. Is not a integer (dec or hex) number.
10:22:15 [    main] main@160                     app_config.max_pool_cnt 128
10:22:15 [    main] main@161                     app_config.blk_cnt 4
10:22:15 [   hisdk] start_sdk@1606               App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
10:22:15 [   hisdk] start_sdk@1609               Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
10:22:15 [   hisdk] start_sdk@1623               chipid 3516E200
10:22:15 [   hisdk] start_sdk@1631               sensor_config.sensor_type stSnsSc2235Obj
10:22:15 [   hisdk] start_sdk@1632               sensor_config.dll_file /usr/lib/sensors/libsns_sc2235_eee.so
10:22:15 [   hisdk] start_sdk@1633               sensor_config.dev_attr 2
10:22:15 [   hisdk] start_sdk@1635               sensor_config.vichn.pix_format 26
10:22:15 [   hisdk] start_sdk@1637               sensor_config.input_mode 0
10:22:15 [   hisdk] start_sdk@1638               sensor_config.mode 0
10:22:15 [   hisdk] start_sdk@1641               sensor_config.isp.isp_x 0
10:22:15 [   hisdk] start_sdk@1642               sensor_config.isp.isp_y 0
10:22:15 [   hisdk] start_sdk@1643               sensor_config.isp.isp_w 1920
10:22:15 [   hisdk] start_sdk@1644               sensor_config.isp.isp_h 1080
10:22:15 [   hisdk] start_sdk@1647               sensor_config.isp.isp_frame_rate 30
10:22:15 [   hisdk] start_sdk@1648               sensor_config.isp.isp_bayer 0
10:22:15 [  sensor] tryLoadLibrary@23            try to load: /usr/lib/sensors/libsns_sc2235_eee.so
10:22:15 [  sensor] tryLoadLibrary@25            libsns_so 0x00000000B6D0CE20
10:22:15 [   hisdk] free_mem@813                 Free MMZ mem before allocation: 32736KB
10:22:15 [   hisdk] init_mem@1738                u32AlignWidth: 64
10:22:15 [   hisdk] dump_vb_configuration@1715   VB configuration:
10:22:15 [   hisdk] dump_vb_configuration@1725     [0]: 3110400 x 4
10:22:15 [   hisdk] free_mem@813                 Free MMZ mem after allocation: 20580KB
10:22:15 [   hisdk] init_sensor@1923             Sensor driver has been loaded
[Func]:cmos_set_wdr_mode [Line]:642 [Info]:SC2235_SENSOR_1080P_30FPS_LINEAR_MODE
Driver based on sc2235
==============================================================
===Smart sc2315e sensor 1080P30fps(MIPI) init success!=====Thu Jan 16 09:40:49 2020===
==============================================================
10:22:15 [   hisdk] init_vpss@2409               HI_MPI_SYS_Bind: VIU(0)->VPSS(grp 0)
10:22:15 [   hisdk] log_venc_chn@924             H.264 vpss_chn(0)->venc_chn(0)   1920x1080 25fps 4096Kbit
10:22:15 [   hisdk] create_vpss_chn@603            new venc: 0   vpss_grp: 0,   vpss_chn: 0
10:22:15 [   hisdk] create_vpss_chn@689            create_venc_chn  VPSS(0)->VENC(0)
10:22:15 [   hisdk] start_video@1328               venc took 7988KB of MMZ [1518]
10:22:15 [   hisdk] start_jpeg@828               JPEG snapshot venc_chn 1  1920x1080
10:22:15 [   hisdk] bind_vpss_venc@726             bind_vpss_venc  VPSS(0)->VENC(1)
10:22:15 [   hisdk] start_jpeg@915                 jpeg_enc took 2048KB of MMZ [2040]
10:22:15 [   audio] init_audio_in@583            Ai(0,0) bind to AencChn:0 ok!
10:22:15 [   hisdk] start_sdk@1700                 Audio took 512KB of MMZ
10:22:15 [   hisdk] start_sdk@1707               HiSilicon SDK has been started.
10:22:15 [   hisdk] free_mem@813                 Free MMZ mem finally: 6328KB
10:22:15 [image_pa] image_params_tuning@45       Image tuning thread has been started
10:22:15 [  evhttp] start_web_server@123         Using LAME v3.100
10:22:15 [  evhttp] start_web_server@180         HTTP server has been started on 0.0.0.0:8888.
10:22:15 [    rtsp] rtsp_init@143                RTSP server started on port 554
10:22:15 [      vi] VI_DRV_GetFrmVbBlk@4106      Pipe 0 chn 0 get buffer fail,size 3110400 !
10:22:15 [      vi] VI_COMM_CapPrepareFrame@2915 pipe id 0 get vb fail
(the last two lines will repeat many times)
```
dmesg (last lines):
```
FUNC:parse_sensor_clock line:126  SNS: is [sc2315e]!
FUNC:parse_sensor_bus_type line:85  SNS is [sc2315e]  !
==== online_flag=0, cmos_yuv_flag=0, sensor=sc2315e, chip=hi3516ev200, board=demo====
==== g_quick_start_flag=0 ====
sysconfig init success!
Module himedia: init ok
Hisilicon Media Memory Zone Manager
hi_osal 1.0 init success!
hi3516ev200_base: module license 'Proprietary' taints kernel.
Disabling lock debugging due to kernel taint
load sys.ko for Hi3516EV200...OK!
load tde.ko for HI3516EV200...OK!
load region.ko for Hi3516EV200...OK!
load vgs.ko for Hi3516EV200...OK!
load vi.ko for Hi3516EV200...OK !
ISP Mod init!
load vpss.ko for Hi3516EV200...OK!
load chnl.ko for Hi3516EV200...OK!
load vedu.ko for Hi3516EV200...OK!
load rc.ko for Hi3516EV200...OK!
load venc.ko for Hi3516EV200...OK!
load h264e.ko for Hi3516EV200...OK!
load h265e.ko for Hi3516EV200...OK!
load jpege.ko for Hi3516EV200...OK!
load ive.ko for Hi3516EV200...OK!
Load sensor_spi.ko for Hi3516EV200...OK !
load mipi_rx driver successful!
register dev
Hisilicon Watchdog Timer: 0.01 initialized. default_margin=60 sec (nodeamon= 0)
hiwtdg init ok. ver=Oct 18 2019, 18:21:00.
hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
(the last line will repeat)
```

widgetii, 2021-04-25:
Thank you for your report! Unfortunately, the supplier didn't provide a driver for the sensor we force to RE it and it will take time.

widgetii, 2021-04-26:
Please test the new driver version
[libsns_sc2235_eee.so.zip](https://github.com/OpenIPC/openipc-2.1/files/6379454/libsns_sc2235_eee.so.zip)

Make sure that `Isp_Bayer  = 3` in `/etc/sensors/sc2315e_i2c.ini`, it's crucial.

widgetii, 2021-05-12:
closed because no answer, it resolved based on my tests

closed 2021-05-12 by widgetii