OpenIPC/firmware #17: Backup/Flash without tftp (hi3516ev200)

Pimmetje, 2021-05-03:
I am trying to flash a hi3516ev200 without tftp (i am unable to get a link).

Normal boot:
```
System startup

Uncompress Ok!

U-Boot 2016.11-g2fc5f58-dirty (Sep 06 2019 - 15:13:30 +0800)hi3516ev200

Relocation Offset is: 0371b000
Relocating to 43f1b000, new gd at 43edaef0, sp at 43edaed0
SPI Nor:  eFlashType: 24.
Flash Name: XM_XT25F64B-S{0xB4017), 0x800000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Quad was Disabled, SRx: [2, 0x0].
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
Current level[6], lock_level_max:7.
at xm_get_locked_range() sr:0x18, level:6.
lk[6 => 0x400000]
SRx val: {[1, 0x38], [1, 0x0], [0, 0x0], [0, 0x0]}.
In:    serial
Out:   serial
Err:   serial
Net:   eth0
Hit ctrl+c to stop autoboot:  0
@do_spi_flash_probe() flash->erase_size:65536
device 0 offset 0x40000, size 0x550000

SF: 5570560 bytes @ 0x40000 Read: OK
srcAddr 0x43000000, dstAddr 0x42000000
created_inode 0x43edb810
find_squashfs_file: name bin, start_block 0, offset 2001, type 1
find_squashfs_file: name boot, start_block 0, offset 2181, type 1
read inode: name boot, sb 0, of 2181, type 1
find_squashfs_file: name uImage, start_block 0, offset 2033, type 2
read inode: name uImage, sb 0, of 2033, type 2
write_file: regular file, blocks 29
len 1861831
### FS load complete: 1861831 bytes loaded to 0x42000000
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1861767 Bytes = 1.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
```
I won't get any output afterwards. 

Interrupting:
```
System startup

Uncompress Ok!

U-Boot 2016.11-g2fc5f58-dirty (Sep 06 2019 - 15:13:30 +0800)hi3516ev200

Relocation Offset is: 0371b000
Relocating to 43f1b000, new gd at 43edaef0, sp at 43edaed0
SPI Nor:  eFlashType: 24.
Flash Name: XM_XT25F64B-S{0xB4017), 0x800000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Quad was Disabled, SRx: [2, 0x0].
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
Current level[6], lock_level_max:7.
at xm_get_locked_range() sr:0x18, level:6.
lk[6 => 0x400000]
SRx val: {[1, 0x38], [1, 0x0], [0, 0x0], [0, 0x0]}.
In:    serial
Out:   serial
Err:   serial
Net:   eth0
Hit ctrl+c to stop autoboot:  0
hisilicon 
```

Help
```
hisilicon # help
?       - alias for 'help'
base    - print or set address offset
bdinfo  - print Board Info structure
bitwait - bit compare and wait for equal
boot    - boot default, i.e., run 'bootcmd'
bootd   - boot default, i.e., run 'bootcmd'
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
clearenv- clear env partition.
cmp     - memory compare
coninfo - print console devices and information
cp      - memory copy
crc32   - checksum calculation
dcache  - enable or disable data cache
ddr     - ddr training function
dispaddr- display the value of 'addr'
dispenv - display the value of 'env_var'
dispver - display the uboot version
editenv - edit environment variable
env     - environment handling commands
erase   - erase FLASH memory
exit    - exit script
false   - do nothing, unsuccessfully
flinfo  - print FLASH memory information
flwrite - SPI flash sub-system
getinfo - print hardware information
go      - start application at address 'addr'
gzwrite - unzip and write memory to block device
help    - print command description/usage
icache  - enable or disable instruction cache
loadb   - load binary file over serial line (kermit mode)
loadx   - load binary file over serial line (xmodem mode)
loady   - load binary file over serial line (ymodem mode)
loop    - infinite loop on address range
md      - memory display
mii     - MII utility commands
mm      - memory modify (auto-incrementing address)
mw      - memory write (fill)
nm      - memory modify (constant address)
part    - disk partition related commands
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
protect - enable or disable FLASH write protection
pxe     - commands to get and boot from pxe files
reset   - Perform RESET of the CPU
run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
setenv  - set environment variables
sf      - SPI flash sub-system
showvar - print local hushshell variables
sleep   - delay execution for some time
squashfsload- fsload  - load binary file from a filesystem image

sysboot - command to get and boot from syslinux files
test    - minimal test like /bin/sh
tftpboot- boot image via network using TFTP protocol
true    - do nothing, successfully
unzip   - unzip a memory region
version - print monitor, compiler and linker version
waitus  - wait for n us
```

```
hisilicon # bdinfo
arch_number = 0x00001F40
boot_params = 0x40000100
DRAM bank   = 0x00000000
-> start    = 0x40000000
-> size     = 0x04000000
eth0name    = eth0
ethaddr     = 00:12:31:xx:xx:xx
current eth = eth0
ip_addr     = 192.168.2.168
baudrate    = 115200 bps
TLB addr    = 0x43FF0000
relocaddr   = 0x43F1B000
reloc off   = 0x0371B000
irq_sp      = 0x43EDAEE0
sp start    = 0x43EDAED0
```

I played around with some commands. (setting other IP's/subnets etc).  But i was unable to get the ethernet leds to turn on. 
So i guess it would require flashing it over serial only.

I have seen a tool (ubootwrite.py) that could help with that. But that's something i still need to find out.

Pimmetje, 2021-05-04:
Using https://m.habr.com/en/post/486856/ and https://github.com/tothi/hs-dvr-telnet

and 

```
from telnetlib import Telnet

def telnet_login(password):
    print('Attempting', password)
    with Telnet('10.9.2.200', 23) as tn:
        username = 'root'
        tn.set_debuglevel(1000)
        tn.read_until(b"login: ")
        tn.write(username.encode('ascii') + b"\n")
        tn.read_until(b"Password: ")
        tn.write(password.encode('ascii') + b"\n")
        data = tn.read_until(b"Login incorrect")
        if b'Login incorrect' in data:
            return False
        return True

file1 = open('passwords.txt', 'r')
lines = file1.readlines()
for line in lines:
    if telnet_login(line):
        break

```

passwords.txt
```
00000000
059AnkJ
4uvdzKqBkj.jg
7ujMko0admin
7ujMko0vizxv
123
1111
1234
1234qwer
2601hx
12345
54321
123456
666666
888888
1111111
/*6.=_ja
anko
anni2013
annie2012
avtech97
cat1029
ccadmin
cxlinux
default
dreambox
fxjvt1805
hdipc%No
hi3518
hichiphx
hipc3518
hkipc2016
hslwificam
ikwb
ipc71a
IPCam@sw
ivdev
juantech
jvbzd
jvtsmart123
klv123
klv1234
meinsm
OxhlwSG8
pass
password
realtek
root
hi3518
S2fGqNFs
service
smcadmin
supervisor
support
system
tech
tlJwpbo6
ubnt
user
vhd1206
vizxv
xc3511
xmhdipc
zlxx.
Zte521
```

My telnet password is xmhdipc (username root)
Now i have to find a way to extract data and upload new programs.

widgetii, 2021-05-05:
Try to run [ipctool](https://github.com/OpenIPC/ipctool/releases/download/latest/ipctool) on it

Pimmetje, 2021-05-05:
I was able to transfer the ipctool using serial

```
---
board:
  vendor: Xiongmai
  model: HI3516EV200_50H20AI_S38
  cloudId: 828e52bc1xxxxxxx
chip:
  vendor: HiSilicon
  model: 3516EV200
  id: 02383a867e0038e9ee70030ac2d7f693262867xxxxxxxxx
ethernet:
  mac: "00:12:31:xx:xx:xx"
  u-mdio-phyaddr: 1
  phy-id: 0x206xxxxx
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    chip:
      name: "XM_XT25F64B-S"
      id: 0x0b4017
    partitions:
      - name: boot
        size: 0x40000
        sha1: 9c3185f5
        contains:
          - name: xmcrypto
            offset: 0x2fc00
          - name: uboot-env
            offset: 0x30000
      - name: romfs
        size: 0x2e0000
        path: /,squashfs
        sha1: 548fe861
      - name: user
        size: 0x420000
        path: /usr,squashfs
        sha1: 64b32f0a
      - name: web
        size: 0x40000
        path: /mnt/web,cramfs
        sha1: ffac9f55
      - name: custom
        size: 0x30000
        path: /mnt/custom,cramfs
        sha1: 3ef98758
      - name: mtd
        size: 0x50000
        path: /mnt/mtd,jffs2,rw
    size: 8M
    addrMode: 3-byte
ram:
  total: 64M
  media: 26M
firmware:
  u-boot: "2016.11 (Oct 29 2018 - 16:06:38)"
  kernel: "4.9.37 (Fri Sep 6 15:29:30 CST 2019)"
  toolchain: gcc version 6.3.0 (Heterogeneous Compiler&Codesign V100R002C00B003)
  libc: uClibc 0.9.33.2
  sdk: "Hi3516EV200_MPP_V1.0.1.1 B030 Release (Jun 17 2019, 11:19:14)"
  god-app: /usr/bin/Sofia
sensors:
- vendor: Sony
  model: IMX307
  control:
    bus: 0
    type: i2c
    addr: 0x34
  data:
    type: MIPI
    input_data_type: DATA_TYPE_RAW_12BIT
    laneId:
    - 0
    - 1
    image: 1920x1080
  clock: 37.125MHz
```

Pimmetje, 2021-05-05:
Mount seems to support nfs :)
```
mkdir /var/utils
mount -o nolock 127.0.0.1:/volume1/nfs /var/utils
```

widgetii, 2021-05-05:
Your board (SoC + sensor) is supported by OpenIPC firmware
https://github.com/OpenIPC/openipc-2.1/wiki/install_hisi

Pimmetje, 2021-05-05:
I think i need to unblock my uboot first as it seems to me it is missing the ethernet driver or something.
Or at least figure out a way to get the ethernet to work on uboot.

Pimmetje, 2021-05-06:
Is it possible to copy the firmware over mtdblocks?

Pimmetje, 2021-05-06:
Apparently the eth leds do not work and my pc does not reply to ping. So tftp works. 

I flashed the camera using the following commands:

```
setenv bootargs 'mem=${osmem:-32M} console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)'
setenv bootcmd 'setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read 0x42000000 0x50000 0x200000; bootm 0x42000000'
saveenv

setenv soc hi3516ev200
setenv osmem 32M
setenv totalmem 64M        
setenv ipaddr 10.9.2.200
setenv serverip 10.9.2.250
saveenv

sf probe 0; mw.b 0x42000000 ff 1000000; tftp 0x42000000 uImage.${soc}; sf erase 0x50000 0x200000; sf write 0x42000000 0x50000 ${filesize}
sf probe 0; mw.b 0x42000000 ff 1000000; tftp 0x42000000 rootfs.squashfs.${soc}; sf erase 0x250000 0x500000; sf write 0x42000000 0x250000 ${filesize}
reset
```

tftp server (10.9.2.250) used
```
dffd94dc5b8907558b11a8dc41db63bd  rootfs.squashfs.hi3516ev200
ca7328f53e986615e3f7fd424c18a83e  uImage.hi3516ev200
```

Boot output
```
System startup

Uncompress Ok!

U-Boot 2016.11-g2fc5f58-dirty (Sep 06 2019 - 15:13:30 +0800)hi3516ev200

Relocation Offset is: 0371b000
Relocating to 43f1b000, new gd at 43edaef0, sp at 43edaed0
SPI Nor:  at Mxic_hasTB() cr:0x8.
at Mxic8M_getTypeBySFDP() sfdp[0x68:0x69]:0xfe,0xcf.
eFlashType: 8.
Flash Name: XM_MX25L6433F{0xC22017), 0x800000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Quad was Disabled, SRx: [1, 0x0].
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
Current level[0], lock_level_max:8.
unlock all.
SRx val: {[1, 0x0], [1, 0x8], [0, 0x0], [0, 0x0]}.
In:    serial
Out:   serial
Err:   serial
Net:   eth0
Hit ctrl+c to stop autoboot:  0
@do_spi_flash_probe() flash->erase_size:65536
device 0 offset 0x50000, size 0x200000

SF: 2097152 bytes @ 0x50000 Read: OK
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1978940 Bytes = 1.9 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az50-139) (gcc version 7.5.0 (Buildroot 2020.02-ga15ab4b) ) #1 Mon May 3 14:10:21 UTC 2021
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
[... 163 lines cut ...]
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #1
Hardware name: Generic DT based system
Backtrace:
[<c0013120>] (dump_backtrace) from [<c0013418>] (show_stack+0x18/0x1c)
 r7:c1a3a009 r6:c047de58 r5:00000000 r4:c053f2e8
[<c0013400>] (show_stack) from [<c01d4458>] (dump_stack+0x24/0x28)
[<c01d4434>] (dump_stack) from [<c0077540>] (panic+0xe4/0x24c)
[<c0077460>] (panic) from [<c04ed440>] (mount_block_root+0x280/0x2dc)
 r3:d3053b52 r2:d3053b52 r1:c183de8c r0:c047de58
 r7:c1a3a009
[<c04ed1c0>] (mount_block_root) from [<c04ed6e4>] (mount_root+0x138/0x154)
 r10:c050d838 r9:c04ec620 r8:c050d834 r7:c053f024 r6:c051d18c r5:01f00003
 r4:c051a808
[<c04ed5ac>] (mount_root) from [<c04ed884>] (prepare_namespace+0x184/0x1cc)
 r10:c050d838 r9:c04ec620 r8:c050d834 r7:c053f000 r6:c053f000 r5:c053f024
 r4:c050d858
[<c04ed700>] (prepare_namespace) from [<c04ecf34>] (kernel_init_freeable+0x1d0/0x1e0)
 r6:c053f000 r5:00000009 r4:c04e93a4
[<c04ecd64>] (kernel_init_freeable) from [<c0400a58>] (kernel_init+0x10/0xfc)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0400a48
 r4:00000000
[<c0400a48>] (kernel_init) from [<c000fee8>] (ret_from_fork+0x14/0x2c)
 r5:c0400a48 r4:00000000
Rebooting in 20 seconds..
```

Pimmetje, 2021-05-07:
The default/stock firmware mtdblock i backupped are 
```
byte     Size  filename  md5                                 sha1
 262144  256K  mtdblock0 ebb09384c3bf15f02a49d158773e3f60    9c3185f5fb44dde38a07ccadecece13f500a1c0d
3014656  2.9M  mtdblock1 023c1b9dc1e4a9ea24e03ee0303d7b83    548fe8617c5449b020fdff7459445700581986f3
4325376  4.2M  mtdblock2 3b5e3752cee328c23c51b3b137b0b504    64b32f0aa3325828ca4b8322b8e1ec64ed14016d
 262144  256K  mtdblock3 9f3fde3ae968a4d092316313da9e258e    ffac9f55bd3726e4deaac3ab9323aed13fe91b34
 196608  192K  mtdblock4 c997ffb67bf70c549273bae4a653ae2b    3ef98758071c54a4bee14c0bdb0269ea53ee6c02
 327680  320K  mtdblock5 7e84ad8a5bcc15efd54bbfe0469f2872    f78be4df5434238de8ee4eab201077ad93fb246b
 
sha512
3d64f3dd69fcd36649891d1428122c4985050541ffd131fafcb27b1939f9d1bed157ea470c561a0ac79a74adb82c748648bd5d4af994d51c53317709164f0136  mtdblock0
881abfc1d092580a49faed29e3189d68376e035bc9394ff728b436b11f62001386b3274a5ca68b3df126d723502a2120b14e76652092b1237439e4f7c5f60bfa  mtdblock1
63055d989a7afdc680a1a1c09701643d6890c8b60f120c4a1b433fac225d34d8417f9c8db5671bf176c365e55fe1e204f73c9989930079f9cff5c3390b839900  mtdblock2
f3ef9cb32e59e61959165e6f6a10ca1cac276f49abc9597b43fcc632fa3e2e75623979f100d9d70cb9a2548de002472f580ad28ea0597f9eb35596e4b5aaa7c9  mtdblock3
c62559c684aa7ade915f099581e2505a5180a430a250675ddcec2286382eb6b45a32cf6673a1108d37cfeea4f7ca321d80e25fc0ff82d49e4c8b4d311e8f23a7  mtdblock4
4a0c8357bdaf78f3c585b60e6ae33e91ebb7a6b3f4aaa4443490eb88a24bf4c0036e38f55434ef49583d3a684df1fad466012b26f29ee1c134e68dde785c1672  mtdblock5
```
The size of the files i used. So it would not fit. 
```
 5107712 4.9M  rootfs.squashfs.hi3516ev200
 1979004 1.9M  uImage.hi3516ev200
```

Could it be that the sizes of the blocks are wrong?

Pimmetje, 2021-05-07:
Using another uImage (using lower spi speed) solved it. Now i can boot OpenIPC. Thanks everyone for there help.

closed 2021-05-07 by Pimmetje

Mishotpl, 2021-09-17:
how do you backup firmware?
original firmware don't have "tftp", only tftpboot

widgetii, 2021-09-17:
Consider using [described technique](https://github.com/openipc/burn#ev300) but download `ev200` version of U-Boot to run it from RAM and then do backup