OpenIPC/firmware #20: Is it support ipcam on xm550ai chip?

esuslik, 2021-05-13:
and how to install firmware on ipcam xm550ai?

ZigFisher, 2021-05-14:
https://github.com/OpenIPC/openipc-2.1/wiki/install_xm530

esuslik, 2021-05-15:
Unfortunately i cannot understand. Which program i need to use and how to upload new firmware to ipcam.
Need step by step instruction

widgetii, 2021-05-15:
You will need UART adapter connected to your computer and camera and then boot to U-boot command line and do commands applicable for your board.

Don't forget about backup of your original firmware

esuslik, 2021-05-24:
i can't find uart interface on board. Maybe you can help?
[Parameters for IVG-85X50PYA-S.pdf](https://github.com/OpenIPC/openipc-2.1/files/6530231/Parameters.for.IVG-85X50PYA-S.pdf)

widgetii, 2021-05-24:
[image]

Check top connector with pin marked from 1 to 4, first find GND with your multimeter and then find TX and RX by trial and error method

closed 2021-05-24 by widgetii