OpenIPC/firmware #22: Compiling

schneemaier, 2021-05-19:
Hi, i tried to compile the project to different supported hardwares, however i always recevie an error:

/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/bin/../lib/gcc/arm-openipc-linux-musleabi/7.5.0/../../../../arm-openipc-linux-musleabi/bin/ld: CMakeFiles/ipctool.dir/src/main.c.o: in function `get_git_version':
main.c:(.text.get_git_version+0x8): undefined reference to `GIT_TAG'
/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/bin/../lib/gcc/arm-openipc-linux-musleabi/7.5.0/../../../../arm-openipc-linux-musleabi/bin/ld: CMakeFiles/ipctool.dir/src/main.c.o: in function `get_git_revision':
main.c:(.text.get_git_revision+0x8): undefined reference to `GIT_REV'
/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/bin/../lib/gcc/arm-openipc-linux-musleabi/7.5.0/../../../../arm-openipc-linux-musleabi/bin/ld: CMakeFiles/ipctool.dir/src/main.c.o: in function `get_git_branch':
main.c:(.text.get_git_branch+0x8): undefined reference to `GIT_BRANCH'
collect2: error: ld returned 1 exit status
CMakeFiles/ipctool.dir/build.make:848: recipe for target 'ipctool' failed
make[5]: *** [ipctool] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ipctool.dir/all' failed
make[4]: *** [CMakeFiles/ipctool.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
[ 26%] Linking C executable ipcinfo
/usr/bin/cmake -E cmake_link_script CMakeFiles/ipcinfo.dir/link.txt --verbose=1
/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/bin/arm-openipc-linux-musleabi-gcc --sysroot=/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/arm-openipc-linux-musleabi/sysroot -std=gnu99 -s -Os -ffunction-sections -Wl,--gc-sections -DNDEBUG  -DNDEBUG  -rdynamic CMakeFiles/ipcinfo.dir/example/ipcinfo.c.o  -o ipcinfo libipchw.a -lm 
/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/bin/../lib/gcc/arm-openipc-linux-musleabi/7.5.0/../../../../arm-openipc-linux-musleabi/bin/ld: CMakeFiles/ipcinfo.dir/example/ipcinfo.c.o: in function `get_git_version':
ipcinfo.c:(.text.get_git_version+0x8): undefined reference to `GIT_TAG'
/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/bin/../lib/gcc/arm-openipc-linux-musleabi/7.5.0/../../../../arm-openipc-linux-musleabi/bin/ld: CMakeFiles/ipcinfo.dir/example/ipcinfo.c.o: in function `get_git_revision':
ipcinfo.c:(.text.get_git_revision+0x8): undefined reference to `GIT_REV'
/home/schnee/Workspace/openipc-2.1/output/per-package/ipctool/host/bin/../lib/gcc/arm-openipc-linux-musleabi/7.5.0/../../../../arm-openipc-linux-musleabi/bin/ld: CMakeFiles/ipcinfo.dir/example/ipcinfo.c.o: in function `get_git_branch':
ipcinfo.c:(.text.get_git_branch+0x8): undefined reference to `GIT_BRANCH'
collect2: error: ld returned 1 exit status
CMakeFiles/ipcinfo.dir/build.make:95: recipe for target 'ipcinfo' failed
make[5]: *** [ipcinfo] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/ipcinfo.dir/all' failed
make[4]: *** [CMakeFiles/ipcinfo.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
package/pkg-generic.mk:266: recipe for target '/home/schnee/Workspace/openipc-2.1/output/build/ipctool-2745429e7add31b6e52b0079a8b28b04daf7a349/.stamp_built' failed
make[2]: *** [/home/schnee/Workspace/openipc-2.1/output/build/ipctool-2745429e7add31b6e52b0079a8b28b04daf7a349/.stamp_built] Error 2
Makefile:84: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/schnee/Workspace/openipc-2.1/buildroot-2020.02.12'
Makefile:91: recipe for target 'all' failed
make: *** [all] Error 2

This is how i try to compile:

export PLATFORM=hisilicon ; make BOARD=hi3516ev200 all

System: Ubuntu 18.04LTS

What am i missing?

Thanks

ZigFisher, 2021-05-23:
Good day

Please note that you do not apply the key "-SKIP_VERSION=ON" for some reason

https://github.com/OpenIPC/openipc-2.1/blob/master/general/package/ipctool/ipctool.mk

schneemaier, 2021-05-24:
Thank you! Where should i apply the key? I downloaded the repositiory and did not modify any files.

widgetii, 2021-06-01:
It's definitely is not an ipctool related issue. It works on my system so you'll need to debug make stuff on yours.

closed 2021-06-01 by widgetii

schneemaier, 2021-06-03:
I did some more tests. It seems that the project does not compile on Ubuntu 18.04LTS, but does compile on Ubunto 20.04LTS

ZigFisher, 2021-06-04:
It would be very good to find out exactly what the reason is..