OpenIPC/firmware #24: Possibly misconfigured kernel on SSC335

operutka, 2021-06-01:
The kernel for the SSC335 boards seems to be misconfigured. Based on the following line from the boot log and the information present in `br-ext-chip-sigmastar/board/infinity6b0/kernel/ssc335.generic.config`, the kernel defconfig appears to be for INFINITY6B0 SSC009B-S01A (which is SSC335DE).
```
OF: fdt:Machine model: INFINITY6B0 SSC009B-S01A QFN128
```

According to the SigmaStar documentation (see the screenshot below), it should be INFINITY6B0 SSC009A-S01A QFN88:

[image]

If I try to boot using this kernel, the kernel freezes before starting init. If I replace `br-ext-chip-sigmastar/board/infinity6b0/kernel/ssc335.generic.config` with `linux-4.9.84/arch/arm/configs/infinity6b0_ssc009a_s01a_defconfig` and build a new kernel image using this configuration, the device boots up fine.

ZigFisher, 2021-06-01:
Thank you very much !
Corrections will be made as soon as possible.

ZigFisher, 2021-06-01:
https://github.com/OpenIPC/openipc-2.1/commit/57f1e9489d3ffc49dad98503f1d004e373d7d723

operutka, 2021-06-01:
I've just tested it and, unfortunately, changing `CONFIG_SS_DTB_NAME` alone does not help. The kernel prints the correct SoC model this time but it still freezes right before the init. There must be another option (or maybe options?) affecting this. The original config `linux-4.9.84/arch/arm/configs/infinity6b0_ssc009a_s01a_defconfig` from SigmaStar works fine.

I'll try to iterate from the SigmaStar config to the OpenIPC config by updating options one by one. Hopefully, I'll be able to find the problematic one.

ZigFisher, 2021-06-01:
Please check with this commit https://github.com/OpenIPC/openipc-2.1/commit/76307b28e34ce8ace429df5ae26c7a451d5fe8ce#diff-fbdc6ab1a13a61676d2734d02b194fbbc636ddbbaeea400333fceb36fc6ef4af

operutka, 2021-06-02:
Oh, perfect. I can see you did quite a lot of changes in the end. The kernel image works fine now. Thank you. There is one final thing that cripples the system though - missing overlay FS support.

operutka, 2021-06-03:
Fixed. Closing the issue.

closed 2021-06-03 by operutka