OpenIPC/firmware #30: Kernel does not boot from KH25L6433F flash drive

darkbyte-ru, 2021-06-20:
On new 85HF20PY board (Hi3516Ev200 + IMX307) stock flash drive was 8M Macronix KH25L6433F (chip ID: 0xC2201716)

```
hisi-sfc hisi_spi_nor.0: SPI Nor ID Table Version 1.2
hisi-sfc hisi_spi_nor.0: all blocks is unlocked.
@spi_nor_scan(), no "m25p,fast-read".
@spi_nor_scan(), modes->rd_modes:0x3d.
hisi-sfc hisi_spi_nor.0: (Fast) Read:  opcode=EBh, protocol=144, mode=8, wait=16
hisi-sfc hisi_spi_nor.0: nor->read_opcode[3: Read; 0B: Fast Read; 3B: Dual; BB: Dual IO; 6B: Quad; EB: Quad IO]: 0xeb.
hisi-sfc hisi_spi_nor.0: mx25l6436f (Chipsize 8 Mbytes, Blocksize 64KiB)
5 cmdlinepart partitions found on MTD device hi_sfc
5 cmdlinepart partitions found on MTD device hi_sfc
Creating 5 MTD partitions on "hi_sfc":
0x000000000000-0x000000040000 : "boot"
0x000000040000-0x000000050000 : "env"
0x000000050000-0x000000250000 : "kernel"
0x000000250000-0x000000750000 : "rootfs"
0x000000750000-0x000000800000 : "rootfs_data"
SPI Nand ID Table Version 2.7
Cannot found a valid SPI Nand Device
hisi_spi_nand_probe(175): Error: driver probe, result: -19
```
...
```
1f00             256 mtdblock0  (driver?)
1f01              64 mtdblock1  (driver?)
1f02            2048 mtdblock2  (driver?)
1f03            5120 mtdblock3  (driver?)
1f04             704 mtdblock4  (driver?)
No filesystem could mount root, tried:  squashfs

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
```

Full boot attached: [kernel-panic-KH25L6433F.txt](https://github.com/OpenIPC/openipc-2.1/files/6682856/kernel-panic-KH25L6433F.txt)

widgetii, 2021-06-20:
Please try this kernel instead of image from release
[uImage.zip](https://github.com/OpenIPC/openipc-2.1/files/6682888/uImage.zip)

It's a special version with patched FMC driver and Quad modes disabled

darkbyte-ru, 2021-06-21:
This uImage works! Kernel boot ok.

widgetii, 2021-06-21:
[disable.quad.patch.zip](https://github.com/OpenIPC/openipc-2.1/files/6687088/disable.quad.patch.zip)

Actual patch for the image.

ZigFisher, 2021-06-21:
Is there a suggestion to include this patch directly in the project ?

widgetii, 2021-06-22:
Not "as is" due to we'll get performance drop for good boards.

Need to experiment with additional kernel parameter/possible auto detection

widgetii, 2021-12-08:
Quad mode is disabled in master by default

closed 2021-12-08 by widgetii