OpenIPC/firmware #31: Problem with mount sd card on boot. Board hi3516ev300 + imx335 (pcb rev: 16EV3_335AF_1.1)

Truevoxdei, 2021-07-06:
usbhid: USB HID core driver
Initializing XFRM netlink socket
NET: Registered protocol family 17
NET: Registered protocol family 15
Key type dns_resolver registered
hibvt_rtc 120e0000.rtc: setting system clock to 1970-01-01 00:03:46 UTC (226)
VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
devtmpfs: mounted
Freeing unused kernel memory: 176K (c04ec000 - c0518000)
This architecture does not have kernel memory protection.
mmc0: new high speed SDHC card at address 0001
mmcblk0: mmc0:0001 EB1QT 29.8 GiB
 mmcblk0: p1
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Starting mdev...
yaffs: dev is 187695105 name is "mmcblk0p1" rw
yaffs: passed flags ""
yaffs: dev is 187695105 name is "mmcblk0p1" rw
yaffs: passed flags ""
mount: mounting /dev/mmcblk0p1 on /mnt/mmcblk0p1 failed: Invalid argument
Saving random seed: random: dd: uninitialized urandom read (512 bytes read)
OK

Truevoxdei, 2021-07-06:
**info from ipctool:**

root@openipc:~# ipctool
---
board:
  vendor: OpenIPC
  version: 2020.02.12
chip:
  vendor: HiSilicon
  model: 3516EV300
  id: 001437073e0038e1c170030a4ac8ca1426a58508504f39e3
ethernet:
  u-mdio-phyaddr: 1
  phy-id: 0x20669903
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 6d8ccd77
        contains:
          - name: uboot-env
            offset: 0x30000
      - name: env
        size: 0x10000
        sha1: 472a55b0
      - name: kernel
        size: 0x200000
        sha1: e5778e26
      - name: rootfs
        size: 0x500000
        sha1: c100d46d
      - name: rootfs_data
        size: 0x8b0000
        path: /overlay,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
  media: 96M
firmware:
  kernel: "4.9.37 (Mon Jul 5 11:45:28 UTC 2021)"
  toolchain: gcc version 7.5.0 (Buildroot -g0296a76)
  sdk: "Hi3516EV200_MPP_V1.0.1.2 B030 Release (Oct 18 2019, 18:21:00)"
  god-app: /usr/bin/majestic
sensors:
- vendor: Sony
  model: IMX335
  control:
    bus: 0
    type: i2c
    addr: 0x34
  data:
    type: MIPI
    input-data-type: DATA_TYPE_RAW_10BIT
    lane-id:
    - 0
    - 1
    - 2
    - 3
    image: 2592x1520
  clock: 37.125MHz

**variables from environment:**

System startup

Uncompress Ok!

U-Boot 2016.11 (Oct 18 2020 - 01:56:55 +0800)hi3516ev300

Relocation Offset is: 0775e000
Relocating to 47f5e000, new gd at 47f1def0, sp at 47f1ded0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
SPI Nor(cs 0) ID: 0xa1 0x40 0x18
Block:64KB Chip:16MB Name:"FM25Q128A-SOB-T-G"
SPI Nor total size: 16MB
NAND:  0 MiB
In:    serial
Out:   serial
Err:   serial
Net:   eth0
Hit any key to stop autoboot:  0
### Please input uboot password: ###
**********
hisilicon # printenv
arch=arm
baudrate=115200
board=hi3516ev300
board_name=hi3516ev300
bootargs=mem=128M console=ttyAMA0,115200 panic=20 rootfstype=squashfs root=/dev/mtdblock3 init=/init mtdparts=sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data) mmz_allocator=cma mmz=anonymous,0,0x42000006M
bootcmd=sf probe 0; sf read 0x42000000 0x50000 0x200000; bootm 0x42000000
bootdelay=1
cpu=armv7
ethact=eth0
ethaddr=00:12:23:34:xx:xx
fileaddr=42000000
filesize=4fd000
ipaddr=192.168.1.10
osmem=32M
sensor=imx335
serverip=192.168.1.7
soc=hi3516ev300
stderr=serial
stdin=serial
stdout=serial
totalmem=128M
vendor=hisilicon
verify=n

Environment size: 853/65532 bytes

**u-boot available commands:**
hisilicon # help
?       - alias for 'help'
base    - print or set address offset
bdinfo  - print Board Info structure
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
cmp     - memory compare
coninfo - print console devices and information
cp      - memory copy
crc32   - checksum calculation
ddr     - ddr training function
dhcp    - boot image via network using DHCP/TFTP protocol
echo    - echo args to console
editenv - edit environment variable
env     - environment handling commands
erase   - erase FLASH memory
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatsize - determine a file's size
flinfo  - print FLASH memory information
getinfo - print hardware information
go      - start application at address 'addr'
gzwrite - unzip and write memory to block device
help    - print command description/usage
itest   - return true/false on integer compare
loop    - infinite loop on address range
md      - memory display
mii     - MII utility commands
mm      - memory modify (auto-incrementing address)
mw      - memory write (fill)
nand    - NAND sub-system
nboot   - boot from NAND device
nfs     - boot image via network using NFS protocol
nm      - memory modify (constant address)
part    - disk partition related commands
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
protect - enable or disable FLASH write protection
pxe     - commands to get and boot from pxe files
reset   - Perform RESET of the CPU
saveenv - save environment variables to persistent storage
setenv  - set environment variables
setexpr - set environment variable as the result of eval expression
sf      - SPI flash sub-system
source  - run script from memory
sysboot - command to get and boot from syslinux files
tftpboot- boot image via network using TFTP protocol
tftpput - TFTP put command, for uploading files to a server
ugzip   - Compress gzipfile with hardware IP
unzip   - unzip a memory region
version - print monitor, compiler and linker version

Truevoxdei, 2021-07-06:
[image]

ZigFisher, 2021-08-23:
Good afternoon
Please format the card as vfat or f2fs

closed 2021-12-08 by widgetii

cronyx, 2023-05-05:
> @Truevoxdei I have the same PCB. How the hell get you OpenIPC on your device?? Over UART. ? Can you plz explain me the Pinout ?

It is a CamHi board, flash via uart.

[image]

matteius, 2023-12-12:
@cronyx I have that board, got it last week, and I just finally got UART working, the pinout is nuts because they claim the 5-pin connector is a 7-pin, but its actually those 4 micropads in the corner, 2 of them are the Rx/Tx.    My problem now is the bootloader and the root account are password protected and I cannot figure out bypassing it yet.