OpenIPC/firmware #32: test_venc each time gets a bigger and bigger snapshot

ZigFisher, 2021-07-08:
Good afternoon, colleagues !

We noticed an interesting situation..
test_venc each time gets a bigger and bigger snapshot

```
...
reading a frame...
  waiting for a frame...
  frame #16
  frame pts: 158409505
  packets: 1
  packet #0 size: 118239
  ok
reading a frame...
  waiting for a frame...
  frame #17
  frame pts: 158909505
  packets: 1
  packet #0 size: 123403
  ok
reading a frame...
  waiting for a frame...
  frame #18
  frame pts: 159409504
  packets: 1
  packet #0 size: 135289
  ok
reading a frame...
  waiting for a frame...
  frame #19
  frame pts: 159909505
  packets: 1
  packet #0 size: 141950
  ok
```
Is it some kind of resource leaking issue?

operutka, 2021-07-08:
I don't think there is any resource leakage. There is a few things to consider.

1. The image complexity grows here because it takes a few frames until the exposure gets adjusted.
2. The CBR rate control adjusts the image quality to meet the bitrate settings over a certain period of time. It does not mean that every frame will have the same size.
3. The total number of frames read from VENC is quite low to make any conclusions.

I've also found a small bug in the frame rate settings. VENC receives only two frames per second:
https://github.com/OpenIPC/openipc-2.1/blob/master/br-ext-chip-sigmastar/package/sigmastar-osdrv-ssc335/files/sample/test_venc.c#L569

However, the rate control expects to receive 30 frames per second:
https://github.com/OpenIPC/openipc-2.1/blob/master/br-ext-chip-sigmastar/package/sigmastar-osdrv-ssc335/files/sample/test_venc.c#L484

This can probably lead to some miscalculations done by the rate control but it shouldn't cause any resource leakage. The rate control simply thinks that it is below the bitrate limit and it can increase the size of the frames.

I suggest fixing this bug to tell the rate control the correct frame rate. Once this is done, you can increase the number of frames read from VENC:
https://github.com/OpenIPC/openipc-2.1/blob/master/br-ext-chip-sigmastar/package/sigmastar-osdrv-ssc335/files/sample/test_venc.c#L716

to see if the frame size grows indefinitely or if it stabilizes after some time.

ZigFisher, 2021-07-08:
Thank you so much for the answer !

ZigFisher, 2021-07-08:
Thank you so much for the answer !
If possible, please send a patch for this test utility. 
I think it will be interesting for all of us.

operutka, 2021-07-12:
This is my result after reading 460 frames. As you can see, the frame size stabilizes relatively quickly slightly above 200 kB. With 2 frames per second, it's still below the 4 Mbps bitrate limit. So the rate control is clearly doing its job. Anyway, I'll create a PR fixing the frame rate issue.

[image]

ZigFisher, 2021-07-12:
Thank you very much, Sir !

ZigFisher, 2021-07-12:
I also want to ask your opinion about cam_os_wrapper, which must be connected statically to the test utility. Do you know why this happens and you can't make a dynamic connection ?

operutka, 2021-07-13:
No problem. I'm glad I could help.

> I also want to ask your opinion about cam_os_wrapper, which must be connected statically to the test utility. Do you know why this happens and you can't make a dynamic connection ?

The dynamic version of the lib is either broken or there is another linking issue between the cam_os_wrapper lib and the other SigmaStar libs because if I link it dynamically, it always ends with the following runtime error:
```text
/tmp/venc-test: can't resolve symbol 'CamOsMemCalloc'
```

ZigFisher, 2021-07-13:
Yes, I have a similar experience. 
Thank you for the explanations.

closed 2021-09-22 by widgetii