OpenIPC/firmware #36: Compile problems for xm530

schneemaier, 2021-08-08:
Hi,

i tried to compile the source for xm530 target, but i received several errors. Some of them i was able to fix.

Here is what i did:
```
git clone --depth=1 https://github.com/OpenIPC/openipc-2.1.git
cd openipc-2.1
export PLATFORM=xiongmai ; make distclean ; make prepare ; make BOARD=unknown_unknown_xm530_openipc all
```
Here are the errors i receive and the fix for eacf of them.

**1st Error**
```
home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/external.mk:18: /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package/vtund-lite/vtund-openipc.mk: No such file or directory vtund-openipc
```

solution:
```
cd /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package
ln -s vtund-openipc vtund-lite
```

**2nd error**
```
/home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/external.mk:17: /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package/uacme-lite/uacme-openipc.mk: No such file or directory
```

solution:
```
cd /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package
ln -s uacme-openipc uacme-lite
```

**3rd error**
```
/home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/external.mk:13: /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package/majestic-xm530/majestic-xm530.mk: No such file or directory
```

solution:
```
cd /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package/majestic-xm530/
ln -s majestic.mk majestic-xm530.mk
```

**4th error:**
```
/home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/external.mk:12: /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package/majestic-xm510/majestic-xm510.mk: No such file or directory
```

solution:
```
cd /home/user/Workspace/openipc-2.1/br-ext-chip-xiongmai/package/majestic-xm510/
ln -s majestic.mk majestic-xm510.mk
```

After this the compilation works fine till it tries to download the majestic packages, where it generates a 403 error with amazon aws:
```
>>> majestic-xm510 current Downloading
--2021-08-08 14:38:49--  http://openipc.s3-eu-west-1.amazonaws.com/majestic.xm510.master.tar.bz2
Resolving openipc.s3-eu-west-1.amazonaws.com (openipc.s3-eu-west-1.amazonaws.com)... 52.218.117.82
Connecting to openipc.s3-eu-west-1.amazonaws.com (openipc.s3-eu-west-1.amazonaws.com)|52.218.117.82|:80... connected.
HTTP request sent, awaiting response... 403 Forbidden
2021-08-08 14:38:50 ERROR 403: Forbidden.
```

Thanks

schneemaier, 2021-08-19:
Thanks for the fixes for xm530 it works now

closed 2021-08-19 by schneemaier

ZigFisher, 2021-08-23:
Thank you for providing the report, sir !