OpenIPC/firmware #40: No filesystem could mount root on Escam PVR008

001Tycoon, 2021-08-22:
Hello
Model Escam PVR008 
Board S834-A0-B 
Sensor F23
CPU Hi3518ERNCV300
Network Realtk 8188ETV (M8188FU3)

openipc-2.1  commit ab38bfa0824ec2fd8eb8ef7cb4cf6e9f404cca02
serial flash
u-boot 0x000000 Length 0x100000
uImage 0x100000 Length 0x200000
rootfs.squashfs 0x300000 Length 0x500000

bootargs=mem=${osmem:-32M} console=ttyAMA0,115200 panic=5 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
bootcmd=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read 0x42000000 0x100000 0x300000; bootm 0x42000000

System startup

Uncompress Ok!

U-Boot 2020.01 (Sep 02 2020 - 17:34:02 +0800)hi3518ev300

DRAM:  SPI Nor:  hifmc_ip_ver_check(54): Check Flash Memory Controller v100 ...hifmc_ip_ver_check(60):  Found
hifmc_spi_nor_probe(2070): SPI Nor ID Table Version 1.0
hifmc_spi_nor_probe(2095): SPI Nor(cs 0) ID: 0x20 0x70 0x18
hifmc_init_print(2016): Block:64KB hifmc_init_print(2017): Chip:16MB hifmc_init_print(2018): Name:"XM25QH128A"
hifmc100_spi_nor_probe(145): SPI Nor total size: 16MB
NAND:  0 MiB
Loading Environment from SPI Flash... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0
Error: eth0 address not set.

Hit any key to stop autoboot:  0
device 0 offset 0x100000, size 0x300000

SF: 3145728 bytes @ 0x100000 Read: OK
   Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1978216 Bytes = 1.9 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (ingo@acer-p256-m) (gcc version 7.5.0 (Buildroot -gab38bfa) ) #1 Sun Aug 22 09:27:48 CEST 2021
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3518EV300 DEMO Board
cmz zone is not set!
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=5 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 26776K/32768K available (4084K kernel code, 148K rwdata, 912K rodata, 176K init, 243K bss, 5992K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
[... 187 lines cut ...]
board=hi3518ev300
board_id=MHgyMDdm
board_name=hi3518ev300
bootargs=mem=${osmem:-32M} console=ttyAMA0,115200 panic=5 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
bootcmd=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read 0x42000000 0x100000 0x300000; bootm 0x42000000
bootdelay=1
cpu=armv7
ethact=eth0
ipaddr=192.168.2.221
netmask=255.255.255.0
osmem=32M
ptzsupport=1
sensor_name=F23
serverip=192.168.2.40
soc=hi3518ev300
stderr=serial
stdin=serial
stdout=serial
totalmem=64M
vendor=hisilicon
verify=n

Environment size: 889/262140 bytes

what can I do that he finds the root filesystem

ZigFisher, 2021-08-23:
Good afternoon
Your type of flash drive is not supported in the current version of Linux in the openipc-2.1 project. 
Please send a photo of the flash chip (8 pin) or a text description.

dimerr, 2021-08-24:
Try with the latest commit/build

001Tycoon, 2021-08-24:
Hello
@dimerr Try with the latest commit/build Patch error 

```
patching file scripts/Makefile.clean
patching file scripts/setlocalversion
patching file scripts/dtc/dtc-lexer.lex.c_shipped
patching file drivers/mtd/spi-nor/spi-nor.c
patching file drivers/mtd/spi-nor/spi-nor.c
patching file drivers/mtd/spi-nor/spi-nor.c
patching file drivers/mtd/spi-nor/spi-nor.c
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 1219 with fuzz 2.
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/drivers/mtd/spi-nor/spi-nor.c<--->2021-08-24 08:00:01.771026933 +0300
|+++ b/drivers/mtd/spi-nor/spi-nor.c<--->2021-08-24 08:04:33.973630596 +0300
--------------------------
File to patch: 
```

@ZigFisher 
[image]

With other settings it is boot up
```
bootargs=mem=${osmem} console=ttyAMA0,115200 panic=5 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:960k(boot),64k(env),2048k(kernel),6144k(rootfs),-(rootfs_data)
bootcmd=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read 0x41000000 0x100000 0x200000; bootm 0x41000000
```
```
SF: 2097152 bytes @ 0x100000 Read: OK
## Booting kernel from Legacy Image at 41000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1980080 Bytes = 1.9 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (ingo@acer-p256-m) (gcc version 7.5.0 (Buildroot -gcc7fac4-dirty) ) #1 Mon Aug 23 22:56:57 CEST 2021
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3518EV300 DEMO Board
cmz zone is not set!
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=5 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:960k(boot),64k(env),2048k(kernel),6144k(rootfs),-(rootfs_data)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 26772K/32768K available (4085K kernel code, 149K rwdata, 916K rodata, 172K init, 243K bss, 5996K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc04056d8   (4086 kB)
      .init : 0xc04ed000 - 0xc0518000   ( 172 kB)
      .data : 0xc0518000 - 0xc053d760   ( 150 kB)
       .bss : 0xc053f000 - 0xc057bcc8   ( 244 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
[... 115 lines cut ...]
FAIL
Starting ntpd: OK
Starting dropbear sshd: OK
Starting mini-snmpd: OK
hisilicon: Check MAC for Xiongmai devices
Cannot parse config file '/etc/fw_env.config': No such file or directory
Cannot parse config file '/etc/fw_env.config': No such file or directory
hisilicon: The eth0 interface has a correct MAC -
hisilicon: Loading of kernel modules and initialization of the video system has started
get_hisi_chip_id() got unexpected 0x3518e300
Cannot parse config file '/etc/fw_env.config': No such file or directory
Cannot parse config file '/etc/fw_env.config': No such file or directory
mmz_start: 0x42000000, mmz_size: 32M
Cannot parse config file '/etc/fw_env.config': No such file or directory
get_hisi_chip_id() got unexpected 0x3518e300
rmmod: can't unload module 'cma_osal': No such file or directory
hisilicon: Get data from ipcinfo and set SENSOR as unknown
Cannot parse config file '/etc/fw_env.config': No such file or directory
Error: environment not initialized
Starting majestic: OK

Welcome to OpenIPC v2.1
hi3518ev300-openipc login:

```

dimerr, 2021-08-24:
Sorry ,copy-paste issue, try again

001Tycoon, 2021-08-24:
@dimerr Thanks works

System booted 
i have a problem with fw_env.config
root@hi3518ev300-openipc:~# fw_printenv
Cannot parse config file '/etc/fw_env.config': No such file or directory

what information do you need

dimerr, 2021-08-24:
I believe you need to replace bootloader...

001Tycoon, 2021-08-24:
I tried that 
```
System startup

Uncompress Ok!

U-Boot 2016.11 (May 09 2019 - 15:47:33 +0800)hi3518ev300

Relocation Offset is: 03737000
Relocating to 43f37000, new gd at 43e96ef0, sp at 43e96ed0
SPI Nor:  NAND:  0 MiB
MMC:
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
hisilicon #
```
```
System startup

Uncompress Ok!

U-Boot 2020.01 (Sep 02 2020 - 17:34:02 +0800)hi3518ev300

DRAM:  SPI Nor:  hifmc_ip_ver_check(54): Check Flash Memory Controller v100 ...hifmc_ip_ver_check(60):  Found
hifmc_spi_nor_probe(2070): SPI Nor ID Table Version 1.0
hifmc_spi_nor_probe(2095): SPI Nor(cs 0) ID: 0x20 0x70 0x18
hifmc_init_print(2016): Block:64KB hifmc_init_print(2017): Chip:16MB hifmc_init_print(2018): Name:"XM25QH128A"
hifmc100_spi_nor_probe(145): SPI Nor total size: 16MB
NAND:  0 MiB
Loading Environment from SPI Flash... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0
Error: eth0 address not set.

Hit any key to stop autoboot:  0
hisilicon #
```
the same mistakes everywhere
where can I get other bootloaders

dimerr, 2021-08-24:
https://github.com/OpenIPC/openipc-2.1/releases/download/latest/u-boot-hi3518ev300-beta.bin try this

001Tycoon, 2021-08-24:
Hello
@dimerr u-boot-hi3518ev300-beta.bin
After saving the settings saveenv system does not boot.
```
hisilicon # saveenv
Saving Environment to SPI Flash...
Erasing SPI flash...Writing to SPI flash...done
hisilicon # reset
resetting ...

System startup

```
nothing more 

boots without saveenv 
```
System startup

Uncompress Ok!

U-Boot 2016.11-g0c517e3-dirty (Aug 23 2021 - 16:24:40 +0300)hi3516ev200

Relocation Offset is: 03750000
Relocating to 43f50000, new gd at 43f0fef0, sp at 43f0fed0
SPI Nor:  hifmc_ip_ver_check(44): Check Flash Memory Controller v100 ...hifmc_ip_ver_check(50):  Found
hifmc_spi_nor_probe(1664): SPI Nor ID Table Version 1.0
hifmc_spi_nor_probe(1689): SPI Nor(cs 0) ID: 0x20 0x70 0x18
hifmc_spi_nor_probe(1754): Block:64KB hifmc_spi_nor_probe(1755): Chip:16MB hifmc_spi_nor_probe(1756): Name:"XM25QH128A"
hifmc100_spi_nor_probe(147): SPI Nor total size: 16MB
NAND:  0 MiB
MMC:
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   eth0
Warning: eth0 (eth0) using random MAC address - 42:48:5c:c5:d2:a3

Hit any key to stop autoboot:  0
device 0 offset 0x50000, size 0x200000

SF: 2097152 bytes @ 0x50000 Read: OK
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1978264 Bytes = 1.9 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (ingo@acer-p256-m) (gcc version 7.5.0 (Buildroot -g917914b) ) #2 Tue Aug 24 15:03:27 CEST 2021
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3518EV300 DEMO Board
cmz zone is not set!
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 26776K/32768K available (4084K kernel code, 148K rwdata, 912K rodata, 176K init, 243K bss, 5992K reserved, 0K cma-reserved)
Virtual kernel memory layout:
[... 130 lines cut ...]
FAIL
Starting ntpd: OK
Starting dropbear sshd: OK
Starting mini-snmpd: OK
hisilicon: Check MAC for Xiongmai devices
Cannot parse config file '/etc/fw_env.config': No such file or directory
Cannot parse config file '/etc/fw_env.config': No such file or directory
hisilicon: The eth0 interface has a correct MAC -
hisilicon: Loading of kernel modules and initialization of the video system has started
get_hisi_chip_id() got unexpected 0x3518e300
Cannot parse config file '/etc/fw_env.config': No such file or directory
Cannot parse config file '/etc/fw_env.config': No such file or directory
mmz_start: 0x42000000, mmz_size: 32M
Cannot parse config file '/etc/fw_env.config': No such file or directory
get_hisi_chip_id() got unexpected 0x3518e300
rmmod: can't unload module 'cma_osal': No such file or directory
hisilicon: Get data from ipcinfo and set SENSOR as unknown
Cannot parse config file '/etc/fw_env.config': No such file or directory
Error: environment not initialized
Starting majestic: OK

Welcome to OpenIPC v2.1
hi3518ev300-openipc login:
```
other logs are required?

dimerr, 2021-08-24:
Try same link, updated

001Tycoon, 2021-08-24:
@dimerr u-boot-hi3518ev300-beta.bin is running now 
saveenv is work now 
fw_printenv  fw_setenv is work now.
Thanks very much

closed 2021-08-24 by dimerr

Droid-MAX, 2022-11-25:
@dimerr this link are no longer availiable, where can I download that uboot file? thx

cronyx, 2022-11-28:
https://github.com/OpenIPC/firmware/releases