OpenIPC/firmware #64: Issue on Escam QF518 with XM530/XM550 and SC5332

Nickbert7, 2021-11-03:
I don't get any picture when I try to run the latest firmware (2.1.11.03) on an Escam QF518.
I tried the XM550 and XM530 firmware.

Unfortunately when I try to run majestic manual I receive following result:
```
root@openipc-xm530:~# majestic
LIBH264: Complied at Apr 30 2019 16:55:22
23:39:45 [    main] main@124                      Majestic Free for Xiongmai, version master+70ebb25, built on 2021-11-03
23:39:45 [app_conf] load_config@122               Using /etc/majestic.yaml as main configuration
23:39:45 [     sdk] sdk_specific_config@777       sdk_specific_config()
23:39:45 [watchdog] watchdog_start@85             Watchdog timeout set to 10 seconds
DEBUG: [ISP_SDK]BUILD TIME:Sep 21 2021 11:58:04
DEBUG: OSC:12M
23:39:45 [     log] printf@229                    SYS/VI set in [ Sofia ]
23:39:45 [     log] printf@229                    Vstd_init     [ PAL ]
23:39:45 [     log] printf@229                    Rslt  [ 0 ]
23:39:45 [     log] printf@229                    IPC_Venc:     [ H.264 ]
23:39:45 [     log] printf@229                    Sns_IF:       [ By src ]

DEBUG: sensor_get_chip:
23:39:47 [     log] printf@229                    0x36
 23:39:47 [     log] printf@229                    0x2
 23:39:47 [     log] printf@229                    0x1
 23:39:47 [     log] printf@229                    0x0
23:39:47 [    puts]
DEBUG: DspChip:XM530
DEBUG: SnsIF:DVP
DEBUG: Vstd:PAL[25fps]
ERR: Not Support this sensor!
ERR: sensor_register_callback failed!
DEBUG: isp_sample end!
DEBUG: ProductType: 0x0
ERR: libfvideo: not support this product! 0x82
23:39:47 [     sdk] ViChnConfig@124               ViChn = 0, enNorm = 0, enSize = 14
23:39:47 [     sdk] COMM_VENC_Start@175           VencChn = 0,enType = 265,enNorm = 0,enSize = 14,enRcMode = 0
23:39:47 [     sdk] COMM_SYS_GetPicSize@152       enNorm = 0,enPicSize = 14
DEBUG: black and white
DEBUG: EShutter 1/1
23:39:49 [     sdk] COMM_VENC_Start@447           XM_MPI_VENC_CreateChn [0] failed with 0xa0078007!
23:39:49 [     sdk] start_sdk@663                 Start VencChn 0 failed!
23:39:49 [     sdk] sdk_dealloc_static_bufs@707   sdk_dealloc_static_bufs()
23:39:49 [watchdog] watchdog_stop@126             Watchdog closed
23:39:49 [    main] main@194                      Shutdown main thread
root@openipc-xm530:~#
```

ipctool result:

```
root@openipc-xm530:~# ipctool
---
board:
  vendor: Xiongmai
chip:
  vendor: Xiongmai
  model: XM550
ethernet:
  mac: "00:12:41:b5:bc:51"
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 7fd8e955
        contains:
          - name: xmcrypto
            offset: 0x2fc00
          - name: uboot-env
            offset: 0x30000
      - name: env
        size: 0x10000
        sha1: b0152796
      - name: kernel
        size: 0x200000
        sha1: 48c811df
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: 712470fd
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
ram:
  total: 64M
  media: 32M
firmware:
  kernel: "3.10.103+ (SMP Wed Nov 3 14:59:04 UTC 2021)"
  toolchain: gcc version 7.5.0 (Buildroot -gf380b16)
sensors:
- vendor: SmartSens
  model: SC5332
  control:
    bus: 0
    type: i2c
    addr: 0x30
root@openipc-xm530:~#

```

What is irritating for me is that majestic tells me XM530 and ipctool XM550

Here is a picture of the SOC:
[image]

Here is the device:
https://www.banggood.com/ESCAM-QF518-5MP-Pan-or-Tilt-AI-Humanoid-Detection-Auto-Tracking-Cloud-Storage-Waterproof-WiFi-IP-Camera-with-Two-Way-Audio-Night-Vision-p-1731086.html

widgetii, 2021-11-04:
According to your logs:

`ERR: Not Support this sensor!`

XiongMai SDK doesn't support this sensor

closed 2021-11-04 by widgetii

Nickbert7, 2021-11-04:
Oh, too bad... 
Do you see any chance that it might be supported somewhen in the future?

widgetii, 2021-11-04:
I doubt so, nobody from the core team has such hardware

kodidu, 2021-11-14:
Hello!
Maybe you need to copy this file from the original firmware dump like I did:
/mnt/mtd/Config/SensorType.bat otherwise majestic showed some error message that SensorType.bat was not found.

widgetii, 2021-11-14:
It didn't change the situation because there is no sensor driver in the software