OpenIPC/firmware #71: XM530 rtsp does not work

kodidu, 2021-11-11:
Hello!

Using openipc 2.1 firmware for xm530.
Majestic only works in hls h265 mode, mp4, jpg, rtsp does not work....
Should this other modes also work with the current state of XM530 development?
How can I provide more info to get rtsp working?

See also https://github.com/OpenIPC/ipctool/issues/30

widgetii, 2021-11-12:
JPG snapshots and specific settings for video channels are under development now but RTSP should work (at least everything is ok on my hardware)

Could you provide your logs while RTSP client connects (`killall majestic; majestic`) and what client do you use?

kodidu, 2021-11-12:
Using VLC with rtsp://10.0.0.99/stream=0
VLC seems to connect fine e.g. play button switches to pause button and codec information shows hvec h265.

http://10.0.0.99/video.mp4 seems to work with VLC with 10s+ delay. 
Previously I only tried it in the browser, maybe the browser does not like h265.
yucca does still not support h265 (https://gitlab.com/yuccastream/yucca/-/issues/729).
So i want to activate h264 mode.
Is h264 mode under development, because this setting does not change the reported codec?
```
video0:
  enabled: true
  codec: h264
```

I already copied this file from the original firmware dump:
`/mnt/mtd/Config/SensorType.bat` otherwise majestic showed some error message.

```
root@openipc-xm530:~# killall majestic; majestic
killall: majestic: no process killed
LIBH264: Complied at Apr 30 2019 16:55:22
19:42:39 [    main] main@124                      Majestic Free for Xiongmai, version master+9fcd445, built on 2021-11-08
19:42:39 [app_conf] load_config@122               Using /etc/majestic.yaml as main configuration
19:42:39 [     sdk] sdk_specific_config@777       sdk_specific_config()
19:42:39 [watchdog] watchdog_start@85             Watchdog timeout set to 10 seconds
DEBUG: [ISP_SDK]BUILD TIME:Sep 21 2021 11:58:04
DEBUG: OSC:12M
19:42:39 [     log] printf@229                    SYS/VI set in [ Sofia ]
19:42:39 [     log] printf@229                    Vstd_init     [ PAL ]
19:42:39 [     log] printf@229                    Rslt  [ 0 ]
19:42:39 [     log] printf@229                    IPC_Venc:     [ H.264 ]
19:42:39 [     log] printf@229                    Sns_IF:       [ By src ]

DEBUG: SnsCfgFile: 55
DEBUG: sensor_get_chip:
19:42:39 [     log] printf@229                    0x40
 19:42:39 [     log] printf@229                    0x1
 19:42:39 [     log] printf@229                    0x1
 19:42:39 [     log] printf@229                    0x37
19:42:39 [    puts]
DEBUG: Venc:495
DEBUG: DspChip:XM530
DEBUG: SnsIF:MIPI
DEBUG: Vstd:PAL[25fps]
DEBUG: ------------- SOIF37 2M 25fps  init ok! (@20190909)----------------
DEBUG: sensor_init Ok!
DEBUG: cmos_fps_set F37 2M 25fps
DEBUG: cmos_fps_set F37 2M 25fps
DEBUG: Init over ~~~~~~~~
DEBUG: ProductType: 0x0
DEBUG: libfvideo: F37!
DEBUG: isp_sample end!
DEBUG: imageStyle XM!
19:42:39 [     sdk] ViChnConfig@124               ViChn = 0, enNorm = 0, enSize = 14
19:42:39 [     sdk] COMM_VENC_Start@175           VencChn = 0,enType = 265,enNorm = 0,enSize = 14,enRcMode = 0
19:42:39 [     sdk] COMM_SYS_GetPicSize@152       enNorm = 0,enPicSize = 14
19:42:39 [root_med] RootMediaSupplier_new@10      new RootMediaSupplier(NAL)
19:42:39 [root_med] RootMediaSupplier_new@10      new RootMediaSupplier(MJPEG)
19:42:39 [  evhttp] new_http_server@288           HTTP server started on 0.0.0.0:80
19:42:39 [root_med] RootMediaSupplier_new@10      new RootMediaSupplier(MP4)
19:42:39 [root_med] RootMediaSupplier_new@10      new RootMediaSupplier(MP4A)
19:42:39 [    rtsp] rtsp_init@153                 RTSP server started on port 554
DEBUG: black and white
DEBUG: EShutter 1/1
DEBUG: color
19:42:41 [     sdk] sdk_fps_by_stream@724         sdk_fps_by_stream(0)
19:42:41 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:42:41 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:42:42 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:42:43 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
[... 52 lines cut ...]
19:43:16 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:17 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:18 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:19 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:19 [watchdog] watchdog_keepalive@103        Watchdog keepalive
19:43:20 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:21 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:22 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:23 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:24 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:24 [  client] RtspClient_dispatch@405       RTSP TEARDOWN rtsp://10.0.0.99:554/stream=0 CSeq=6 from 10.0.0.123:50929
19:43:24 [    rtsp] conn_eventcb@66               RTSP connection closed.
19:43:24 [watchdog] watchdog_keepalive@103        Watchdog keepalive
19:43:25 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:26 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
19:43:27 [     sdk] sdk_secsInGop_by_stream@747   sdk_secsInGop_by_stream(0)
^C19:43:27 [     log] printf@229                    Interrupt signal received
19:43:27 [root_med] RootMediaSupplier_drop@37     drop RootMediaSupplier(MP4)
19:43:27 [root_med] RootMediaSupplier_drop@37     drop RootMediaSupplier(MP4A)
19:43:27 [root_med] RootMediaSupplier_drop@37     drop RootMediaSupplier(NAL)
19:43:27 [root_med] RootMediaSupplier_drop@37     drop RootMediaSupplier(MJPEG)
19:43:27 [     sdk] sdk_dealloc_static_bufs@707   sdk_dealloc_static_bufs()
19:43:27 [watchdog] watchdog_stop@126             Watchdog closed
19:43:27 [    main] main@194                      Shutdown main thread
```

widgetii, 2021-11-13:
Xiongmai support is in MVP state now (only main video stream is supported, no image snapshots and additional settings) and we're looking for sponsors to continue development for XM platform.

But RTSP should work currently and I've checked described issue on XM530 camera and found a similar issue with VLC (it shows nothing but other players do, for example, `ffplay`).

It seems that VLC cannot parse NALs in HEVC bitstream produced by XM530 encoder, but for me upgrade from vlc 3.0.8 to 3.0.16 resolved the issue. Could you try do the same?

kodidu, 2021-11-14:
A newer VLC works, but more settings (h264) would be great.
So XM530 has no focus and its is better to stick with e.g. 3516CV300.
Last time I ordered a 3516CV300 but got a XM530...

closed 2021-11-14 by kodidu

widgetii, 2021-11-14:
Latest build also supports H264/H265 codec selection

kodidu, 2021-11-14:
Seems to work, but getting random scrambled video every second or so
[image]
.

widgetii, 2021-11-15:
Do you use WiFi between client and camera? Try also `vlc --rtsp-tcp rtsp://10.0.0.99`

kodidu, 2021-11-15:
nope, direct LAN connection, also happens with video.mp4 stream