OpenIPC/firmware #74: SerComm camera contribution

johnwc, 2021-11-27:
We have a few different models of SerComm cameras. I was able to open one up and get UART access as root, and I was able to pull the firmware off to a file. This specific camera is ambarella based. I was able to mount the NFS share and run ipctool. Running without any options caused the camera to reboot. Here is some info that I thought was important running it with dmesg option.

```
Linux version 2.6.38.8 (arvin@ubuntu) (gcc version 4.6.1 (Sourcery CodeBench Lite 2011.09-70) ) #4 PREEMPT Thu Jun 8 11:00:05 CST 2017
CPU: ARMv6-compatible processor [4117b365] revision 5 (ARMv6TEJ), cr=00c5387f
CPU: VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Machine: Coconut
Kernel command line: console=ttyS0 root=/dev/mtdblock5 ro rootfstype=squashfs init=/linuxrc bootimageid=2 video=amb0fb:720x480,720x480,1,0
Ambarella Coconut:
      chip id:                5100
      board type:             3
      board revision:         10
      chip name:              a5m
      HAL version:            207871
      reference clock:        24000000
      system configuration:   0x040004ea
      boot type:              0x00000002
      hif type:               0x00000000
RT6352iNIC: 802.11n WLAN MII driver v2.7.0.3 (July 26, 2013)
```

I am a seasoned developer and would like to contribute to OpenIPC to have it support SerComm cameras. How can I get started?

ZigFisher, 2021-11-27:
Good afternoon
We need more information and a photo of the device, and preferably the processor name. 
Or remove and send a dump from the device.

ZigFisher, 2021-11-27:
Thanks
Ambarella chips are not supported by us yet

johnwc, 2021-11-27:
I think you missed the last part of my post. `I am a seasoned developer and would like to contribute to OpenIPC to have it support SerComm cameras. How can I get started?`

widgetii, 2021-11-28:
There is nothing to do for the project without a platform SDK because everything needs to be built from source code

johnwc, 2021-11-28:
Does anyone have access to the Ambarella SDK?

widgetii, 2021-11-28:
Everything from Ambarella is under strict NDA.
If you want to help the project you may choose any SoC from https://github.com/OpenIPC/firmware#downloads-for-research-only-without-any-guarantee- page with `help needed` label.

closed 2021-11-30 by widgetii