OpenIPC/firmware #75: Error in host-m4-1.4.18/lib/c-stack.c breaks build

carbofos, 2021-11-29:
During building  i am getting the next error

```
In file included from /usr/include/signal.h:328,
                 from ./signal.h:52,
                 from c-stack.c:49:
c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384

```

I did try to build some different configurations and this bug does not depend on it as I can see.

The problem is in this file:

./openipc-2.1/output/build/host-m4-1.4.18/lib/c-stack.c

I did remove preprocessor directive lines so the code could be compiled now:

#ifndef SIGSTKSZ
#define SIGSTKSZ 16384
#endif

But I am not sure if the  bug descibed in the comment of the c-stack.c is still actual ( there is a whitespace between # and define so it's interpreted as a comment by preprocessor )

widgetii, 2022-01-09:
Couldn't confirm the issue on Github Actions used for firmware CI.

Please use Docker environments to remove differences between platforms

closed 2022-01-09 by widgetii

iddq, 2023-03-06:
Is this a compiler bug?

tim-ka, 2024-01-16:
Currently buildroot using m4-1.4.19 without such kind of problems so workaround could consist removing patches and replace m4.hash and m4.mk to recent versions for m4-1.4.19. I met the same issue building OpenIPC firmware on Ubuntu 22.04. Unfortunatelly it's not only one issue with Ubuntu 22.04 (next one will be segfault crash for make ver. 4.3 on printvars) so I decide to downgrade the building environment to 20.04

widgetii, 2024-01-17:
We use `ubuntu-latest` on GitHub CI which is Ubuntu 22.04