OpenIPC/firmware #122: Рефакторинг Github Actions

p0isk, 2021-12-24:
Сейчас все релизы собираются в одном старом сообщении, куда вручную загружены файлы и описание. Предлагаю:
- разделить sdk-buildroot, firmware и u-boot на разные релизы
- обновлять не только файлы, но данные самого релиза, для того чтобы:
  - получать данные о версии
  - делать заметки к релизу: накопительное/критическое обновление
- вынести повторяющийся и опциональный код из workflow-файлов
- запускать сборку под все платформы из одного workflow

Посмотреть, на что можно поменять старый actions/create-release@v1, svenstaro/upload-release-action@v2 в https://github.com/marketplace?type=actions

widgetii, 2022-09-12:
Было сделано:

1. Разделены на этапы сборка тулчейнов (с кэшированием в релизах) и firmware. U-Boot полагаю, что имеет смысл собирать через CI в своих собственных репо
2. Workflow был заменен на универсальный

themactep, 2023-05-07:
so, is it solved? the issue looks stale.

closed 2023-11-19 by viktorxda