OpenIPC/firmware #126: Anyka ak3918

bolt1502, 2021-12-25:
Hello.This chip is supported ak3918 ?

ZigFisher, 2021-12-25:
Hello
This character is in the plans for the future
While there is no support, you need to wait

ricardojlrufino, 2022-02-26:
I make progress in reverse engineering of camera model: v380

# Info Boot
[13:59:32:303] Anyka Linux Kernel Version: 2.5.02
[13:59:32:343] Booting Linux on physical CPU 0
[13:59:32:343] Linux version 3.4.35 (ma@ma-PC) (gcc version 4.8.5 (anyka (gcc-4.8.5 + binutils-2.24 + ulcibc-0.9.33.2)(20170223)) ) #5 Wed Sep 18 15:36:19 CST 2019
[13:59:32:343] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[13:59:32:343] CPU: VIVT data cache, VIVT instruction cache
[13:59:32:366] Machine: Cloud39E_AK3918E+H42_V1.0.2
[13:59:32:366] Memory policy: ECC disabled, Data cache writeback
[13:59:32:366] ANYKA CPU AK3918 (ID 0x20150200)

I got: 
read raw image
get mic input
audio output.

But this using original Kernel and Libraries.

And samples from:
files extrated from: http://www.iotelink.com/frank/ak3918ev300v18
and from: https://github.com/helloworld-spec/qiwen/tree/main/anycloud39ev300

I managed to compile some examples, but the original camera libraries (.so) are slightly different from the original, and there are some limitations.

One problem is being able to use Hardware Video Decoder/Encoder (ISP), because the file 'isp_sc1245.conf" is incompatible with the libraries of (ak3918ev300v18 repository)

--
I tried to use OpenIPC, but neither rootfs nor Kernel does not boot my camera.

dimerr, 2022-06-13:
dup https://github.com/OpenIPC/firmware/issues/207

closed 2022-06-13 by dimerr

bolt1502, 2022-06-13:
Спасибо.Значит я в пролёте,жаль.Буду ждать,может кто дамп выкинет на просторы.

> 
> 
>Closed  #126 as not planned.

berserkerkz, 2024-01-12:
> Спасибо.Значит я в пролёте,жаль.Буду ждать,может кто дамп выкинет на просторы.  
>     -- Alex Alex Отправлено из Почты Mail.ru

У меня есть дамп

berserkerkz, 2024-03-12:
да не вопрос, скинул на почту.