OpenIPC/firmware #131: RTSP stream authentication,need some help

sh1970, 2021-12-30:
The current version can directly play RTSP streams without authentication. This is insecure. Please help what configuration needs to be modified for RTSP stream authentication.

widgetii, 2021-12-30:
Currently, RTSP authentication is not supported

esuslik, 2022-01-18:
многие облачные сервисы а также по для видеонаблюдения используют rtsp как универсальное протокол для подключения потока. Камера может оказаться открыта через интернет наружу для подключения и нужна обязательно авторизация на любой поток и протокол. Особенно камеры стоят в коммерческих организациях.

widgetii, 2022-01-19:
We do have it in the project backlog but with low priority for now

widgetii, 2022-05-11:
Implemented in recent version