OpenIPC/firmware #137: Contribution to Wiki - adding xm510 install instructions

schneemaier, 2022-01-03:
Hi, unfortunatelly github does not support pull requests for Wiki artikles. I had some time and tested my xm510 camera with the firmware. I tried to follow the intrsuctions for xm530/550 boards with some modifications and documented the results in the md files.

Attached to this bug to wiki files:
- install_xm510.md - installation instructons for xm510 board
- _Sidebar.md - modified to include xm510 insatllation link
[_Sidebar.md](https://github.com/OpenIPC/firmware/files/7799997/_Sidebar.md)
[install_xm510.md](https://github.com/OpenIPC/firmware/files/7799998/install_xm510.md)

Thanks

ZigFisher, 2022-01-03:
Good afternoon, sir!
Thank you so much for your work.
Your information has been added to the Wiki.
Yes, we are considering alternative options for maintaining documentation, working with the current Wiki collectively is damn inconvenient..

ZigFisher, 2022-01-03:
I see you're a big fan of Ingenic, sir.
We plan to support the entire line in the near future. We already have boards based on T10, T20, T21, T31... 
I hope this will also be interesting to you.

schneemaier, 2022-01-06:
Thanks for the Wiki update! I will definitely try the Ingenic version too

For the issues, experience i have with xm510, should i open separate bugs? Or should i dump all findings into one report?

ZigFisher, 2022-01-06:
Do as you like, it is better of course to break errors and edits into small parts.

closed 2022-01-06 by ZigFisher