OpenIPC/firmware #140: Driver for the sensor IMX385

gitgayhub, 2022-01-04:
Hi, first I want to thank you for this wonderful project. I have a camera with 3516CV300 and a 1/1.8' Sony sensor IMX385. I have manged to install OpenIPC to the camera successfully, but only to find that the sensor IMX385 is not supported.

I have found the driver source code for IMX385, but don't have the skills to cross-build the ko module, the driver source code can be found at [](https://github.com/ZigFisher/Glutinium/tree/52f59a674caea013d135014021047008380a6232/hisi-sensors/src/sony_imx385__soc_v3)

I'm wondering if you can add the driver for IMX385 to this project, or if you can tell more about how to cross-compile the driver myself

Thank you very much

dimerr, 2022-01-04:
hi, i can add the driver, no problem, but are you able to make sensor config yourself?

gitgayhub, 2022-01-04:
@dimerr Hi, thanks for the quick reply. Honestly, I have no idea about the config file either, I’m not a tech guy. Is it hard to write? Can I just modify the existing config file for other Sony sensor like imx291? Thanks

dimerr, 2022-01-05:
Well, yes, you can modify, but you need to know sensor connection type and some other properties. Does ipctool give you any hints?

gitgayhub, 2022-01-05:
@dimerr ipctool inside OpenIPC doesn't give any sensor info, since there's no driver for imx385.  And I can't get shell access for the manufacture's original firmware to run ipctool. However I managed to get a dump of the entire flash and extracted the file app.tar.gz which contains the driver and config information, can you please take a look and see if I can do anything with that? Thanks

[app.tar.gz](https://github.com/OpenIPC/firmware/files/7812394/app.tar.gz)

root@openipc-hi3516cv300:/tmp# ./ipctool
---
board:
  vendor: OpenIPC
  version: 2.2.01.04
chip:
  vendor: HiSilicon
  model: 3516CV300
ethernet:
  mac: "00:24:b8:ff:ff:ff"
  u-mdio-phyaddr: 1
  phy-id: 0x001cc816
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 979deb21
      - name: env
        size: 0x40000
        sha1: a5f9dc40
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel
        size: 0x280000
        sha1: f6b96c16
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: 2bd91069
      - name: rootfs_data
        size: 0x800000
        path: /overlay,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
firmware:
  u-boot: "2010.06 (Mar 21 2018 - 18:48:28)"
  kernel: "3.18.20 (Tue Jan 4 04:13:46 UTC 2022)"
  toolchain: gcc version 7.5.0 (Buildroot -g6cd560e)

gitgayhub, 2022-01-05:
Just in case you may need to analyse for further project development, I've attached the 4 partition dumps from the original flash, it's a 16MB flash, mtd layout as below, the app.tar.gz file is inside usr partition

```
Creating 5 MTD partitions on "hi_sfc":
0x000000000000-0x000000080000 : "boot"
0x000000080000-0x000000300000 : "kernel"
0x000000300000-0x000000580000 : "rootfs"
0x000000580000-0x000000c80000 : "usr"
0x000000c80000-0x000001000000 : "ext"
```
[3516cv300_imx385_dump.zip](https://github.com/OpenIPC/firmware/files/7812514/3516cv300_imx385_dump.zip)

part of printenv:
```
bootargs=mem=72M console=ttyAMA0,115200 root=/dev/mtdblock2 rootfstype=jffs2 rw mtdparts=hi_sfc:512K(boot),2560K(kernel),2560K(rootfs),7168K(usr),3584K(ext) coherent_pool=4M
bootcmd=sf probe 0;sf read 0x82000000 0x80000 0x280000;bootm 0x82000000
```

dimerr, 2022-01-05:
Thanks, that might be helpful. I'm gonna add imx385 module soon, but as for config file: unless you are able to provide remote ssh access to your cam, we will have to try 'guessing' config with many many tries here...

dimerr, 2022-01-06:
try sysupgrade

then run 

SENSOR=imx385 majestic

gitgayhub, 2022-01-06:
@dimerr Thank you very much. sysupgrade gave error message
 `Cannot parse config file '/etc/fw_env.config': No such file or directory
SOC not defined in U-Boot ENV...` 

Same error message `Cannot parse config file '/etc/fw_env.config': No such file or directory` appears several times during booting process. I have set soc=hi3516cv300 in uboot env

Since the sysupgarde command didn't work, so I upgraded in uboot environment by TFTP, and the `Cannot parse config file '/etc/fw_env.config' error message still there. `SENSOR=imx385 majestic` gave error message too

I'm using the original uboot from manufacturer,  do I have to update it to OpenIPC's uboot too? I was worried that I may brick the camera so I didn't update uboot

```root@openipc-hi3516cv300:~# SENSOR=imx385 majestic
07:09:07 [    main] main@135                      Majestic Lite for HiSilicon, version master+3f13a01, built on 2022-01-04
07:09:07 [app_conf] load_config@342               Using /etc/majestic.yaml as main configuration
07:09:07 [     sdk] sdk_specific_config@3532      SENSOR=imx385
07:09:07 [     sdk] find_sensor_config@3489       matched sensor config: imx385_i2c_1080p.ini
07:09:07 [     sdk] find_sensor_config@3506       Using /etc/sensors/imx385_i2c_1080p.ini as sensor configuration
07:09:07 [watchdog] watchdog_start@59             Watchdog device (/dev/watchdog) not found
07:09:07 [     hal] HiSi_HAL_start_logger@570     Cannot open /dev/logmpp
07:09:07 [     sdk] start_sdk@250                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
07:09:07 [     sdk] start_sdk@253                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
open sys: No such file or directory
07:09:07 [     sdk] start_sdk@262                 Cannot get chip id
07:09:07 [     sdk] start_sdk@262                 Unknown error code 2
07:09:07 [    main] main@227                      Shutdown main thread
```
printenv

```
gsf3516cv300 # print
bootdelay=1
baudrate=115200
bootfile="uImage"
hichip=3479235
paramMagic=1217463299
wifinetmask=255.255.255.0
wifiip=192.168.1.161
wifigateway=192.168.1.2
ethaddr=xxxxxx
wifimac=00:11:22:33:44:56
devType=11
diNum=1
doNum=1
chnNum=1
rs232Num=1
rs485Num=1
netCardNum=2
hwVer=34325266
coreVer=16843009
swVer=67196928
sensorType=18
serialId=xxxxxxxxxxx
devFeature=2
ivsType=0
protocol=3
productName=
user=admin
passwd=12345
cmdPort=8080
rtspPort=554
httpPort=80
rtmpPort=1935
istPort=9660
productDate=1522325569
ethDHCP=0
zoom=30
skipCheckSns=1
ircutPwl=1
p2pType=0
irFilter=2
language=0
companyName=
companyAddr=
viCfgMask=0
irManage=0
irSignal=0
stand=0
camName=Camera
charEncode=1
pppoeEnable=0
pppoeUser=
pppoePassword=
thirdPfType=0
thirdPfGrpAddr=
thirdPfLUrl=
thirdPfDevId=
thirdPfDevKey=
thirdPfDevMode=
resolutionLimit=0
defIma=136
cropEn=0
cropX=0
cropY=0
cropWidth=0
cropHeight=0
bootargs=mem=72M console=ttyAMA0,115200 root=/dev/mtdblock3 rootfstype=squashfs init=/init rw mtdparts=hi_sfc:256K(boot),256K(env),2560K(kernel),5120K(rootfs),-(rootfs_data)
bootcmd=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0;sf read 0x82000000 0x80000 0x280000;bootm 0x82000000
osmem=32M
totaolmem=128M
soc=hi3516cv300
stdin=serial
stdout=serial
stderr=serial
verify=n
ver=U-Boot 2010.06 (Mar 21 2018 - 18:48:28)
filesize=4E6000
fileaddr=82000000
gatewayip=192.168.1.1
netmask=255.255.255.0
ipaddr=192.168.2.28
serverip=192.168.2.88
```

dimerr, 2022-01-06:
create /etc/fw_env.config with the following contents:

`/dev/mtd1    0x0    0x40000    0x10000`

This is due to very non-standard env offset

dimerr, 2022-01-06:
But before you do that, can you also confirm that ipctool --printenv returns correct info?

gitgayhub, 2022-01-06:
> But before you do that, can you also confirm that ipctool --printenv returns correct info?

Yes, before creating /etc/fw_env.config, ipctool -printenv return the correct env info, same as the printenv result in uboot.

After creating /etc/fw_env.config with `/dev/mtd1 0x0 0x40000 0x10000`, the `Cannot parse config file '/etc/fw_env.config': No such file or directory` erorr message didn't appear anymore, but "SENSOR=imx385 majestic" still gave the same error result:

```
root@openipc-hi3516cv300:~# SENSOR=imx385 majestic
12:44:44 [    main] main@135                      Majestic Lite for HiSilicon, version master+3f13a01, built on 2022-01-04
12:44:44 [app_conf] load_config@342               Using /etc/majestic.yaml as main configuration
12:44:44 [     sdk] sdk_specific_config@3532      SENSOR=imx385
12:44:44 [     sdk] find_sensor_config@3489       matched sensor config: imx385_i2c_1080p.ini
12:44:44 [     sdk] find_sensor_config@3506       Using /etc/sensors/imx385_i2c_1080p.ini as sensor configuration
12:44:44 [watchdog] watchdog_start@59             Watchdog device (/dev/watchdog) not found
12:44:44 [     hal] HiSi_HAL_start_logger@570     Cannot open /dev/logmpp
12:44:44 [     sdk] start_sdk@250                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
12:44:44 [     sdk] start_sdk@253                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
open sys: No such file or directory
12:44:44 [     sdk] start_sdk@262                 Cannot get chip id
12:44:44 [     sdk] start_sdk@262                 Unknown error code 2
12:44:44 [    main] main@227                      Shutdown main thread
```
ipctool still didn't provide any sensor info either, does that mean I have to get shell access to the original firmware?

dimerr, 2022-01-06:
fw_setenv sensor imx385 then reboot and try

gitgayhub, 2022-01-06:
> fw_setenv sensor imx385 then reboot and try

Thanks. I did that, and confirmed that soc=hi3516cv300 and soc=imx385 exist in fw_printenv after rebooting. But still it's the same error message by running `SENSOR=imx385 majestic`

widgetii, 2022-01-06:
There is no /dev/sys device, probably kernel modules weren't loaded

gitgayhub, 2022-01-06:
Well that sounds beyond my ability, guess I have to give up this camera.

I have several other cameras from hikvision and dahua based on hi3516AV100 and imx307, but their uboot tool is limited, I can't write 3rd party uimage and rootfs into it. I have another uniview camera with hi3516DV300 and 2GB RAM/8GB EMMC,, but the sensor is the Chinese [SmartSens SC4210](https://www.smartsenstech.com/page?id=115), 1/1.8" 4MP, hard to get driver support

Anyway, thank you again for this great project and for the quick free support. I will test OpenIPC on more devices

dimerr, 2022-01-06:
Its not a problem to load modules.
try

load_hisilicon -i 

and see what happens. I'm sure you just lack something important in env

gitgayhub, 2022-01-06:
Thanks, `load_hisilicon -i` gave error message `[err] os_mem[72], over total_mem[64]`. After some trial I found that the mem size in bootargs should be smaller than 64M(don't know the reason), so I set it to 63M and I can see the following message before the login notice

```
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0x83f00000, mmz_size: 1M
insert audio
==== Your input Sensor type is imx385 ====
Starting majestic: OK

Welcome to OpenIPC v2.1
openipc-hi3516cv300 login: root
```
`load_hisilicon -i` and `SENSOR=imx385 majestic` gave new error message s

```
root@openipc-hi3516cv300:~# load_hisilicon -i
mmz_start: 0x83f00000, mmz_size: 1M
insmod: can't insert 'sys_config.ko': File exists
insmod: can't insert 'hi_osal.ko': File exists
******* Error: There's something wrong, please check! *****
```
```
root@openipc-hi3516cv300:~# SENSOR=imx385 majestic
14:34:51 [    main] main@135                      Majestic Lite for HiSilicon, version master+3f13a01, built on 2022-01-04
14:34:51 [app_conf] load_config@342               Using /etc/majestic.yaml as main configuration
14:34:51 [     sdk] sdk_specific_config@3532      SENSOR=imx385
14:34:51 [     sdk] find_sensor_config@3489       matched sensor config: imx385_i2c_1080p.ini
14:34:51 [     sdk] find_sensor_config@3506       Using /etc/sensors/imx385_i2c_1080p.ini as sensor configuration
14:34:51 [watchdog] watchdog_start@78             Detected Hisilicon Watchdog
14:34:51 [watchdog] watchdog_start@98             Watchdog timeout set to 10 seconds
14:34:52 [     sdk] start_sdk@250                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
14:34:52 [     sdk] start_sdk@253                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
14:34:52 [     sdk] start_sdk@272                 Can't load config /etc/sensors/imx385_i2c_1080p.ini
14:34:52 [    main] main@227                      Shutdown main thread

```

ipctool still didn't give sensor info. dmesg ended at multiple lines of this 

```
[  252.504162] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  252.508367] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  252.508558] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  252.508750] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
```

dimerr, 2022-01-06:
ipctool has no idea about this sensor, dont rely on it's output for now.

fw_setenv totalmem 128M
fw_setenv osmem 32M

gitgayhub, 2022-01-07:
> fw_setenv totalmem 128M
> fw_setenv osmem 32M

After executing these two commands, booting process stuck at `mmz_start: 0x82000000, mmz_size: 32M` and then keep rebooting, no matter what mem/osmem size I change in uboot env

dimerr, 2022-01-07:
how much mem you actually have there?

dimerr, 2022-01-07:
```
setenv bootargs "mem=32M console=ttyAMA0,115200 root=/dev/mtdblock3 rootfstype=squashfs init=/init rwmtdparts=hi_sfc:256K(boot),256K(env),2560K(kernel),5120K(rootfs),-(rootfs_data)"
setenv osmem 32M
setenv totalmem 128M
```

please keep env as close to installation manual as possible, I've just noticed that you have a lot of differences

gitgayhub, 2022-01-07:
> how much mem you actually have there?
I have 128M memory, the mem size in bootargs of the original firmware is 72M

> please keep env as close to installation manual as possible, I've just noticed that you have a lot of differences

Since the mtd layout of the flash is different from the manual, I modified some mtd offset to make it working, this is the original flash layout

```
0x000000000000-0x000000080000 : "boot"
0x000000080000-0x000000300000 : "kernel"
0x000000300000-0x000000580000 : "rootfs"
0x000000580000-0x000000c80000 : "usr"
0x000000c80000-0x000001000000 : "ext"
```
env info is stored from 0x40000 to 0x80000

The camera can boot to system now, but majestic error still the same

dimerr, 2022-01-07:
https://github.com/OpenIPC/firmware/blob/master/general/package/hisilicon-osdrv-hi3516cv300/files/sensor/config/imx385_i2c_1080p.ini try this config, replace in /etc/sensors/ dir

gitgayhub, 2022-01-07:
Thanks, it's getting close but still there's some error message, no image output

```
root@openipc-hi3516cv300:~# SENSOR=imx385 majestic
07:39:43 [    main] main@135                      Majestic Lite for HiSilicon, version master+3f13a01, built on 2022-01-04
07:39:43 [app_conf] load_config@342               Using /etc/majestic.yaml as main configuration
07:39:43 [     sdk] sdk_specific_config@3540      Using /etc/sensors/imx385_i2c_1080p.ini sensor configuration file
07:39:43 [watchdog] watchdog_start@78             Detected Hisilicon Watchdog
07:39:43 [watchdog] watchdog_start@98             Watchdog timeout set to 10 seconds
07:39:43 [     sdk] start_sdk@250                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
07:39:43 [     sdk] start_sdk@253                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
07:39:43 [    vpss] VPSS_DRV_MallocNrBuffer@1769  malloc vpss Ref&Mad buffer fail!
07:39:43 [    vpss] VPSS_CreateGrp@4258           [grp0]:drv ctx init fail!
07:39:43 [    vpss] VPSS_SetChnMode@5696          [grp0]:vpss unexist!
07:39:43 [    venc] VencGetChnAttr@14124          venc check create:venc 0 chn not create!
07:39:43 [    vpss] VPSS_SetChnMode@5696          [grp0]:vpss unexist!
07:39:43 [    vpss] VPSS_SetChnMode@5696          [grp0]:vpss unexist!
07:39:43 [    vpss] VPSS_SetChnMode@5696          [grp0]:vpss unexist!
07:39:43 [    venc] VencGetChnAttr@14124          venc check create:venc 0 chn not create!
07:39:43 [    venc] VencGetChnAttr@14124          venc check create:venc 0 chn not create!
07:39:43 [    venc] VencGetChnAttr@14124          venc check create:venc 0 chn not create!
07:39:43 [    vpss] VPSS_DisableChn@5456          [grp0]:vpss unexist!
07:39:43 [    vpss] VPSS_DisableChn@5456          [grp0]:vpss unexist!
07:39:43 [     sys] SysCheckBindChn@185           ModId:0(unknow mod) is not supported !
07:39:43 [     sdk] start_sdk@277                 sensor imx385
07:39:43 [     sdk] start_sdk@282                   input_mode MIPI, WDR NONE
07:39:43 [     sdk] start_sdk@289                   dev [1920x1080]@4x4 25fps, RGGB
07:39:43 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx385.so
07:39:44 [     sdk] dump_vb_configuration@2484    VB configuration:
07:39:44 [     sdk] dump_vb_configuration@2494      [0]: 3136320 x 9
07:39:44 [     sdk] dump_vb_configuration@2494      [1]: 4147200 x 1
07:39:44 [     sdk] init_sensor@2687              Sensor driver loaded
07:39:44 [     isp] VREG_DRV_Exit@102             The vreg of u32BaseAddr 0x10000 has not registerd!
07:39:44 [    puts]                               linear mode
07:39:44 [    puts]                               ===IMX385 1080P 30fps 12bit LINE Init OK!===
07:39:44 [    vpss] VPSS_DRV_MallocNrBuffer@1769  malloc vpss Ref&Mad buffer fail!
07:39:44 [    vpss] VPSS_CreateGrp@4258           [grp0]:drv ctx init fail!
07:39:44 [     sdk] init_vpss@3136                Cannot create VPSS grp 0
07:39:44 [     sdk] init_vpss@3136                ERR_VPSS_NOMEM: The memory fails to be allocated
07:39:44 [image_tu] start_image_params_tuning@63  Image tuning task started
07:39:44 [     sdk] start_sdk@825                 HiSilicon SDK started
07:39:44 [   httpd] new_http_server@342           HTTP server started on 0.0.0.0:80
07:39:44 [    rtsp] rtsp_init@30                  RTSP server started on port 554
07:39:44 [   netip] netip_start@2007              NETIP server started on port 34567
07:39:47 [     hal] take_yuv420sp_video_fram@1059 Cannot set chn mode for vpssGrp=0, vpssChn=0
07:39:47 [     hal] take_yuv420sp_video_fram@1059 ERR_VPSS_UNEXIST: No VPSS group is created
07:39:47 [image_co] image_control_compute_his@100 Cannot take yuv420sp VI frame
07:39:47 [image_co] image_control_compute_his@100 ERR_VPSS_UNEXIST: No VPSS group is created
07:39:47 [image_tu] tune_image_params@180         get_histogram errored
07:39:47 [    vpss] VPSS_SetChnMode@5696          [grp0]:vpss unexist!
07:39:50 [     hal] take_yuv420sp_video_fram@1059 Cannot set chn mode for vpssGrp=0, vpssChn=0
07:39:50 [     hal] take_yuv420sp_video_fram@1059 ERR_VPSS_UNEXIST: No VPSS group is created
07:39:50 [image_co] image_control_compute_his@100 Cannot take yuv420sp VI frame
07:39:50 [image_co] image_control_compute_his@100 ERR_VPSS_UNEXIST: No VPSS group is created
07:39:50 [image_tu] tune_image_params@180         get_histogram errored
07:39:50 [    vpss] VPSS_SetChnMode@5696          [grp0]:vpss unexist!
07:39:55 [     hal] take_yuv420sp_video_fram@1059 Cannot set chn mode for vpssGrp=0, vpssChn=0
07:39:55 [     hal] take_yuv420sp_video_fram@1059 ERR_VPSS_UNEXIST: No VPSS group is created
07:39:55 [image_co] image_control_compute_his@100 Cannot take yuv420sp VI frame
07:39:55 [image_co] image_control_compute_his@100 ERR_VPSS_UNEXIST: No VPSS group is created
07:39:55 [image_tu] tune_image_params@180         get_histogram errored
07:39:55 [    vpss] VPSS_SetChnMode@5696          [grp0]:vpss unexist!

```

dimerr, 2022-01-07:
can you post again current printenv and then a full bootlog?

gitgayhub, 2022-01-07:
```
hi3516cv300 System startup

U-Boot 2010.06 (Mar 21 2018 - 18:48:28)

Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xc2 0x20 0x18
Block:64KB Chip:16MB Name:"MX25L128XX"
SPI Nor total size: 16MB
In:    serial
Out:   serial
Err:   serial
Hit any key to stop autoboot:  0
16384 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.18.20-hi3516cv300
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1814195 Bytes = 1.7 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.18.20 (runner@fv-az177-150) (gcc version 7.5.0 (Buildroot -gccf718b) ) #1 Thu Jan 6 06:09:25 UTC 2022
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Hisilicon Hi3516CV300 DEMO Board
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: mem=32M console=ttyAMA0,115200 root=/dev/mtdblock3 rootfstype=squashfs init=/init rw mtdparts=hi_sfc:256K(boot),256K(env),2560K(kernel),5120K(rootfs),-(rootfs_data)
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Memory: 27392K/32768K available (3400K kernel code, 169K rwdata, 980K rodata, 152K init, 242K bss, 5376K reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xc2800000 - 0xff000000   ( 968 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc044f568   (4382 kB)
[    0.000000]       .init : 0xc0450000 - 0xc0476000   ( 152 kB)
[    0.000000]       .data : 0xc0476000 - 0xc04a0560   ( 170 kB)
[    0.000000]        .bss : 0xc04a0560 - 0xc04dcfb0   ( 243 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] VIC @c2800000: id 0x00641190, vendor 0x41
[    0.000033] sched_clock: 32 bits at 3000kHz, resolution 333ns, wraps every 1431655765682ns
[    0.000772] Console: colour dummy device 80x30
[    0.000829] Calibrating delay loop... 398.13 BogoMIPS (lpj=1990656)
[    0.050025] pid_max: default: 32768 minimum: 301
[    0.050185] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.050203] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.051396] Initializing cgroup subsys devices
[    0.051438] Initializing cgroup subsys freezer
[    0.051461] Initializing cgroup subsys blkio
[    0.051520] CPU: Testing write buffer coherency: ok
[    0.052218] Setting up static identity map for 0x8033e760 - 0x8033e7b8
[    0.054257] devtmpfs: initialized
[    0.055888] VFP support v0.3: not present
[    0.056700] pinctrl core: initialized pinctrl subsystem
[    0.064769] NET: Registered protocol family 16
[    0.066206] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.080808] Serial: AMBA PL011 UART driver
[... 227 lines cut ...]
cropEn=0
cropX=0
cropY=0
cropWidth=0
cropHeight=0
filesize=4E0000
fileaddr=82000000
gatewayip=192.168.1.1
netmask=255.255.255.0
ipaddr=192.168.2.28
serverip=192.168.2.88
soc=hi3516cv300
sensor=imx385
bootcmd=sf probe 0;sf read 0x82000000 0x80000 0x280000;bootm 0x82000000
osmem=32M
totalmem=128M
stdin=serial
stdout=serial
stderr=serial
verify=n
ver=U-Boot 2010.06 (Mar 21 2018 - 18:48:28)
bootargs=mem=32M console=ttyAMA0,115200 root=/dev/mtdblock3 rootfstype=squashfs init=/init rw mtdparts=hi_sfc:256K(boot),256K(env),2560K(kernel),5120K(rootfs),-(rootfs_data)
root@openipc-hi3516cv300:~#

```

dimerr, 2022-01-07:
setenv bootargs "mem=32M totalmem=128M console=ttyAMA0,115200 root=/dev/mtdblock3 rootfstype=squashfs init=/init rw mtdparts=hi_sfc:256K(boot),256K(env),2560K(kernel),5120K(rootfs),-(rootfs_data)"

gitgayhub, 2022-01-07:
Wow, it's working now, unbelievable, first good thing in 2022. Thank you again for the generous help

```
root@openipc-hi3516cv300:~# SENSOR=imx385 majestic
12:38:24 [    main] main@135                      Majestic Lite for HiSilicon, version master+3f13a01, built on 2022-01-04
12:38:24 [app_conf] load_config@342               Using /etc/majestic.yaml as main configuration
12:38:24 [     sdk] sdk_specific_config@3540      Using /etc/sensors/imx385_i2c_1080p.ini sensor configuration file
12:38:24 [watchdog] watchdog_start@78             Detected Hisilicon Watchdog
12:38:24 [watchdog] watchdog_start@98             Watchdog timeout set to 10 seconds
12:38:24 [     sdk] start_sdk@250                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
12:38:24 [     sdk] start_sdk@253                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
12:38:24 [    venc] VENC_VpssSend@1545            venc check create:venc 0 chn not create!
12:38:24 [    venc] VENC_VpssSend@1545            venc check create:venc 1 chn not create!
12:38:24 [    venc] VENC_VpssSend@1545            venc check create:venc 1 chn not create!
12:38:24 [     sdk] start_sdk@277                 sensor imx385
12:38:24 [     sdk] start_sdk@282                   input_mode MIPI, WDR NONE
12:38:24 [     sdk] start_sdk@289                   dev [1920x1080]@4x4 25fps, RGGB
12:38:24 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx385.so
12:38:24 [     sdk] dump_vb_configuration@2484    VB configuration:
12:38:24 [     sdk] dump_vb_configuration@2494      [0]: 3136320 x 9
12:38:24 [     sdk] dump_vb_configuration@2494      [1]: 4147200 x 1
12:38:24 [     sdk] init_sensor@2687              Sensor driver loaded
12:38:24 [    puts]                               linear mode
12:38:24 [    puts]                               ===IMX385 1080P 30fps 12bit LINE Init OK!===
12:38:24 [     sdk] log_venc_chn@1605             H.265 1920x1080 25fps 4096Kbit  25 GOP
12:38:24 [     sdk] create_vpss_chn@1358            new venc: 0   vpss_grp: 0,   vpss_chn: 0
12:38:24 [     sdk] init_chn@1640                 JPEG snapshot snapshot venc_chn 1  1920x1080
12:38:24 [image_tu] start_image_params_tuning@63  Image tuning task started
12:38:24 [     sdk] start_sdk@825                 HiSilicon SDK started
12:38:24 [   httpd] new_http_server@342           HTTP server started on 0.0.0.0:80
12:38:24 [    rtsp] rtsp_init@30                  RTSP server started on port 554
12:38:25 [   netip] netip_start@2007              NETIP server started on port 34567
12:38:28 [image_co] set_image_contrast@55         Autoadjust image param cscAttrContrField from 50 to 90
                                         ╭╮╭╮
                                         │╰╮│ ╭╮
                                        ╭╯││╰╮│╰╮╭─╮
                                        │ │╰──╯╰╰╯ │
                                        │ │     │  ╰╮
                                       ╭╯ │     │   │
                                      ╭╯  │     ╰╮  │
                                     ╭╯  ╭╯      │  ╰╮
                                     │  ╭╯       │   │
                                    ╭╯  │        ╰╮  ╰╮
                                    │   │         │   │
╮                                  ╭╯  ╭╯         ╰╮  ╰╮
│                                  │   │           │   ╰╮
│                                  │   │           │    │
│                                 ╭╯   │           │    │
│                                 │    │           ╰╮   ╰╮
│                                ╭╯   ╭╯            │    │
│                   ╭──╮  ╭───╮╭─╯╭╮  │             │    │
│               ╭───╯  ╰──╯ ╭─╰╯──╯╰──╯             │    │
│              ╭╯           │                       │    │
│       ╭──────╯           ╭╯                       │    ╰╮
│   ╭╭──╯             ╭────╯                        ╰╮    │
│   ╭╯─╮           ╭──╯                              ╰────╰────────────────╮
╰───╯  ╰───────────╯                                         ╰─────────────╰─────────
12:40:17 [  client] RtspClient_dispatch@458       RTSP OPTIONS rtsp://192.168.2.207:554/stream=0 CSeq=2 from 192.168.2.88:11176
12:40:17 [  client] RtspClient_dispatch@458       RTSP DESCRIBE rtsp://192.168.2.207:554/stream=0 CSeq=3 from 192.168.2.88:11176
12:40:17 [  client] describe_sdp_video@555        RTSP chn 0 sps_len 0
12:40:17 [  client] describe_sdp_video@558        RTSP chn 0 pps_len 0
12:40:17 [  client] describe_sdp_video@564        RTSP chn 0 profile_level_id 000000
12:40:17 [  client] RtspClient_dispatch@458       RTSP SETUP rtsp://192.168.2.207:554/stream=0/video CSeq=4 from 192.168.2.88:11176
12:40:17 [  client] RtspClient_setup@803          UDP RTSP SETUP done, stream_id=0 from 192.168.2.88:11176
12:40:17 [  client] RtspClient_dispatch@458       RTSP PLAY rtsp://192.168.2.207:554/stream=0 CSeq=5 from 192.168.2.88:11176
12:41:15 [  client] RtspClient_dispatch@458       RTSP OPTIONS rtsp://192.168.2.207:554/stream=0 CSeq=6 from 192.168.2.88:11176
12:42:13 [  client] RtspClient_dispatch@458       RTSP OPTIONS rtsp://192.168.2.207:554/stream=0 CSeq=7 from 192.168.2.88:11176
12:43:11 [  client] RtspClient_dispatch@458       RTSP OPTIONS rtsp://192.168.2.207:554/stream=0 CSeq=8 from 192.168.2.88:11176

```

gitgayhub, 2022-01-07:
If you plan to add support for the sensor SmartSens SC4210, I'm willing to do the testing job

dimerr, 2022-01-07:
If you've got the .so file for sensor (UNV often has them as separate files) feel free to open another issue for that sensor

closed 2022-01-07 by dimerr

gitgayhub, 2022-01-07:
> If you've got the .so file for sensor (UNV often has them as separate files) feel free to open another issue for that sensor

I found some problem with that. The tftp command inside that uboot doesn't have the feature to upload file to server, it can only download from server. And the shell environment is protected too, I can only run a few commands.  Thus I can't get a dump of the EMMC

But I do have the firmware for upgrade, does that help?

dimerr, 2022-01-08:
Please open new issue about this sensor and lets continue there.