OpenIPC/firmware #143: xm510: ssh login fails

schneemaier, 2022-01-07:
When i try to SSH into my xm510 camera i receive an error:

```
:~$ssh root@192.168.10.14
root@192.168.10.14's password: 
PTY allocation request failed on channel 0
shell request failed on channel 0
```

Running top on the camera shows that syslogd uses ~25% CPU time:
```
Mem: 14092K used, 1348K free, 112K shrd, 696K buff, 2240K cached
CPU:  67% usr  28% sys   0% nic   0% idle   0% io   0% irq   3% sirq
Load average: 1.84 1.04 0.43 2/44 865
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  486     1 root     S    15560 101%  71% /usr/bin/majestic -s
  333     1 root     S     1624  11%  23% /sbin/syslogd -n -C64
  490   486 root     S     3832  25%   4% isp
  361     1 root     S     1560  10%   1% /usr/sbin/rngd -r /dev/urandom
....
```

widgetii, 2022-01-07:
This is a well-known issue https://github.com/OpenIPC/firmware/wiki/install_xm510#known-issues

schneemaier, 2022-01-08:
Thanks

Regarding the ~25% CPU for syslog. I found that after boot majestic is spamming the logs with the following errors:
```
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Unknown error code FFFFFFFF
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Empty data has been returned for JPEG request
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Cannot release VENC stream for chn=2
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Unknown error code FFFFFFFF
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Cannot get VENC stream for chn=2
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Unknown error code FFFFFFFF
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Cannot release VENC stream for chn=2
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Unknown error code FFFFFFFF
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Cannot get VENC stream for chn=2
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Unknown error code FFFFFFFF
Jan  8 00:23:15 sch-openipc-xm510 user.err majestic[486]: Cannot release VENC stream for chn=2
```

Eventually it kills majestic at after 20-180 minutes. I also found the if is open a video preview stream (http://192.168.10.14:85/cgi-bin/preview-video.cgi) and start streaming the error reporting stops and the CPU usage drops both for majestic and syslogd. Majestic still dies after some time though and the only way to restart is either reboot the camera or kill the isp process and then run "S95xiongmai start"

schneemaier, 2022-01-08:
Maybe related thing: when the camera starts the stream delay is around 3 sec, after ~1 hours it is 20 sec and it is constantly increasing until majestic stops functioning.

widgetii, 2022-01-08:
[uImage.xm510.zip](https://github.com/OpenIPC/firmware/files/7832587/uImage.xm510.zip)
Try to burn this kernel image to mitigate PTY allocation request issue.
It's the same kernel but built with original XM toolchain

For out of mem issue I suggest disabling HLS setting in majestic.yaml and check if it helps

schneemaier, 2022-01-08:
Thanks, will try the kernel.

I disabled HLS and it seems that it help the stabilty as the RTSP stream seems stable. However after booting the camera i still see high SPU usage for majestic (~60%) and syslog (30%) which only goes away when i load http://192.168.10.14:85/cgi-bin/preview-video.cgi, which generates a few error mesages in the logs, but after that the CPU usage drops below 10 percent in idle condition
```
Jan  8 15:38:29 sch-openipc-xm510 user.err majestic[1677]: Stuck in progress
Jan  8 15:38:29 sch-openipc-xm510 user.err majestic[1677]: Empty data has been returned for JPEG request
Jan  8 15:38:29 sch-openipc-xm510 user.err majestic[1677]: Stuck in progress
Jan  8 15:38:29 sch-openipc-xm510 user.err majestic[1677]: Empty data has been returned for JPEG request
Jan  8 15:39:02 sch-openipc-xm510 user.err majestic[1677]: Stuck in progress
Jan  8 15:39:02 sch-openipc-xm510 user.err majestic[1677]: Empty data has been returned for JPEG request
Jan  8 15:39:02 sch-openipc-xm510 user.err majestic[1677]: Stuck in progress
Jan  8 15:39:02 sch-openipc-xm510 user.err majestic[1677]: Empty data has been returned for JPEG request
```

The jpg snapshot and the mjpg stream do not work though they are enabled in the majestic config.

widgetii, 2022-01-08:
Actually, XM510 is an ancient SoC and there is no way to get both JPEG and MJPEG streams from it. For XM platform we're looking for sponsors to continue development because it will take a vast amount of resources to make it good

schneemaier, 2022-01-10:
Do you mean that the xm510 is not capable of JPEG? I did not know that. 

I played around with the majestic config and the huge CPU load is going away if i disable the JPEG functionality. After this the system becames stable and can run for several hours. I tried night mode and other settings, but it does not work. I guess because GPIOs are not enabléed for the xm510?

I would be happy to help with the development on xm510 platform, but i don't have the knowledge to be really useful. I can help with testing and figuring out problems. Example: i can play around and figure out the GPIO pins for the xm510, but i have no idea how to enable GPIO support for the kernel.

widgetii, 2022-01-10:
What about the main topic of the issue? Have you resolved ssh login issue with the updated kernel?

schneemaier, 2022-01-10:
With the kernel you [shared](https://github.com/OpenIPC/firmware/issues/143#issuecomment-1007887408) it does work. I can open a ssh session to the camera

schneemaier, 2022-01-15:
Resolving

widgetii, 2022-01-20:
I'm going to reopen the issue because it needs to be fixed in the mainstream, otherwise, it produces different issues such mentioned in #157

closed 2022-02-11 by widgetii

schneemaier, 2022-02-11:
Thank you for the fix!  It works great.

commit f4a327e9 (2022-02-11): Fix openpty behaviour on recent toolchains, fixes #143 (#172)