OpenIPC/firmware #149: Driver for sensor SC4210

gitgayhub, 2022-01-09:
Hi, I have an Uniview camera with Hi3516DV300, CMOS sensor is [SmartSens SC4210](https://smartsens.oss-cn-beijing.aliyuncs.com/web/products/SC4210_V4.0.pdf).  It has 2GB memory and 8GB onboard EMMC storage. As told in another issue, you can add support for this sensor if I have the .so files for the sensor.

Unfortunately, the shell access is limited to only a few commands, the uboot tftp command is limited to download only, so I can't get a dump of the EMMC chip. But I do have a firmware image for upgrade.  The entire firmware is more than 500MB large, I can't upload it here on github, so I upload it somewhere else: [https://ufile.io/se1e8xam](https://ufile.io/se1e8xam) The [uimage.bin](https://github.com/OpenIPC/firmware/files/7834646/uimage.zip) is 12MB large

```
uboot # printenv
arch=arm
baudrate=115200
board=hi3516dv300
board_name=hi3516dv300
bootargs=mem=512M console=ttyAMA0,115200 blkdevparts=mmcblk2:1M(boot),2M(bootlogo),16M(kernel),16M(kernel_bak),16M(config),16M(config_bak),1M(cliinfo),1M(cliinfo_bak),1M(mtd_runtime),32M(calibration),512K(update),1224M(program),1224M(cache),4208M(data),-(other)
bootcmd=mmc read 0 82000000 1800 8000;bootm 0x82000000
bootdelay=2
cpu=armv7
ethact=eth0
ethaddr=42:c1:84:f1:25:6e
ipaddr=192.168.2.22
netmask=255.255.255.0
serverip=192.168.2.88
soc=hi3516dv300
stderr=serial
stdin=serial
stdout=serial
vendor=hisilicon
verify=n
```

```
uboot # help
?       - alias for 'help'
base    - print or set address offset
bdinfo  - print Board Info structure
boot    - boot default, i.e., run 'bootcmd'
bootd   - boot default, i.e., run 'bootcmd'
bootelf - Boot from an ELF image in memory
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
bootvx  - Boot vxWorks from an ELF image
cipher_test- CIPHER Encrypt And Decrypt Test
cmp     - memory compare
config  - config  --- config mac address,ddr test flag

coninfo - print console devices and information
cp      - memory copy
crc32   - checksum calculation
dcache  - enable or disable data cache
ddr     - ddr training function
decjpg  - jpgd   - decode jpeg picture.
decjpg [format]
dhcp    - boot image via network using DHCP/TFTP protocol
dm      - Driver model low level access
echo    - echo args to console
editenv - edit environment variable
env     - environment handling commands
erase   - erase FLASH memory
exit    - exit script
false   - do nothing, unsuccessfully
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatsize - determine a file's size
flinfo  - print FLASH memory information
getinfo - print hardware information
go      - start application at address 'addr'
go_cpu1 - Perform power on and unreset  CPU1_A7
gzwrite - unzip and write memory to block device
hash    - Calcluate hash
hash_test- hash_test [x]:[0] SHA1; [1] SHA256; [2] HMAC-SHA1; [3] HMAC-SHA256;
help    - print command description/usage
icache  - enable or disable instruction cache
iminfo  - print header information for application image
imxtract- extract a part of a multi-image
itest   - return true/false on integer compare
klad_test- KLAD Test
loadb   - load binary file over serial line (kermit mode)
loads   - load S-Record file over serial line
loadx   - load binary file over serial line (xmodem mode)
loady   - load binary file over serial line (ymodem mode)
loop    - infinite loop on address range
manuinfo- download,erase,display manuinfo

md      - memory display
mii     - MII utility commands
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcinfo - display MMC info
mw      - memory write (fill)
nfs     - boot image via network using NFS protocol
nm      - memory modify (constant address)
part    - disk partition related commands
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
protect - enable or disable FLASH write protection
pxe     - commands to get and boot from pxe files
reset   - Perform RESET of the CPU
rng_test- RNG Test
rsa_enc_test- RSA PKCS1# V1_5 encrypt/decrypt Test
rsa_sign_test- RSA PKCS1# V1_5 Sign Test
run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
setenv  - set environment variables
setexpr - set environment variable as the result of eval expression
setvobg - setvobg   - set vo backgroud color.
        - setvobg [dev color]
showvar - print local hushshell variables
sleep   - delay execution for some time
source  - run script from memory
startgx - startgx   - open graphics layer.
        - startgx [layer addr stride x y w h]

startvl - startvl   - open video layer.
        - startvl [layer addr stride x y w h]

startvo - startvo   - open vo device with a certain output interface.
        - startvo [dev intftype sync]
stopgx  - stopgx   - close graphics layer.
        - stopgx [layer]
stopvl  - stopvl   - close video layer.
        - stopvl [layer]
stopvo  - stopvo   - close interface of vo device.
        - stopvo [dev]
sysboot - command to get and boot from syslinux files
test    - minimal test like /bin/sh
tftpboot- boot image via network using TFTP protocol
true    - do nothing, successfully
unzip   - unzip a memory region
update  - update image via network using TFTP protocol and write into emmc/flash.
usb     - USB sub-system
usbboot - boot from USB device
version - print monitor, compiler and linker version

```

Please take a look and see if you can extract the .so files from it

Thank you very much

dimerr, 2022-01-09:
Unfortunately this firmware is encrypted, so emmc dump is needed. I see uboot has support for mmc, do you have mmc slot on board?

gitgayhub, 2022-01-09:
I noticed that too before I used the mmc write command to flash the uimage.bin file to EMMC chip.  Now I can’t even boot to kernel, uboot says can’t find kernel image :( tftp 0x82000000 uimage.bin then bootm 0x82000000 gave the same result

Seems you can’t update the kernel partition in uboot, you can only do that in the web panel, there should be a decryption process before writing kernel files to EMMC chip

Yes there’s a micro SD card slot, I will try to get a dump of the other partitions which I haven’t overwritten tomorrow, and contact Uniview support to see if I can get a replacement

gitgayhub, 2022-01-10:
I tried the micro-SD card slot, but the mmc command inside uboot can only find the onboard emmc chip. I will send the device back to UNV support service, they will likely replace the PCB board. I will continue to try to get the partition dumps after I got it back

```
uboot # mmcinfo
Device: himci
Manufacturer ID: 15
OEM: 100
Name: 8GTF4
Tran Speed: 200000000
Rd Block Len: 512
MMC version 5.1
High Capacity: Yes
Bus Width: 4-bit
Erase Group Size: 512 KiB
HC WP Group Size: 8 MiB
User Capacity: 7.3 GiB ENH WRREL
User Enhanced Start: 0 Bytes
User Enhanced Size: 8 MiB
Boot Capacity: 4 MiB ENH
RPMB Capacity: 512 KiB ENH
```

```
uboot # mmc dev 0
scan edges:2 p2f:6 f2p:1
mix set temp-phase 3
scan elemnts: startp:120 endp:111
Tuning SampleClock. mix set phase:[03/07] ele:[10d/15]
** First descriptor is NOT a primary desc on 0:1 **
switch to partitions #0, OK
mmc0(part 0) is current device
uboot # mmc dev 1
MMC Device 1 not found
no mmc device at slot 1
uboot # mmc dev 2
MMC Device 2 not found
no mmc device at slot 2

```

dimerr, 2022-01-15:
Refer to this message from our telegram channel

`also found I can use md.b to display memory in hex.. I also found a python program to convert this back into binary. Seems to work...`

this will be extremely long but possible...

gitgayhub, 2022-01-15:
Unfortunately the nand chip is already flashed with encrypted image, and UNV refused my request to replace a PCB board, now I'm looking for another same model to get a dump, this will take some time

The method you mentioned could be very useful when there's no tftp upload function or no network access at all, could you please give the name of that python program, thanks

gitgayhub, 2022-01-15:
I made some search and found that the xxd command in debian can directly convert the copy-pasted hex texts to binary file, but it does take a long time

`xxd -r -seek -0x82000000 input.txt output.bin`

0x82000000 is the memory offset

dimerr, 2022-01-15:
https://github.com/smorin/hextool/blob/master/hexdump.py prolly this, but xxd will work too, but it will generate a 2gig file, which you will have to cut manually later

gitgayhub, 2022-01-15:
>  but it will generate a 2gig file, which you will have to cut manually later

Use the -seek parameter to ignore the none existing contents before

gitgayhub, 2022-01-15:
Just curious, Is it possible to create an uimage together with rootfs inside it, so we can load it into memory by tftp and boot into it with the bootm command, then mount the partitions to get a dump

dimerr, 2022-01-15:
It's possible, but much easier is to use nfsroot

dimerr, 2022-01-16:
I've just noticed that you've asked about cv500 family, have you checked available sensors? I see yours is there....

https://github.com/OpenIPC/firmware/blob/master/general/package/hisilicon-osdrv-hi3516cv500/files/sensor/

gitgayhub, 2022-01-16:
> I've just noticed that you've asked about cv500 family, have you checked available sensors? I see yours is there....

Thanks but I didn't asked about cv500, I don't have one either. Currently I have 3516av100+imx185, 3516av200+imx385, 3516dv100+imx385 and 3516dv300+sc4210

dimerr, 2022-01-16:
3516dv300 is 3516cv500

gitgayhub, 2022-01-16:
thanks, so do I need to install the cv500 firmware instead of 3516dv300?

dimerr, 2022-01-16:
install firmware for your soc, but they share same family, so there will be same files in all of them

closed 2022-01-20 by dimerr

gitgayhub, 2022-02-25:
Hi, I have manged to get a dump of the emmc chip, please let me know if you still need them, attached are the libsns files
[libsns.zip](https://github.com/OpenIPC/firmware/files/8138743/libsns.zip)
 
```
root@root:/nfs/HIC5641# ./ipctool
---
chip:
  vendor: HiSilicon
  model: 3516DV300
ethernet:
  mac: "6c:xxxxxxxx"
  u-mdio-phyaddr: 1
  phy-id: 0xffffffff
  d-mdio-phyaddr: 0
ram:
  total: 2048M
  media: 1535M
firmware:
  kernel: "4.9.37 (SMP Thu Oct 29 16:38:14 CST 2020)"
  toolchain: gcc version 6.3.0 (HC&C V100R002C00B017_20180728)
  sdk: "Hi3516CV500_MPP_V2.0.1.0 B090 Release (Jan 14 2019, 20:23:45)"
  main-app: mwareserver
sensors:
- vendor: SmartSens
  model: SC4210
  control:
    bus: 0
    type: i2c
    addr: 0x60
```