OpenIPC/firmware #151: Не работает видео на hi3516ev300

esuslik, 2022-01-10:
Установил прошивку на камеру чере веб
Модель камеры 000529E9

При включеном watchdog иде постоянная перезагрузка каждую минуту. После отключения бюольше не перезагружается.

Не работает видео preview 

Также обратил внимание на сообщение во вкладке firmware:
Bundled version of Majestic is used.
Not enough space to update Majestic.
Required KB, available 8531 KB.

Как заставит работать видео и звук?

dimerr, 2022-01-10:
Какой сенсор установлен на этой плате?

Покажите вывод ipctool

Если о сенсоре ничего не будет, попробуйте `fw_setenv sensor sc4236 `

esuslik, 2022-01-10:
а где запускать ipctool?

esuslik, 2022-01-10:
через веб консоль /bin/sh: eval: line 1: ipctool: not found

ошибку что нет места удалось решить, путем обновления web gui до development и потом обновил majestic

esuslik, 2022-01-10:
Hardware

SoC           hi3516ev300
SoC Family    hi3516ev300
Sensor        imx335_i2c
Flash         16 MB
SoC temp.     86.75°C

Firmware

Version       2020.02.12-lite
Build         master+29dde73, 2022-01-10

esuslik, 2022-01-10:
[root@openipc-hi3516ev300 /tmp]# SENSOR=imx335 majestic

22:05:15 INFO  [main] main@135: Majestic Lite for HiSilicon, version master+f9fbbda, built on 2022-01-09
22:05:15 INFO  [app_config] load_config@342: Using /etc/majestic.yaml as main configuration
22:05:15 DEBUG [sdk] sdk_specific_config@3533: SENSOR=imx335
22:05:15 DEBUG [sdk] find_sensor_config@3490: matched sensor config: imx335_i2c_4M.ini
22:05:15 INFO  [sdk] find_sensor_config@3507: Using /etc/sensors/imx335_i2c_4M.ini as sensor configuration
22:05:15 DEBUG [app_config] parse_app_config@683: app_config.osd_template %a %e %B %Y %H:%M:%S %Z
22:05:15 DEBUG [sdk] start_sdk@250: App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
22:05:15 DEBUG [sdk] start_sdk@253: Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
22:05:15 DEBUG [sdk] start_sdk@277: sensor stSnsImx335Obj
22:05:15 DEBUG [sdk] start_sdk@282:   input_mode MIPI, WDR NONE
22:05:15 DEBUG [sdk] start_sdk@289:   dev [2592x1520]@200x20 25fps, RGGB
22:05:15 DEBUG [sensor] try_to_load@19: trying to load /usr/lib/sensors/libsns_imx335.so
22:05:16 DEBUG [sdk] dump_vb_configuration@2485: VB configuration:
22:05:16 DEBUG [sdk] dump_vb_configuration@2495:   [0]: 5909760 x 6
22:05:16 DEBUG [sdk] dump_vb_configuration@2495:   [2]: 608256 x 1
22:05:16 DEBUG [sdk] init_sensor@2688: Sensor driver loaded
22:05:16 INFO  [puts] @0: linear mode
22:05:16 INFO  [puts] @0: binit false IMX335 i2c init
22:05:16 INFO  [puts] @0: -------Sony IMX335_init_5M_2592x1944_12bit_linear30 Initial OK!-------
22:05:16 DEBUG [sdk] log_venc_chn@1605: H.264 2592x1520 25fps 4096Kbit  25 GOP
22:05:16 DEBUG [sdk] create_vpss_chn@1358:   new venc: 0   vpss_grp: 0,   vpss_chn: 0
22:05:16 DEBUG [sdk] log_venc_chn@1605: H.265 704x576 15fps 1024Kbit  15 GOP
22:05:16 DEBUG [sdk] create_vpss_chn@1358:   new venc: 1   vpss_grp: 0,   vpss_chn: 1
22:05:16 DEBUG [sdk] init_chn@1640: JPEG snapshot snapshot venc_chn 2  2592x1520
22:05:16 DEBUG [osd] init_osd@92: OSD initialized
22:05:16 INFO  [image_tuning] start_image_params_tuning@63: Image tuning task started
22:05:16 DEBUG [sdk] start_sdk@825: HiSilicon SDK started
22:05:16 INFO  [httpd] new_http_server@346: HTTP server started on :::80
22:05:16 INFO  [rtsp] rtsp_init@30: RTSP server started on port 554
22:05:16 INFO  [netip] netip_start@2007: NETIP server started on port 34567
Killed

esuslik, 2022-01-10:
[root@openipc-hi3516ev300 /tmp]# SENSOR=imx335 majestic

22:06:46 INFO  [main] main@135: Majestic Lite for HiSilicon, version master+f9fbbda, built on 2022-01-09
22:06:46 INFO  [app_config] load_config@342: Using /etc/majestic.yaml as main configuration
22:06:46 INFO  [sdk] sdk_specific_config@3541: Using /etc/sensors/5M_imx335.ini sensor configuration file
22:06:46 DEBUG [app_config] parse_app_config@683: app_config.osd_template %a %e %B %Y %H:%M:%S %Z
22:06:46 DEBUG [sdk] start_sdk@250: App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
22:06:46 DEBUG [sdk] start_sdk@253: Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
22:06:46 DEBUG [sdk] start_sdk@277: sensor stSnsImx335Obj
22:06:46 DEBUG [sdk] start_sdk@282:   input_mode MIPI, WDR NONE
22:06:46 DEBUG [sdk] start_sdk@289:   dev [2592x1944]@200x20 20fps, RGGB
22:06:46 DEBUG [sensor] try_to_load@19: trying to load /usr/lib/sensors/libsns_imx335.so
22:06:46 DEBUG [sdk] dump_vb_configuration@2485: VB configuration:
22:06:46 DEBUG [sdk] dump_vb_configuration@2495:   [0]: 7558272 x 6
22:06:46 DEBUG [sdk] dump_vb_configuration@2495:   [2]: 608256 x 1
22:06:46 DEBUG [sdk] init_sensor@2688: Sensor driver loaded
22:06:46 INFO  [puts] @0: linear mode
22:06:46 INFO  [puts] @0: binit false IMX335 i2c init
22:06:46 INFO  [puts] @0: -------Sony IMX335_init_5M_2592x1944_12bit_linear30 Initial OK!-------
22:06:46 DEBUG [sdk] log_venc_chn@1605: H.264 2592x1944 15fps 4096Kbit  15 GOP
22:06:46 DEBUG [sdk] create_vpss_chn@1358:   new venc: 0   vpss_grp: 0,   vpss_chn: 0
22:06:46 DEBUG [sdk] log_venc_chn@1605: H.265 704x576 15fps 1024Kbit  15 GOP
22:06:46 DEBUG [sdk] create_vpss_chn@1358:   new venc: 1   vpss_grp: 0,   vpss_chn: 1
22:06:46 DEBUG [sdk] init_chn@1640: JPEG snapshot snapshot venc_chn 2  2592x1944
22:06:46 DEBUG [osd] init_osd@92: OSD initialized
22:06:46 INFO  [image_tuning] start_image_params_tuning@63: Image tuning task started
22:06:46 DEBUG [sdk] start_sdk@825: HiSilicon SDK started
22:06:46 INFO  [httpd] new_http_server@346: HTTP server started on :::80
22:06:46 INFO  [rtsp] rtsp_init@30: RTSP server started on port 554
22:06:46 INFO  [netip] netip_start@2007: NETIP server started on port 34567
22:06:50 DEBUG [log] printf@261: Error buf: BUF_MALLOC_ERROR     put(...)    /context/src/file_format/mp4/bitbuf.c:70
22:06:50 DEBUG [log] printf@261: Error buf: BUF_MALLOC_ERROR     write_mdat(...)    /context/src/file_format/mp4/moof.c:44
22:06:50 DEBUG [log] printf@261: Error buf: BUF_MALLOC_ERROR     flush_video(...)    /context/src/file_format/mp4/mp4.c:318
Killed

dimerr, 2022-01-10:
set_allocator cma

esuslik, 2022-01-10:
SENSOR=imx335 majestic

22:11:14 INFO  [main] main@135: Majestic Lite for HiSilicon, version master+f9fbbda, built on 2022-01-09
22:11:14 INFO  [app_config] load_config@342: Using /etc/majestic.yaml as main configuration
22:11:14 INFO  [sdk] sdk_specific_config@3541: Using /etc/sensors/5M_imx335.ini sensor configuration file
22:11:14 DEBUG [app_config] parse_app_config@683: app_config.osd_template %a %e %B %Y %H:%M:%S %Z
22:11:14 DEBUG [sdk] start_sdk@250: App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
22:11:14 DEBUG [sdk] start_sdk@253: Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
22:11:14 DEBUG [sdk] start_sdk@277: sensor stSnsImx335Obj
22:11:14 DEBUG [sdk] start_sdk@282:   input_mode MIPI, WDR NONE
22:11:14 DEBUG [sdk] start_sdk@289:   dev [2592x1944]@200x20 20fps, RGGB
22:11:14 DEBUG [sensor] try_to_load@19: trying to load /usr/lib/sensors/libsns_imx335.so
22:11:14 ERROR [sdk] init_mem@2585: Cannot set VB conf
22:11:14 ERROR [sdk] init_mem@2585: ERR_VB_BUSY: The system is busy
22:11:14 ERROR [sdk] start_sdk@315: Cannot init mem
22:11:14 INFO  [main] main@227: Shutdown main thread

esuslik, 2022-01-10:
менял в настройках Path to sensor configuration file
и вот разные выводы дает , сейчас при любых настройках выдает последний лог

dimerr, 2022-01-10:
fw_printenv

esuslik, 2022-01-10:
в логе вот еще пишет что типа памяти не хватает

Jan 10 17:19:47 openipc-hi3516ev300 kern.emerg kernel: Node 0 active_anon:14384kB inactive_anon:108kB active_file:104kB inactive_file:116kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:136kB dirty:0kB writeback:0kB shmem:112kB writeback_tmp:0kB unstable:0kB pages_sca
Jan 10 17:19:47 openipc-hi3516ev300 kern.emerg kernel: Normal free:624kB min:656kB low:820kB high:984kB active_anon:14384kB inactive_anon:108kB active_file:104kB inactive_file:116kB unevictable:0kB writepending:0kB present:32768kB managed:27192kB mlocked:0kB slab_reclaimable:368kB s
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: lowmem_reserve[]: 0 0
Jan 10 17:19:47 openipc-hi3516ev300 kern.emerg kernel: Normal: 36*4kB (UME) 22*8kB (UME) 3*16kB (UME) 6*32kB (U) 1*64kB (M) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 624kB
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: 85 total pagecache pages
Jan 10 17:19:47 openipc-hi3516ev300 kern.emerg kernel: 8192 pages RAM
Jan 10 17:19:47 openipc-hi3516ev300 kern.emerg kernel: 0 pages HighMem/MovableOnly
Jan 10 17:19:47 openipc-hi3516ev300 kern.emerg kernel: 1394 pages reserved
Jan 10 17:19:47 openipc-hi3516ev300 kern.emerg kernel: 0 pages cma reserved
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  586]     0   586      313       24       3       0        0             0 syslogd
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  590]     0   590      297        8       4       0        0             0 klogd
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  625]     0   625      297       24       3       0        0             0 rngd
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  659]     0   659      297        7       3       0        0             0 udhcpc
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  664]     0   664      297        9       4       0        0             0 ntpd
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  671]     0   671      196        9       4       0        0             0 dropbear
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  673]     0   673      297       24       4       0        0             0 httpd
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  685]     0   685      297        9       4       0        0             0 crond
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [  823]     0   823      297        8       4       0        0             0 getty
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [ 1219]     0  1219      300       10       3       0        0             0 httpd
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [ 1220]     0  1220      171       10       4       0        0             0 ajaxcmd.cgi
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [ 1221]     0  1221      298       10       4       0        0             0 sh
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [ 1224]     0  1224      297        8       3       0        0             0 sh
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [ 1225]     0  1225     9952     3487      25       0        0             0 majestic
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: [ 1230]     0  1230      297        8       3       0        0             0 httpd
Jan 10 17:19:47 openipc-hi3516ev300 kern.err kernel: Out of memory: Kill process 1225 (majestic) score 501 or sacrifice child
Jan 10 17:19:47 openipc-hi3516ev300 kern.err kernel: Killed process 1225 (majestic) total-vm:39808kB, anon-rss:13936kB, file-rss:12kB, shmem-rss:0kB
Jan 10 17:19:47 openipc-hi3516ev300 kern.info kernel: oom_reaper: reaped process 1225 (majestic), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

dimerr, 2022-01-10:
Мне кажется, вы не выполнили set_allocator cma

esuslik, 2022-01-10:
да пропустил сообщение, нужно после этой команды перезапускать камеру?

esuslik, 2022-01-10:
дольше пыталось запуститься, но также было killed
an 10 17:28:39 openipc-hi3516ev300 kern.info kernel: [ 1406]     0  1406      300       12       4       0        0             0 httpd
Jan 10 17:28:39 openipc-hi3516ev300 kern.err kernel: Out of memory: Kill process 1401 (majestic) score 486 or sacrifice child
Jan 10 17:28:39 openipc-hi3516ev300 kern.err kernel: Killed process 1401 (majestic) total-vm:39404kB, anon-rss:13524kB, file-rss:12kB, shmem-rss:0kB
Jan 10 17:28:39 openipc-hi3516ev300 kern.info kernel: oom_reaper: reaped process 1401 (majestic), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

esuslik, 2022-01-10:
после перезапуска камеры, появилось видео, видимо заработало, тестирую дальше

esuslik, 2022-01-10:
SoC temp.     104.12°C
Это нормальная температура? rtsp поток не могу получить на разрешении 1920х1080 через vlc
на stream1 показывает нормально

closed 2022-01-15 by esuslik