OpenIPC/firmware #153: Driver for IMX385 on Hi3516DV100

gitgayhub, 2022-01-13:
Hi, I've got [another camera](https://smtsec.en.alibaba.com/product/60842671862-803005151/H_265_2_0mp_Super_Starlight_IP_Camera_Module_Sony_IMX385_Hi3516D_CCTV_Smart_Security_IPC_PCB_IP_SMTSEC_Board_Camera_SIP_E385D.html) with the sensor imx385, but the soc is hi3516dv100 this time. I installed the newest OpenIPC firmware for hi3516dv100, but seems the imx385 driver and config file wasn't added for this soc

I've got a dump of the entire flash chip: [3516dv100_imx385_dump.zip](https://github.com/OpenIPC/firmware/files/7865217/3516dv100_imx385_dump.zip)

and the extracted rootfs: [3516dv100_imx385_rootfs.zip](https://github.com/OpenIPC/firmware/files/7865221/3516dv100_imx385_rootfs.zip)

```
hisilicon # print
bootdelay=1
baudrate=115200
ethaddr=00:00:23:34:45:66
bootfile="uImage"
bootcmd=sf probe 0;sf read 0x82000000 0x80000 0x280000;bootm 0x82000000
configFlag=ok
cpuFlag=ok
cpuType=3516a
sensorFlag=no
sensorType=53
filesize=39840
fileaddr=82000000
netmask=255.255.255.0
bootargs=mem=108M console=ttyAMA0,115200 root=/dev/mtdblock1 rootfstype=jffs2 mtdparts=hi_sfc:3M(boot),13M(rootfs)
stdin=serial
stdout=serial
stderr=serial
verify=n
ver=U-Boot 2010.06 (Sep 22 2017 - 09:17:14)
mdio_intf=rmii
hisicpu=hi3516d
ipaddr=192.168.2.22
serverip=192.168.2.88
```
I modified bootargs and boocmd to suite OpenIPC, please let me know if there's anything wrong
```
mem=108M totalmem=128M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock2 rootfstype=squashfs init=/init mtdparts=hi_sfc:3072k(boot),2048k(kernel),5120k(rootfs),-(rootfs_data)
bootcmd=sf probe 0;sf read 0x82000000 0x300000 0x300000;bootm 0x82000000
```

```
/ # cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00300000 00010000 "boot"
mtd1: 00d00000 00010000 "rootfs"

```

```
/usr # ./ipctool
---
chip:
  vendor: HiSilicon
  model: 3516DV100
ethernet:
  mac: "e6:98:25:xxxxxxx"
  u-mdio-phyaddr: 86
  phy-id: 0xa5a5a5a5
  d-mdio-phyaddr: a5a5
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x300000
        sha1: b6d9797f
        contains:
          - name: uboot-env
            offset: 0x40000
      - name: rootfs
        size: 0xd00000
        sha1: a11e3d0f
    size: 16M
    addr-mode: 3-byte
ram:
  total: 128M
firmware:
  u-boot: "2010.06 (Sep 22 2017 - 09:17:14)"
  kernel: "3.4.35 (Wed Aug 16 16:10:11 CST 2017)"
  toolchain: gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300)
  libc: uClibc 0.9.33.2
```

gitgayhub, 2022-01-14:
I found the firmware for this camera:
[Hi3516A-D(8.8.3.65).zip](https://github.com/OpenIPC/firmware/files/7867826/Hi3516A-D.8.8.3.65.zip)

dimerr, 2022-01-14:
https://github.com/OpenIPC/firmware/tree/7cfe3d7057e21da6f24872f3d4623d1f740afd0e/general/package/hisilicon-osdrv-hi3516av100/files/sensor try

gitgayhub, 2022-01-15:
Haven’t got time to test it yet, but it seems that the file you sent is for 3516av100, while my camera is 3516dv100, will it work?

widgetii, 2022-01-15:
They are from the same family AV100/DV100

gitgayhub, 2022-01-15:
Thanks, I tried the sensor driver and config file, but there's still some error

```
root@openipc-hi3516dv100:~# load_hisilicon -i
mmz_start: 0x82000000, mmz_size: 96M
hisilicon: Get data from environment and set SENSOR as imx385
insmod: can't insert 'mmz.ko': File exists
******* Error: There's something wrong, please check! *****
```
```
root@openipc-hi3516dv100:~# SENSOR=imx385 majestic
13:39:01 [    main] main@135                      Majestic Lite for HiSilicon, version master+c2f29d8, built on 2022-01-12
13:39:01 [app_conf] load_config@342               Using /etc/majestic.yaml as main configuration
13:39:01 [     sdk] sdk_specific_config@3536      SENSOR=imx385
13:39:01 [     sdk] find_sensor_config@3493       matched sensor config: imx385_i2c_1080p.ini
13:39:01 [     sdk] find_sensor_config@3510       Using /etc/sensors/imx385_i2c_1080p.ini as sensor configuration
13:39:01 [watchdog] watchdog_start@74             Watchdog device (/dev/watchdog) not found
13:39:01 [     sdk] start_sdk@250                 App was built with MPP version: Hi3516A_MPP_V1.0.8.0.B070 Release
13:39:01 [     sdk] start_sdk@253                 Current MPP version: HI_VERSION=Hi3516A_MPP_V1.0.8.0 B070 Release
13:39:01 [     sdk] start_sdk@277                 sensor imx385
13:39:01 [     sdk] start_sdk@282                   input_mode MIPI, WDR NONE
13:39:01 [     sdk] start_sdk@289                   dev [1920x1080]@4x4 25fps, RGGB
13:39:01 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx385.so
13:39:01 [     sdk] dump_vb_configuration@2488    VB configuration:
13:39:01 [     sdk] dump_vb_configuration@2498      [0]: 3110400 x 5
13:39:01 [     sdk] dump_vb_configuration@2498      [1]: 4147200 x 1
13:39:01 [     sdk] HISDK_COMM_VI_SetMipiAttr@844 warning: open hi_mipi dev failed
13:39:01 [     sdk] init_sensor@2628              Cannot set VI Mipi attr
13:39:01 [     sdk] init_sensor@2628              Unknown error code FFFFFFFF
13:39:01 [     sdk] start_sdk@320                 Cannot init sensor
13:39:01 [    main] main@227                      Shutdown main thread

```

```
root@openipc-hi3516dv100:~# fw_printenv
bootdelay=1
baudrate=115200
ethaddr=00:00:23:34:45:66
bootfile="uImage"
configFlag=ok
cpuFlag=ok
cpuType=3516a
sensorFlag=no
sensorType=53
filesize=4A2000
fileaddr=82000000
netmask=255.255.255.0
ipaddr=192.168.2.22
serverip=192.168.2.88
bootcmd=sf probe 0;sf read 0x82000000 0x300000 0x300000;bootm 0x82000000
totalmem=128M
sensor=imx385
soc=3516dv100
osmem=32M
stdin=serial
stdout=serial
stderr=serial
verify=n
ver=U-Boot 2010.06 (Sep 22 2017 - 09:17:14)
mdio_intf=rmii
hisicpu=hi3516d
bootargs=mem=32M totalmem=128M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock2 rootfstype=squashfs init=/init mtdparts=hi_sfc:3072k(boot),2048k(kernel),5120k(rootfs),-(rootfs_data)

```

boot log:

hisilicon # reset
resetting ...
[... 179 lines cut ...]
udhcpc: sending discover
udhcpc: sending discover
udhcpc: sending discover
udhcpc: no lease, forking to background
OK
Starting ntpd: OK
Starting dropbear sshd: OK
Starting httpd: OK
Starting mini-snmpd: DISABLED, OK
Starting telnetd: DISABLED, OK
Starting crond: OK
hisilicon: Check MAC for Xiongmai devices
hisilicon: The eth0 interface has a lousy MAC, let's try to change it..
hisilicon: It is not possible to install a new MAC on the eth0 interface..
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0x82000000, mmz_size: 96M
hisilicon: Get data from environment and set SENSOR as imx385
xxxx Invalid sensor type  xxxx
******* Error: There's something wrong, please check! *****
Starting majestic: OK

Welcome to OpenIPC v2.1
openipc-hi3516dv100 login: root

```

dimerr, 2022-01-15:
vi /usr/bin/load_hisilicon

find this line https://github.com/OpenIPC/firmware/blob/7cfe3d7057e21da6f24872f3d4623d1f740afd0e/general/package/hisilicon-osdrv-hi3516av100/files/script/load_hisilicon#L167

and make it 
`imx178 | imx385)`

gitgayhub, 2022-01-15:
Thanks, got some new error message

```
root@openipc-hi3516dv100:~# load_hisilicon -i
mmz_start: 0x82000000, mmz_size: 96M
hisilicon: Get data from environment and set SENSOR as imx385
insmod: can't insert 'mmz.ko': File exists
******* Error: There's something wrong, please check! *****
root@openipc-hi3516dv100:~# SENSOR=imx385 majestic
15:28:10 [    main] main@135                      Majestic Lite for HiSilicon, version master+c2f29d8, built on 2022-01-12
15:28:10 [app_conf] load_config@342               Using /etc/majestic.yaml as main configuration
15:28:10 [     sdk] sdk_specific_config@3536      SENSOR=imx385
15:28:10 [     sdk] find_sensor_config@3493       matched sensor config: imx385_i2c_1080p.ini
15:28:10 [     sdk] find_sensor_config@3510       Using /etc/sensors/imx385_i2c_1080p.ini as sensor configuration
15:28:10 [watchdog] watchdog_start@74             Watchdog device (/dev/watchdog) not found
15:28:10 [     sdk] start_sdk@250                 App was built with MPP version: Hi3516A_MPP_V1.0.8.0.B070 Release
15:28:10 [     sdk] start_sdk@253                 Current MPP version: HI_VERSION=Hi3516A_MPP_V1.0.8.0 B070 Release
15:28:10 [     sdk] start_sdk@277                 sensor imx385
15:28:10 [     sdk] start_sdk@282                   input_mode MIPI, WDR NONE
15:28:10 [     sdk] start_sdk@289                   dev [1920x1080]@4x4 25fps, RGGB
15:28:10 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx385.so
15:28:10 [     sdk] dump_vb_configuration@2488    VB configuration:
15:28:10 [     sdk] dump_vb_configuration@2498      [0]: 3110400 x 5
15:28:10 [     sdk] dump_vb_configuration@2498      [1]: 4147200 x 1
15:28:10 [     sdk] init_sensor@2691              Sensor driver loaded
15:28:10 [     log] printf@261                    linear mode FullLineStd :1350
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
15:28:10 [    puts]                               Open hi_i2c error!
[... 49 lines cut ...]
15:28:29 [     sdk] venc_read@1306                NOTE: Current frame is NULL!
15:28:31 [     sdk] venc_read@1286                Timeout get any data from venc channel 0
15:28:31 [     sdk] venc_read@1306                NOTE: Current frame is NULL!
15:28:33 [     sdk] venc_read@1286                Timeout get any data from venc channel 0
15:28:33 [     sdk] venc_read@1306                NOTE: Current frame is NULL!
15:28:35 [     hal] take_yuv420sp_video_fram@1081 Cannot get chn 0 frame for vpssGrp=0
15:28:35 [     hal] take_yuv420sp_video_fram@1081 ERR_VPSS_BUF_EMPTY: The picture queue is empty
[... cut ...]
15:28:39 [     hal] take_yuv420sp_video_fram@1081 Cannot get chn 0 frame for vpssGrp=0
15:28:39 [     hal] take_yuv420sp_video_fram@1081 ERR_VPSS_BUF_EMPTY: The picture queue is empty
15:28:39 [image_co] image_control_compute_his@100 Cannot take yuv420sp VI frame
15:28:39 [image_co] image_control_compute_his@100 ERR_VPSS_BUF_EMPTY: The picture queue is empty
15:28:39 [image_tu] tune_image_params@180         get_histogram errored
15:28:39 [    vpss] VpssUserGetFrame@364          [  vpss] vpss(0, 0) user get frame time out, s32MilliSec=10000

```

dimerr, 2022-01-15:
[libsns_imx385_i2c.zip](https://github.com/OpenIPC/firmware/files/7875321/libsns_imx385_i2c.zip)

gitgayhub, 2022-01-15:
> SENSOR=imx385 majestic

I replaced /usr/lib/sensors/libsns_imx385.so with this file, but it's the same error message

dimerr, 2022-01-15:
unfortunately this driver was compiled for an older i2c access method which is not supported now...

gitgayhub, 2022-01-15:
> unfortunately this driver was compiled for an older i2c access method which is not supported now...

do you mean this binary file is extracted from the camera flash dump directly? I thought it was compiled by you

gitgayhub, 2022-01-15:
I got another imx385 camera from the same company, but soc is 3516av200. Do you think it's possible to add support for it too? if no, I won't create another issue and waste your time

dimerr, 2022-01-15:
nobody yet have tested av200 platform. however we do have imx385 for av200. you can ttry flashing images, but i expect it might not even boot

gitgayhub, 2022-01-15:
I will try tomorrow, I see you do have an image for 3516av200. The OEM firmware recognize it as 3519v101, are they the same family too

Do you need me to create an issue if it won’t boot? Or if it won’t load imx385 driver? I’m happy to do the testing job but don’t want to waste your time when there’s no meaning

dimerr, 2022-01-15:
I guess you can, if it at least tries to boot, open an issue and post logs,etc, as much info as possible, because i dont have any boards on that platform so i've added it without any testing

gitgayhub, 2022-01-16:
https://stin.to/m2h3f
will give you login details there

closed 2022-01-20 by dimerr