OpenIPC/firmware #154: Nand flash not supported to create mtdblock

gitgayhub, 2022-01-14:
Hi, I have an UNV camera with hi3516av100, imx185, 512M memory and 128M nand flash.  I installed OpenIPC, but it seems that the nand flash is not supported, I got an error message when booting up

`VFS: Cannot open root device "mtdblock3" or unknown-block(0,0): error -6`

The nand flash is [Toshiba TC58BVG0S3HTA00](https://business.kioxia.com/info/docget.jsp?did=14875&prodName=TC58BVG0S3HTA00)

```
uboot # reset
resetting ...

U-Boot 2010.06 (Nov 01 2016 - 14:15:35)

NAND:  Check nand flash controller v610. found
Special NAND id table Version 1.36
Nand ID: 0x98 0xF1 0x80 0x15 0xF2 0x16 0x08 0x00
Block:128KB Page:2KB Chip:128MB*1 OOB:64B ECC:4bit/512
Check SPI Nand Flash Controller v100 ... Found
SPI Nand ID Table Version 1.3
SPI Nand : Manufacturer ID[0x0] No NAND device found!!!
128 MiB
Check spi flash controller v350... Found
Can't find a valid spi flash chip.
Can't find a valid spi flash chip.
In:    serial
Out:   serial
Err:   serial
UBOOT for QIPCV1R2
Detected MACID:48:ea:xxxxxx
NetBoardType:COPPER + SFP
Press Ctrl+B to abort autoboot in 2 seconds

NAND read: device 0 offset 0xa00000, size 0x200000
 2097152 bytes read: OK
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-4.9.37-hi3516av100
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1949156 Bytes = 1.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az167-852) (gcc version 7.5.0 (Buildroot -g9567237) ) #1 Fri Jan 14 03:47:17 UTC 2022
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3516A DEMO Board
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
Kernel command line: mem=512M totalmem=1024M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hinand:1M(boot),9M(kernel_unv),2M(kernel),5M(rootfs),83M(rootfs_data),1M(cliinfo),1M(clibak),1M(mtd_runtime),16M(config),8M(calibration),512K(update),-(other)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 514180K/524288K available (3983K kernel code, 174K rwdata, 892K rodata, 184K init, 224K bss, 10108K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03ebff0   (3984 kB)
      .init : 0xc04ce000 - 0xc04fc000   ( 184 kB)
      .data : 0xc04fc000 - 0xc0527b20   ( 175 kB)
       .bss : 0xc0529000 - 0xc0561308   ( 225 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
[... 115 lines cut ...]
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #1
Hardware name: Hisilicon Hi3516A (Flattened Device Tree)
Backtrace:
[<c0012a64>] (dump_backtrace) from [<c0012d40>] (show_stack+0x18/0x1c)
 r7:df7b0000 r6:c04ef858 r5:00000000 r4:c05292e8
[<c0012d28>] (show_stack) from [<c019ffd4>] (dump_stack+0x24/0x28)
[<c019ffb0>] (dump_stack) from [<c00797f0>] (panic+0xd0/0x238)
[<c0079724>] (panic) from [<c04cf314>] (mount_block_root+0x1e0/0x2b0)
 r3:00000000 r2:00000000 r1:df44fea0 r0:c0462d08
 r7:df7b0000
[<c04cf134>] (mount_block_root) from [<c04cf5f8>] (mount_root+0x128/0x130)
 r10:c04ef838 r9:c04ce5fc r8:c04ef834 r7:c0529000 r6:c0529024 r5:c050288c
 r4:00000000
[<c04cf4d0>] (mount_root) from [<c04cf784>] (prepare_namespace+0x184/0x1cc)
 r9:c04ce5fc r8:c04ef834 r7:c0529000 r6:c0529000 r5:c0529024 r4:c04ef858
[<c04cf600>] (prepare_namespace) from [<c04ceea8>] (kernel_init_freeable+0x1d0/0x1e0)
 r6:c0529000 r5:00000009 r4:c04cad10
[<c04cecd8>] (kernel_init_freeable) from [<c03e7534>] (kernel_init+0x10/0xfc)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03e7524
 r4:00000000
[<c03e7524>] (kernel_init) from [<c000f9e8>] (ret_from_fork+0x14/0x2c)
 r5:c03e7524 r4:00000000
Rebooting in 20 seconds..

```

gitgayhub, 2022-01-14:
Also I've extracted the driver files for multiple sensors from the UNV firmware, just in case you may find it useful
[lib.zip](https://github.com/OpenIPC/firmware/files/7870598/lib.zip)

dimerr, 2022-01-14:
it has to be hinand instead hi_sfc

dimerr, 2022-01-14:
[uImage.zip](https://github.com/OpenIPC/firmware/files/7871116/uImage.zip)
try this kernel

gitgayhub, 2022-01-15:
> [uImage.zip](https://github.com/OpenIPC/firmware/files/7871116/uImage.zip) try this kernel

Thanks but this kernel can't recognize the nand storage either.

```
bootargs=mem=512M totalmem=1024M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hinand:1M(boot),9M(kernel_unv),2M(kernel),5M(rootfs),83M(rootfs_data),1M(cliinfo),1M(clibak),1M(mtd_runtime),16M(config),8M(calibration),512K(update),-(other)
bootcmd=nand read 0x82000000 0xa00000 0x200000;bootm 0x82000000
```

```
brd: module loaded
loop: module loaded
Spi id table Version 1.22
SPI Nand ID Table Version 2.2
nand: This device[00,00] cannot found in spi nand id table!!
nand: No NAND device found
libphy: hisi_gemac_mii_bus: probed
libphy: Fixed MDIO Bus: probed
hi_gmac_v200 10090000.ethernet (unnamed net_device) (uninitialized): using random MAC address 36:63:5e:28:98:b8
phy 1 not found
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
ehci-platform: EHCI generic platform driver
ehci-platform 100b0000.ehci: EHCI Host Controller
ehci-platform 100b0000.ehci: new USB bus registered, assigned bus number 1
ehci-platform 100b0000.ehci: irq 19, io mem 0x100b0000
ehci-platform 100b0000.ehci: USB 2.0 started, EHCI 1.00
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
ohci-platform: OHCI generic platform driver
ohci-platform 100a0000.ohci: Generic Platform OHCI controller
ohci-platform 100a0000.ohci: new USB bus registered, assigned bus number 2
ohci-platform 100a0000.ohci: irq 20, io mem 0x100a0000
hub 2-0:1.0: USB hub found
hub 2-0:1.0: 1 port detected
usbcore: registered new interface driver usb-storage
i2c /dev entries driver
hisi-i2c-hisilicon 200d0000.i2c: Hisilicon [i2c-0] probed!
hisi-i2c-hisilicon 20240000.i2c: Hisilicon [i2c-1] probed!
hisi-i2c-hisilicon 20250000.i2c: Hisilicon [i2c-2] probed!
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
hisi-dmac 10060000.hidma-controller: hidmac probe!
Initializing XFRM netlink socket
NET: Registered protocol family 10
NET: Registered protocol family 17
NET: Registered protocol family 15
Key type dns_resolver registered
VFS: Cannot open root device "mtdblock3" or unknown-block(0,0): error -6
Please append a correct "root=" boot option; here are the available partitions:
0100           65536 ram0  (driver?)
0101           65536 ram1  (driver?)
0102           65536 ram2  (driver?)
0103           65536 ram3  (driver?)
0104           65536 ram4  (driver?)
0105           65536 ram5  (driver?)
0106           65536 ram6  (driver?)
0107           65536 ram7  (driver?)
0108           65536 ram8  (driver?)
0109           65536 ram9  (driver?)
010a           65536 ram10  (driver?)
010b           65536 ram11  (driver?)
010c           65536 ram12  (driver?)
010d           65536 ram13  (driver?)
010e           65536 ram14  (driver?)
010f           65536 ram15  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #9
Hardware name: Hisilicon Hi3516A (Flattened Device Tree)

```

dimerr, 2022-01-15:
do you have bootlog from stock kernel?

gitgayhub, 2022-01-15:
> do you have bootlog from stock kernel?

do you mean the bootlog of the original firmware?

dimerr, 2022-01-15:
yes

dimerr, 2022-01-15:
[uImage.zip](https://github.com/OpenIPC/firmware/files/7875275/uImage.zip)

gitgayhub, 2022-01-15:
> do you have bootlog from stock kernel?
```

uboot # tftp 0x82000000 uimage.bin
MVL P4 is link up, speed 1000M, full duplex
ETH0: PHY(phyaddr=1, rmii) link UP: DUPLEX=FULL : SPEED=100M
MAC:   48-EA-63-46-83-F2
TFTP from server 192.168.2.88; our IP address is 192.168.2.22
Download Filename 'uimage.bin'.
Download to address: 0x82000000
Downloading: #################################################
done
Bytes transferred = 6953800 (6a1b48 hex)
uboot # bootm 0x82000000
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    6953736 Bytes = 6.6 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Linux version 3.4.35 (jenkins@root123) (gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v400) ) #2 Mon Nov 22 15:41:57 CST 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: hi3516a
[    0.000000] Ignoring tag cmdline (using the default kernel command line)
[    0.000000] chip_id:0, type: Hi3516A, memory size is changed
[    0.000000] reserve : reserve 8Bytes mem
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 41148
[    0.000000] Kernel command line: mem=126M console=ttyAMA0,115200 mtdparts=hinand:1M(boot),9M(kernel),74M(program),16M(cfgbak),1M(cliinfo),1M(clibak),1M(mtd_runtime),16M(config),8M(calibration),512K(update),-(other)
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 162MB = 162MB total
[    0.000000] Memory: 149076k/149076k available, 16812k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xca800000 - 0xff000000   ( 840 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xca200000   ( 162 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc046f000   (4508 kB)
[    0.000000]       .init : 0xc046f000 - 0xc0e96574   (10398 kB)
[    0.000000]       .data : 0xc0e98000 - 0xc0eca300   ( 201 kB)
[    0.000000]        .bss : 0xc0eca324 - 0xc0ee4080   ( 104 kB)
[    0.000000] SLUB: Genslabs=11, HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:128
[    0.000000] sched_clock: 32 bits at 49MHz, resolution 20ns, wraps every 86767ms
[    0.000000] Console: colour dummy device 80x30
[    0.000133] Calibrating delay loop... 1196.85 BogoMIPS (lpj=5984256)
[    0.090074] pid_max: default: 32768 minimum: 301
[    0.090276] Mount-cache hash table entries: 512
[    0.090880] Initializing cgroup subsys freezer
[    0.090934] CPU: Testing write buffer coherency: ok
[    0.091134] Setting up static identity map for 0x80353b00 - 0x80353b58
[    0.093496] dummy:
[    0.093886] NET: Registered protocol family 16
[    0.097097] gpiochip_add: registered GPIOs 0 to 7 on device: hi3516-gpio0
[    0.097344] gpiochip_add: registered GPIOs 8 to 15 on device: hi3516-gpio1
[    0.097579] gpiochip_add: registered GPIOs 16 to 23 on device: hi3516-gpio2
[    0.097807] gpiochip_add: registered GPIOs 24 to 31 on device: hi3516-gpio3
[    0.098032] gpiochip_add: registered GPIOs 32 to 39 on device: hi3516-gpio4
[    0.098249] gpiochip_add: registered GPIOs 40 to 47 on device: hi3516-gpio5
[... 265 lines cut ...]
[   12.310286] UBI: image sequence number:  163536723
[   12.315111] UBI: background thread "ubi_bgt3d" started, PID 985
ubi3_0 exist.
        /cfgbak exists
        mount -t ubifs ubi3:cfgbak /cfgbak
[   12.373693] UBIFS: default file-system created
[   12.453439] UBIFS: mounted UBI device 3, volume 0, name "cfgbak"
[   12.459459] UBIFS: file system size:   14221312 bytes (13888 KiB, 13 MiB, 112 LEBs)
[   12.467165] UBIFS: journal size:       1015809 bytes (992 KiB, 0 MiB, 6 LEBs)
[   12.474321] UBIFS: media format:       w4/r0 (latest is w4/r0)
[   12.480155] UBIFS: default compressor: lzo
[   12.484266] UBIFS: reserved for root:  671707 bytes (655 KiB)
[RCS]: /etc/init.d/S40network
Starting network...
ip: RTNETLINK answers: File exists
[   12.731236] ADDRCONF(NETDEV_UP): eth0: link is not ready
touch: /var/log/messages: No such file or directory
/bin/sh: exec: line 1: /program/bin/init.sh: not found

root login: [   13.921017] PHY: mdio0:15 - Link is Up - 100/Full
[   13.925878] ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   63.861255] Switch logic port: 4 - Link is up(unlink to link) - 1000M/Full
[   63.868228] Switch logic port: 5 - Link is up(unlink to link) - 100M/Full

```

gitgayhub, 2022-01-15:
> [uImage.zip](https://github.com/OpenIPC/firmware/files/7875275/uImage.zip)

This uimage can recognize the nand flash, but can't mount mtdblock3 as squashfs parition

This is my bootargs
`bootargs=mem=512M totalmem=1024M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hinand:1M(boot),9M(kernel_unv),2M(kernel),5M(rootfs),83M(rootfs_data),1M(cliinfo),1M(clibak),1M(mtd_runtime),16M(config),8M(calibration),512K(update),-(other)`

```
uboot # nand read 0x82000000 0xc00000 50

NAND read: device 0 offset 0xc00000, size 0x50
 80 bytes read: OK
uboot # md 0x82000000 50
82000000: 73717368 00000316 61dfa081 00020000    hsqs.......a....
82000010: 0000002a 00110004 000200c0 00000004    *...............
82000020: 121a0f6e 00000000 004a1532 00000000    n.......2.J.....
82000030: 004a152a 00000000 ffffffff ffffffff    *.J.............
82000040: 0049dd38 00000000 0049f334 00000000    8.I.....4.I.....
82000050: 004a10ca 00000000 004a1518 00000000    ..J.......J.....
82000060: 587a37fd 0100005a 36de2269 9ab5c003    .7zXZ...i".6....
82000070: 08808005 000a0121 9713415b efde7de1    ....!...[A...}..
82000080: 3f005dfe 68844591 a6dede3b d4f0230f    .].?.E.h;....#..
82000090: 3c941924 0b9d2d1d 3933fc49 149e0045    $..<.-..I.39E...
820000a0: 09b85a7c b3ff5118 b71101d3 199b04fe    |Z...Q..........
820000b0: 28d54da7 a8b16160 683c84ad bd0a1647    .M.(`a....<hG...
820000c0: 582c8a9e d74efef6 27ffca5f 313f5f8d    ..,X..N._..'._?1
820000d0: afd51fd6 f272d8d3 31a31cb1 b08457fc    ......r....1.W..
820000e0: 050c83eb ebb989c9 58b7c00a 2029d72c    ...........X,.)
820000f0: 91aedfeb 0a3a11a9 ac5bcf6b a027e6a9    ......:.k.[...'.
82000100: bd4689dc 9df68b0d 92c27973 4b877adf    ..F.....sy...z.K
82000110: bb72aec6 12c54439 48863456 e2f0b284    ..r.9D..V4.H....
82000120: 933c4c51 63e058be 5aeba7ee 8bc43910    QL<..X.c...Z.9..
82000130: d06487fd 5cf275c1 6b7f327e 434c25f0    ..d..u.\~2.k.%LC
```

I verified that the contents on nand is the same with the rootfs image

gitgayhub, 2022-01-15:
Forgot to include bootlog

```
Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (pk@ps) (gcc version 7.5.0 (Buildroot -ge849e6b) ) #15 Sat Jan 15 18:31:30 MSK 2022
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3516A DEMO Board
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
Kernel command line: mem=512M totalmem=1024M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hinand:1M(boot),9M(kernel_unv),2M(kernel),5M(rootfs),83M(rootfs_data),1M(cliinfo),1M(clibak),1M(mtd_runtime),16M(config),8M(calibration),512K(update),-(other)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 513992K/524288K available (4120K kernel code, 184K rwdata, 928K rodata, 188K init, 225K bss, 10296K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc040e4e0   (4122 kB)
      .init : 0xc04f9000 - 0xc0528000   ( 188 kB)
      .data : 0xc0528000 - 0xc05561a0   ( 185 kB)
       .bss : 0xc0558000 - 0xc0590448   ( 226 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
sched_clock: 32 bits at 3000kHz, resolution 333ns, wraps every 715827882841ns
Console: colour dummy device 80x30
Calibrating delay loop... 1208.32 BogoMIPS (lpj=6041600)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x80008200 - 0x80008258
devtmpfs: initialized
VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
Serial: AMBA PL011 UART driver
20080000.uart: ttyAMA0 at MMIO 0x20080000 (irq = 18, base_baud = 0) is a PL011 rev2
console [ttyAMA0] enabled
SCSI subsystem initialized
ssp-pl022 200c0000.spi: ARM PL022 driver, device ID: 0x00800022
ssp-pl022 200c0000.spi: mapped registers from 0x200c0000 to fe1c0000
ssp-pl022 200e0000.spi: ARM PL022 driver, device ID: 0x00800022
ssp-pl022 200e0000.spi: mapped registers from 0x200e0000 to fe1e0000
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource arm,sp804
NET: Registered protocol family 2
TCP established hash table entries: 4096 (order: 2, 16384 bytes)
TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
TCP: Hash tables configured (established 4096 bind 4096)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
workingset: timestamp_bits=30 max_order=17 bucket_order=0
[... 92 lines cut ...]
0109           65536 ram9  (driver?)
010a           65536 ram10  (driver?)
010b           65536 ram11  (driver?)
010c           65536 ram12  (driver?)
010d           65536 ram13  (driver?)
010e           65536 ram14  (driver?)
010f           65536 ram15  (driver?)
1f00            1024 mtdblock0  (driver?)
1f01            9216 mtdblock1  (driver?)
1f02            2048 mtdblock2  (driver?)
1f03            5120 mtdblock3  (driver?)
1f04           84992 mtdblock4  (driver?)
1f05            1024 mtdblock5  (driver?)
1f06            1024 mtdblock6  (driver?)
1f07            1024 mtdblock7  (driver?)
1f08           16384 mtdblock8  (driver?)
1f09            8192 mtdblock9  (driver?)
1f0a             512 mtdblock10  (driver?)
1f0b             512 mtdblock11  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #15
Hardware name: Hisilicon Hi3516A (Flattened Device Tree)
Backtrace:

```

dimerr, 2022-01-15:
now you can flash our rootfs... if you have nand backup ofcourse

gitgayhub, 2022-01-15:
> now you can flash our rootfs... if you have nand backup ofcourse

mtdblock3 is flashed with openipc rootfs, but can't be mounted

dimerr, 2022-01-15:
[uImage.zip](https://github.com/OpenIPC/firmware/files/7875353/uImage.zip)

gitgayhub, 2022-01-15:
> [uImage.zip](https://github.com/OpenIPC/firmware/files/7875353/uImage.zip)

Got endless error message about jffs2

```
12 cmdlinepart partitions found on MTD device hinand
Creating 12 MTD partitions on "hinand":
0x000000000000-0x000000100000 : "boot"
0x000000100000-0x000000a00000 : "kernel_unv"
0x000000a00000-0x000000c00000 : "kernel"
0x000000c00000-0x000001100000 : "rootfs"
0x000001100000-0x000006400000 : "rootfs_data"
0x000006400000-0x000006500000 : "cliinfo"
0x000006500000-0x000006600000 : "clibak"
0x000006600000-0x000006700000 : "mtd_runtime"
0x000006700000-0x000007700000 : "config"
0x000007700000-0x000007f00000 : "calibration"
0x000007f00000-0x000007f80000 : "update"
0x000007f80000-0x000008000000 : "other"
libphy: hisi_gemac_mii_bus: probed
libphy: Fixed MDIO Bus: probed
hi_gmac_v200 10090000.ethernet (unnamed net_device) (uninitialized): using random MAC address da:35:75:46:af:60
phy 1 not found
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
ehci-platform: EHCI generic platform driver
ehci-platform 100b0000.ehci: EHCI Host Controller
ehci-platform 100b0000.ehci: new USB bus registered, assigned bus number 1
ehci-platform 100b0000.ehci: irq 19, io mem 0x100b0000
ehci-platform 100b0000.ehci: USB 2.0 started, EHCI 1.00
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
ohci-platform: OHCI generic platform driver
ohci-platform 100a0000.ohci: Generic Platform OHCI controller
ohci-platform 100a0000.ohci: new USB bus registered, assigned bus number 2
ohci-platform 100a0000.ohci: irq 20, io mem 0x100a0000
hub 2-0:1.0: USB hub found
hub 2-0:1.0: 1 port detected
usbcore: registered new interface driver usb-storage
i2c /dev entries driver
hisi-i2c-hisilicon 200d0000.i2c: Hisilicon [i2c-0] probed!
hisi-i2c-hisilicon 20240000.i2c: Hisilicon [i2c-1] probed!
hisi-i2c-hisilicon 20250000.i2c: Hisilicon [i2c-2] probed!
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
hisi-dmac 10060000.hidma-controller: hidmac probe!
Initializing XFRM netlink socket
NET: Registered protocol family 10
NET: Registered protocol family 17
NET: Registered protocol family 15
Key type dns_resolver registered
VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
devtmpfs: mounted
Freeing unused kernel memory: 188K (c04f9000 - c0528000)
This architecture does not have kernel memory protection.
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00000000: 0x4255 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00000004: 0x0001 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00000018: 0xc243 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0000003c: 0x72e5 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00020000: 0x4255 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00020004: 0x0001 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00020018: 0xc243 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0002003c: 0x72e5 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00040000: 0x4255 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00040004: 0x0001 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00040018: 0xc243 instead
random: fast init done
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0004003c: 0x72e5 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00060000: 0x4255 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00060004: 0x0001 instead

```

dimerr, 2022-01-15:
because rootfs_data must be erased before use, otherwise it will try to do that for a very long time

gitgayhub, 2022-01-15:
> because rootfs_data must be erased before use, otherwise it will try to do that for a very long time

thanks, I set rootfs_data to 3M, and there's no jffs2 error now, but boot process stuck at `hisilicon: Get data from environment and set SENSOR as imx185`

and there seems to be some error about the mount command

```
hisi-dmac 10060000.hidma-controller: hidmac probe!
Initializing XFRM netlink socket
NET: Registered protocol family 10
NET: Registered protocol family 17
NET: Registered protocol family 15
Key type dns_resolver registered
VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
devtmpfs: mounted
Freeing unused kernel memory: 188K (c04f9000 - c0528000)
This architecture does not have kernel memory protection.
BusyBox v1.31.1 (2022-01-15 03:30:53 UTC) multi-call binary.

Usage: mount [OPTIONS] [-o OPT] DEVICE NODE

Mount a filesystem. Filesystem autodetrandom: fast init done
ection requires /proc.

        -a              Mount all filesystems in fstab
        -r              Read-only mount
        -t FSTYPE[,...] Filesystem type(s)
        -T FILE         Read FILE instead of /etc/fstab
        -O OPT          Mount only filesystems with option OPT (-a only)
-o OPT:
        loop            Ignored (loop devices are autodetected)
        [a]sync         Writes are [a]synchronous
        [no]atime       Disable/enable updates to inode access times
        [no]diratime    Disable/enable atime updates to directories
        [no]relatime    Disable/enable atime updates relative to modification time
        [no]dev         (Dis)allow use of special device files
        [no]exec        (Dis)allow use of executable files
        [no]suid        (Dis)allow set-user-id-root programs
        [r]shared       Convert [recursively] to a shared subtree
        [r]slave        Convert [recursively] to a slave subtree
        [r]private      Convert [recursively] to a private subtree
        [un]bindable    Make mount point [un]able to be bind mounted
        [r]bind         Bind a file or directory [recursively] to another location
        move            Relocate an existing mount point
        remount         Remount a mounted filesystem, changing flags
        ro              Same as -r

There are filesystem-specific -o flags.
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Loading modules:Saving random seed: OK
Starting rngd: OK
Starting mdev...
Starting network: ip: SIOCGIFINDEX: No such device
FAIL
Starting ntpd: OK
Starting dropbear sshd: OK
Starting httpd: OK
Starting mini-snmpd: DISABLED, OK
Starting telnetd: DISABLED, OK
Starting crond: OK
hisilicon: Check MAC for Xiongmai devices
hisilicon: The eth0 interface has a correct MAC - 48:ea:63:46:83:f2
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0xa0000000, mmz_size: 512M
hisilicon: Get data from environment and set SENSOR as imx185

```

dimerr, 2022-01-16:
Try the latest build from releases,  but it's very not clear what is going on there, we have NAND successfuly working on other devices.

Also do 
`setenv sensor unknown;saveenv`

in uboot, to avoid loading sdk modules, if thats the reason why it hangs

gitgayhub, 2022-01-16:
> setenv sensor unknown;saveenv

Thanks, this solved the problem, now it can boot in to system

but the network device didn't start

I see there's the imx185 driver .so file but no config file under /etc/sensors

```
Loading modules:Saving random seed: OK
Starting rngd: OK
Starting mdev...
Starting network: ip: SIOCGIFINDEX: No such device
FAIL
Starting ntpd: OK
Starting dropbear sshd: OK
Starting httpd: OK
Starting mini-snmpd: DISABLED, OK
Starting telnetd: DISABLED, OK
Starting crond: OK
hisilicon: Check MAC for Xiongmai devices
hisilicon: The eth0 interface has a correct MAC - 48:ea:63:46:83:f2
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0xa0000000, mmz_size: 512M
hisilicon: Get data from environment and set SENSOR as unknown
Starting majestic: OK

```

dimerr, 2022-01-16:
[uImage.zip](https://github.com/OpenIPC/firmware/files/7876617/uImage.zip)
where do you get all that stuff from? what the heck, camera with 7 port gigabit switch?..

gitgayhub, 2022-01-16:
> [uImage.zip](https://github.com/OpenIPC/firmware/files/7876617/uImage.zip) where do you get all that stuff from? what the heck, camera with 7 port gigabit switch?..

I bought it from online used market, it's a traffic camera with a RJ45 and a SFP fiber port, so I guess there should be a switch

Will try it later

gitgayhub, 2022-01-16:
It's still the same error message: `Starting network: ip: SIOCGIFINDEX: No such device`

Since it's not a typical device, let's ignore the switch problem

Thanks for your help

```
libphy: hisi_gemac_mii_bus: probed
libphy: Fixed MDIO Bus: probed
libphy: mdio_driver_register: mv88e6085
hi_gmac_v200 10090000.ethernet (unnamed net_device) (uninitialized): using random MAC address be:3d:xxxxxxx
phy 1 not found
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
ehci-platform: EHCI generic platform driver
ehci-platform 100b0000.ehci: EHCI Host Controller
ehci-platform 100b0000.ehci: new USB bus registered, assigned bus number 1
ehci-platform 100b0000.ehci: irq 19, io mem 0x100b0000
ehci-platform 100b0000.ehci: USB 2.0 started, EHCI 1.00
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
ohci-platform: OHCI generic platform driver
ohci-platform 100a0000.ohci: Generic Platform OHCI controller
ohci-platform 100a0000.ohci: new USB bus registered, assigned bus number 2
ohci-platform 100a0000.ohci: irq 20, io mem 0x100a0000
hub 2-0:1.0: USB hub found
hub 2-0:1.0: 1 port detected
usbcore: registered new interface driver usb-storage
i2c /dev entries driver
hisi-i2c-hisilicon 200d0000.i2c: Hisilicon [i2c-0] probed!
hisi-i2c-hisilicon 20240000.i2c: Hisilicon [i2c-1] probed!
hisi-i2c-hisilicon 20250000.i2c: Hisilicon [i2c-2] probed!
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
hisi-dmac 10060000.hidma-controller: hidmac probe!
Initializing XFRM netlink socket
NET: Registered protocol family 10
NET: Registered protocol family 17
NET: Registered protocol family 15
Key type dns_resolver registered
VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
devtmpfs: mounted
Freeing unused kernel memory: 192K (c051e000 - c054e000)
This architecture does not have kernel memory protection.
BusyBox v1.31.1 (2022-01-16 03:34:55 UTC) multi-call binary.

```

closed 2022-01-16 by gitgayhub