OpenIPC/firmware #155: Размер osd шрифта

Oleglevsha, 2022-01-15:
Возможно сделать адаптивный размер шрифта, в зависимости от разрешения камеры?
сейчас нормальный размер для 5мп
для 1мп огромный
для 2мп приемлемо но все же большой

widgetii, 2022-01-15:
В development группе обсуждали, что в данном виде OSD представляет собой MVP (только платформа HiSilicon)

Для дальнейшего развития этого направления требуется составление всеобъемлющего технического задания по этому блоку включающего в себя:
1) Поддержка одновременно нескольких разных шрифтов и задание их атрибутов пользователем
2) Различные варианты использования OSD, в том числе несколько областей на экране, разные настройки по каналам и т.п.
3) Задание внешних источников для OSD
4) Адаптация для всех поддерживаемых OpenIPC платформ

ТЗ должно быть в виде, готовом для передачи разработчику, незнакомому с данной предметной областью.

OlegLxx, 2023-05-16:
Напишу в виде комментария пожелания\идеи.
1. Реализовать перенос строки согласно документации [strftime(3)](https://man7.org/linux/man-pages/man3/strftime.3.html). Там есть символ переноса строки: **%n** _A newline character._
Однако сейчас он не работает, строка не переносится.
_%Y-%m-%e  %T %Z **%n** test_
[image]

2. Отображение текущего разрешения потока\субпотока.
3. Отображение температуры чипа.

closed 2026-09-26 by openipc-ai