OpenIPC/firmware #156: No network connection on Hi3516AV200

gitgayhub, 2022-01-16:
Hi, so this is the new camera with Hi3516AV200, IMX385, 1GB memory and 32MB flash. I flashed the newest OpenIPC image, it can boot into system without any problem, but without network access

dmesg log show eth0 link is up, default mode is dhcp, but can't get an ip address from router, the ipaddr in uboot env is used instread as eth0 ip address, but there's no connection at all

 I modified /etc/network/interface to use static ip, but it's the same result, eth0 shows as up but no connection at all, ping gateway ip will time out

```
dmesg: 
IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
hi_gmac_v200 10050000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

```

```
U-Boot 2010.06 (Nov 29 2017 - 20:31:10)

Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xc2 0x20 0x19
Block:64KB Chip:32MB Name:"MX25L(256/257)XX"
SPI Nor total size: 32MB
MMC:
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
In:    serial
Out:   serial
Err:   serial
Hit any key to stop autoboot:  1 16av200                                                                                                                                                                          0
32768 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.18.20-hi3516av200
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2021054 Bytes = 1.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Initializing cgroup subsys cpu
Linux version 3.18.20 (runner@fv-az186-929) (gcc version 7.5.0 (Buildroot -g7cfe3d7) ) #1 SMP Sat Jan 15 03:44:14 UTC 2022
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: Hisilicon HI3516AV200 DEMO Board
cmz zone is not set!
cma: Reserved 16 MiB at 0x9f000000
Memory policy: Data cache writealloc
PERCPU: Embedded 9 pages/cpu @debcc000 s7424 r8192 d21248 u36864
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
Kernel command line: mem=512M totalmem=1024M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:384k(boot),128k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 497564K/524288K available (3844K kernel code, 196K rwdata, 1036K rodata, 220K init, 310K bss, 26724K reserved, 0K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
    vmalloc : 0xe0800000 - 0xff000000   ( 488 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc04cc2c4   (4881 kB)
      .init : 0xc04cd000 - 0xc0504000   ( 220 kB)
      .data : 0xc0504000 - 0xc0535220   ( 197 kB)
       .bss : 0xc0535220 - 0xc0582d20   ( 311 kB)
[... 243 lines cut ...]
    pre-up wifi xm711
    pre-up wpa_passphrase "SSID" "password" >/tmp/wpa_supplicant.conf
    pre-up sed -i '2i \\tscan_ssid=1' /tmp/wpa_supplicant.conf
    pre-up (sleep 3; wpa_supplicant -B -Dnl80211 -ieth2 -c/tmp/wpa_supplicant.conf)
    post-down killall -q wpa_supplicant

manual usb0
iface usb0 inet dhcp
    pre-up echo 9 > /sys/class/gpio/export
    pre-up echo out > /sys/class/gpio/gpio9/direction
    pre-up echo 0 > /sys/class/gpio/gpio9/value
    pre-up modprobe usbserial vendor=0x2c7c product=0x6026
    pre-up modprobe rndis_host
    pre-up sleep 10

manual wg0
iface wg0 inet static
    address 192.168.99.1
    netmask 255.255.255.0
    pre-up modprobe wireguard
    pre-up ip link add dev wg0 type wireguard
    pre-up wg setconf wg0 /etc/wireguard.conf
    post-down ip link del dev wg0

```

gitgayhub, 2022-01-16:
And I don't see imx385 driver and config files in this firmware, only very limited sensor files included

```
root@openipc-hi3516av200:/usr/lib/sensors# ls -alh
total 195K
drwxr-xr-x    2 root     root         189 Jan 15  2022 .
drwxr-xr-x    3 root     root        2.1K Jan 15  2022 ..
-rw-r--r--    1 root     root       23.2K Jan 15  2022 libsns_imx226.so
-rw-r--r--    1 root     root       23.6K Jan 15  2022 libsns_imx274.so
-rw-r--r--    1 root     root       32.1K Jan 15  2022 libsns_imx290.so
-rw-r--r--    1 root     root       22.7K Jan 15  2022 libsns_imx326.so
-rw-r--r--    1 root     root       36.1K Jan 15  2022 libsns_os05a.so
-rw-r--r--    1 root     root       28.0K Jan 15  2022 libsns_os08a10.so
-rw-r--r--    1 root     root       26.8K Jan 15  2022 libsns_ov4689_slave.so
```

dimerr, 2022-01-16:
do you have stock kernel bootlog?

gitgayhub, 2022-01-16:
booted the oem kernel from memory, with the openipc rootfs on it. do I have to flash the oem rootfs back to camera too?

```
hisilicon # tftp 0x82000000 3516av200_imx385_mtdblock1
ETH0: PHY(phyaddr=1, rmii) link UP: DUPLEX=FULL : SPEED=100M
MAC:   00-00-23-34-45-23
TFTP from server 192.168.2.88; our IP address is 192.168.2.28
Download Filename '3516av200_imx385_mtdblock1'.
Download to address: 0x82000000
Downloading: #################################################
done
Bytes transferred = 3145728 (300000 hex)
hisilicon # bootm 0x82000000
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.18.20
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2956426 Bytes = 2.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Initializing cgroup subsys cpu
Linux version 3.18.20 (root@wang) (gcc version 4.9.4 20150629 (prerelease) (Hisilicon_v500_20150831) ) #5 SMP Thu Nov 30 15:35:01 CST 2017
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: Hisilicon HI3516AV200 DEMO Board
cmz zone is not set!
cma: Reserved 16 MiB at 0x9f000000
Memory policy: Data cache writealloc
PERCPU: Embedded 9 pages/cpu @debcc000 s7424 r8192 d21248 u36864
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
Kernel command line: mem=512M totalmem=1024M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:384k(boot),128k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 497216K/524288K available (4080K kernel code, 200K rwdata, 1128K rodata, 212K init, 339K bss, 27072K reserved, 0K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
    vmalloc : 0xe0800000 - 0xff000000   ( 488 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc051e5d4   (5210 kB)
      .init : 0xc051f000 - 0xc0554000   ( 212 kB)
      .data : 0xc0554000 - 0xc0586140   ( 201 kB)
       .bss : 0xc0586140 - 0xc05dafbc   ( 340 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Hierarchical RCU implementation.
        RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS:16 nr_irqs:16 16
Architected cp15 timer(s) running at 24.00MHz (phys).
sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 2863311519744ns
Switching to timer-based delay loop, resolution 41ns
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Setting up static identity map for 0x803e4ea0 - 0x803e4ef8
ARM CCI driver probed
CPU1: Booted secondary processor
CPU1: thread -1, cpu 0, socket 1, mpidr 80000100
[... 151 lines cut ...]
[<c0011964>] (show_stack) from [<c03e0c50>] (dump_stack+0x88/0x98)
[<c03e0c50>] (dump_stack) from [<c03dee34>] (panic+0x9c/0x1fc)
[<c03dee34>] (panic) from [<c0520160>] (mount_block_root+0x218/0x298)
[<c0520160>] (mount_block_root) from [<c05204c0>] (prepare_namespace+0x190/0x1d4)
[<c05204c0>] (prepare_namespace) from [<c051fe2c>] (kernel_init_freeable+0x1c8/0x1d8)
[<c051fe2c>] (kernel_init_freeable) from [<c03de238>] (kernel_init+0x8/0xe8)
[<c03de238>] (kernel_init) from [<c000e540>] (ret_from_fork+0x14/0x34)
CPU0: stopping
CPU: 0 PID: 0 Comm: swapper/0 Not tainted 3.18.20 #5
[<c001487c>] (unwind_backtrace) from [<c0011964>] (show_stack+0x10/0x14)
[<c0011964>] (show_stack) from [<c03e0c50>] (dump_stack+0x88/0x98)
[<c03e0c50>] (dump_stack) from [<c0013bc8>] (handle_IPI+0x1cc/0x1ec)
[<c0013bc8>] (handle_IPI) from [<c0008620>] (gic_handle_irq+0x74/0x78)
[<c0008620>] (gic_handle_irq) from [<c0012400>] (__irq_svc+0x40/0x54)
Exception stack(0xc0555f48 to 0xc0555f90)
5f40:                   fffffff0 00000000 fffffff0 c0019980 c0554018 c055c460
5f60: c055c4c4 00000000 00000000 00000000 c0554000 c03e69a8 c055d220 c0555f90
5f80: c000f0e4 c000f0e8 60000113 ffffffff
[<c0012400>] (__irq_svc) from [<c000f0e8>] (arch_cpu_idle+0x38/0x3c)
[<c000f0e8>] (arch_cpu_idle) from [<c0050b98>] (cpu_startup_entry+0x228/0x29c)
[<c0050b98>] (cpu_startup_entry) from [<c051fc58>] (start_kernel+0x398/0x3a4)
[<c051fc58>] (start_kernel) from [<80008074>] (0x80008074)
Rebooting in 20 seconds..

```

dimerr, 2022-01-16:
devmem 120100ec 32 9f

gitgayhub, 2022-01-16:
> devmem 120100ec 32 9f

```
root@openipc-hi3516av200:~# devmem 0x120100ec 32
0x0000003F
root@openipc-hi3516av200:~# devmem 0x120100ec 32 0x9f
root@openipc-hi3516av200:~# devmem 0x120100ec 32
0x0000009F
```

dimerr, 2022-01-16:
after that eth0 should start receiving data.
if not, then in uboot try to ping something and  then without reset do 
`run bootcmd`

gitgayhub, 2022-01-16:
I tried both method, but none worked, tired dhcp and static ip, it's the same

```
hisilicon # sf probe 0
32768 KiB hi_fmc at 0:0 is now current device
hisilicon # sf read 0x82000000 0x80000 0x200000

hisilicon # ping 192.168.2.1
ETH0: PHY(phyaddr=1, rmii) link UP: DUPLEX=FULL : SPEED=100M
MAC:   00-00-23-34-45-23
host 192.168.2.1 is alive
hisilicon # sf read 0x82000000 0x80000 0x200000
hisilicon # bootm 0x82000000
```
```
root@openipc-hi3516av200:~# devmem 0x120100ec 32
0x0000003F
```

widgetii, 2022-01-16:
Could you run `ipctool` on the system and show its output?

gitgayhub, 2022-01-16:
```
root@openipc-hi3516av200:/mnt/mmcblk0p1# ./ipctool
---
board:
  vendor: OpenIPC
  version: 2.2.01.15
chip:
  vendor: HiSilicon
  model: 3519V101
ethernet:
  mac: "00:00:23:34:45:23"
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x60000
        sha1: 02d017de
      - name: env
        size: 0x20000
        sha1: 54135b4c
      - name: kernel
        size: 0x200000
        sha1: 3280b3c6
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: 05018b14
      - name: rootfs_data
        size: 0x1880000
        path: /overlay,jffs2,rw
    size: 32M
ram:
  total: 512M
firmware:
  kernel: "3.18.20 (SMP Sat Jan 15 03:44:14 UTC 2022)"
  toolchain: gcc version 7.5.0 (Buildroot -g7cfe3d7)

```

widgetii, 2022-01-16:
Is it real 3519V101 or 3516AV200?

gitgayhub, 2022-01-16:
it's printed Hi3516ARFCV200 on the back of the chip
in uboot env, there's also a `hisicpu=hi3516av200`, but the oem firmware bootlog says 3519v101, ipctool inside oem firmware says 3519v101 too

gitgayhub, 2022-01-16:
```
root@openipc-hi3516av200:~# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 48.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 2 (v7l)
BogoMIPS        : 48.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x1
CPU part        : 0xc0e
CPU revision    : 2

Hardware        : Hisilicon Hi3516av200 (Flattened Device Tree)
Revision        : 0000
Serial          : 0000000000000000

```

dimerr, 2022-01-16:
[uImage.zip](https://github.com/OpenIPC/firmware/files/7876936/uImage.zip)

gitgayhub, 2022-01-16:
> [uImage.zip](https://github.com/OpenIPC/firmware/files/7876936/uImage.zip)

all results are the same, both devmem 0x120100ec 32 and uboot ping method

dimerr, 2022-01-16:
please provide bootlog around this line

ETH: rgmii, phy_addr=1

gitgayhub, 2022-01-16:
```
0x000000780000-0x000002000000 : "rootfs_data"
SPI Nand ID Table Version 2.4
Cannot found a valid SPI Nand Device
libphy: Fixed MDIO Bus: probed
libphy: hisi_gemac_mii_bus: probed
hi_gmac_v200 10050000.ethernet (unnamed net_device) (uninitialized): using random MAC address ea:66:fb:cb:d8:ee
attached PHY 1 to driver Generic PHY, PHY_ID=0x1cc816
higmac: ETH MAC supporte CCI.
Higmac dma_sg_phy: 0x9dc80000
ETH: rgmii, phy_addr=1
xhci-hcd 10180000.xhci: xHCI Host Controller
xhci-hcd 10180000.xhci: new USB bus registered, assigned bus number 1
xhci-hcd 10180000.xhci: irq 54, io mem 0x10180000

```

dimerr, 2022-01-16:
[uImage.zip](https://github.com/OpenIPC/firmware/files/7876964/uImage.zip)

gitgayhub, 2022-01-16:
I installed the firmware for 3519v101: openipc.hi3519v101-br.tgz, everyhing is the same, same result, same boot log, same error message

dimerr, 2022-01-16:
there is no difference between them,except kernel.

and as for the kernel - for now try  only builds that  im posting

gitgayhub, 2022-01-16:
> [uImage.zip](https://github.com/OpenIPC/firmware/files/7876964/uImage.zip)

Great! flashed this uimage with 3519v101's rootfs remaining on flash, network is working now

I don't see imx385 driver in /usr/lib/sensors

dimerr, 2022-01-16:
Can you arrange remote ssh access to this cam? Because I'm afraid sensor driver is the least problem now, need to work on init scripts first

gitgayhub, 2022-01-16:
https://stin.to/m2h3f
will give you login details there

gitgayhub, 2022-01-17:
ssh ported has been temporarily disabled, please let me know if you need to access it again

gitgayhub, 2022-01-17:
Found [another firmware](http://www.hankvision.com/Admin/Others/DownloadsPage.aspx?nid=14&id=68&ref=down_a&t=ac18a35230a99092) for 3516av200/3519v101 from another oem company, see if it will help

the rootfs is just a compressed file don't know how to download it to flash chip. I do see .so driver files for multiple sensors including imx385

gitgayhub, 2022-01-19:
[av200_imx385_512M.zip](https://github.com/OpenIPC/firmware/files/7896898/av200_imx385_512M.zip)

gitgayhub, 2022-01-19:
[3516av200_yaffs_mounted.zip](https://github.com/OpenIPC/firmware/files/7897862/3516av200_yaffs_mounted.zip)

gitgayhub, 2022-01-21:
do you still need the ssh access?

closed 2022-01-21 by gitgayhub