OpenIPC/firmware #157: xm510: sysupgrade -r fails with losetup error

schneemaier, 2022-01-16:
Running sysupgare from the UI or from the commandline fails with error for the rootfs.

**Note: kernel is from https://github.com/OpenIPC/firmware/issues/143#issuecomment-1007887408 to allow ssh, upgrade with stock kernel works**

```
root@sch-openipc-xm510:/tmp# ls
Isplock           ipctool           ispmsg2
fw_printenv.lock  ispmsg            resolv.conf
root@sch-openipc-xm510:/tmp# sysupgrade -r

Forced shutdown of services...

killall: majestic: no process killed
killall: crond: no process killed
killall: klogd: no process killed
killall: ntpd: no process killed
killall: rngd: no process killed
killall: syslogd: no process killed

Downloading the latest firmware...

https://github.com/OpenIPC/firmware/releases/download/latest/openipc.xm510-br.tgz
rootfs.squashfs.xm510
rootfs.squashfs.xm510.md5sum
uImage.xm510
uImage.xm510.md5sum

Check RAM size...

              total        used        free      shared  buff/cache   available
Mem:          15120        5128        1676           0        8316           0
-/+ buffers/cache:         5128        9992
Swap:             0           0           0

Rootfs: Checksum verified...

losetup: /tmp/rootfs.squashfs.xm510: No such file or directory

Rootfs: Unable to get hostname, execution was interrupted...

root@sch-openipc-xm510:/tmp#
```

Runing losetup ends with error:
```
root@sch-openipc-xm510:/tmp# ls
Isplock                       rootfs
fw_printenv.lock              rootfs.squashfs.xm510
ipctool                       rootfs.squashfs.xm510.md5sum
ispmsg                        uImage.xm510
ispmsg2                       uImage.xm510.md5sum
resolv.conf
root@sch-openipc-xm510:/tmp# losetup /dev/loop0 /tmp/rootfs.squashfs.xm510
losetup: /tmp/rootfs.squashfs.xm510: No such file or directory
root@sch-openipc-xm510:/tmp#
```

widgetii, 2022-01-19:
The kernel was built without loopback device support that's why `sysupgrade` can be used only with `--force` option

schneemaier, 2022-01-19:
Thanks. I was successful with the --force option. As it is not possible to provide the force option via the WebUI i can manually upgrade via ssh or can modifiy the sysupgrade script 

Thanks Again

closed 2022-01-19 by schneemaier