OpenIPC/firmware #158: Перезагрузки камеры

esuslik, 2022-01-18:
Прошил камеру 000529E9
Перезагружается 1-2 раза за 20 минут. Даже при отключеной настройке watchdog
прошил камеру 000529B2
Перестает работать сначала поток через netip(через 40 минут) и камера перезагружается.

Где и как собрать логи, чтобы предоставить вам для выявления причин перезагрузки?

Также обратил внимание на сенсор SoC temp.
При включеной подсветке и долгой работе камеры, данная температура поднимается до 116 градусов. Это нормальная температура для работы камеры? при условии что в помещении примерно 25 градусов.

widgetii, 2022-01-19:
Here is 3516EV300 SoC temperature from my indoor camera (ambient temperature is about 24C)

[image]

I think you have chip overheating caused by hardware issues

esuslik, 2022-01-19:
это новые камеры только из коробки 2ух разных производителей

cronyx, 2022-01-19:
Радиаторы на камерах стоят ?

esuslik, 2022-01-19:
стоят
[image]

esuslik, 2022-01-19:
одна камера работает с температурой 80 но всеравно перезагружается.
И не работает видеопоток через netip. После перезагрузки начинает работать и потом опять перестает.
Какие логи посмотреть\скинуть чтобы понять почему идет перезагрузка и падает поток?

widgetii, 2022-01-19:
Connect to UART, and take your attention to what's going on in the console in the moment camera reboots

esuslik, 2022-01-19:
эх, у меня нет uart

esuslik, 2022-01-19:
еще обратил внимание что камера может не перезагружаться, а видео поток не работает
лог:

Jan 19 14:28:54 openipc-hi3516ev300 kern.emerg kernel: load mipi_rx driver successful!
Jan 19 14:28:54 openipc-hi3516ev300 kern.emerg kernel: register dev
Jan 19 14:28:54 openipc-hi3516ev300 kern.emerg kernel: Hisilicon Watchdog Timer: 0.01 initialized. default_margin=60 sec (nodeamon= 0)
Jan 19 14:28:54 openipc-hi3516ev300 kern.emerg kernel: hiwtdg init ok. ver=Oct 18 2019, 18:21:00.
Jan 19 14:28:54 openipc-hi3516ev300 daemon.info : starting pid 823, tty '/dev/console': '/sbin/getty -L  console 0 vt100 '
Jan 19 14:28:55 openipc-hi3516ev300 user.info majestic[822]: Majestic Lite for HiSilicon, version master+855facc, built on 2022-01-19
Jan 19 14:28:55 openipc-hi3516ev300 user.info majestic[822]: Using /etc/majestic.yaml as main configuration
Jan 19 14:28:55 openipc-hi3516ev300 user.info majestic[822]: Using /etc/sensors/5M_imx335.ini sensor configuration file
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:55 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 14:28:56 openipc-hi3516ev300 user.debug majestic[822]: Main audio codec is opus (other options: mp3, aac, pcm, alaw, ulaw)
Jan 19 14:28:56 openipc-hi3516ev300 user.info majestic[822]: linear mode
Jan 19 14:28:56 openipc-hi3516ev300 user.info majestic[822]: binit false IMX335 i2c init
Jan 19 14:28:56 openipc-hi3516ev300 user.info majestic[822]: -------Sony IMX335_init_5M_2592x1944_12bit_linear30 Initial OK!-------
Jan 19 14:28:56 openipc-hi3516ev300 user.info majestic[822]: Image tuning task started
Jan 19 14:28:56 openipc-hi3516ev300 user.info majestic[822]: HTTP server started on :::80
Jan 19 14:28:56 openipc-hi3516ev300 user.info majestic[822]: NETIP server started on port 34567
Jan 19 14:29:18 openipc-hi3516ev300 user.warn majestic[822]: Slow network for NETIP TCP transmit, skipping data
Jan 19 14:29:18 openipc-hi3516ev300 user.warn majestic[822]: Slow network for NETIP TCP transmit, skipping data
Jan 19 14:29:36 openipc-hi3516ev300 kern.err kernel: mmz_userdev:mmz_userdev_release: 
Jan 19 14:29:36 openipc-hi3516ev300 kern.emerg kernel: MMB LEAK(pid=822): 0x476FE000, 249856 bytes, 'AENC(0)StrmBuf'
Jan 19 14:29:36 openipc-hi3516ev300 kern.warn kernel: mmz_userdev:mmz_userdev_release: 
Jan 19 14:29:36 openipc-hi3516ev300 kern.emerg kernel: mmb<0x476fe000> mapped to userspace 0xb60fb000 will be unmaped!
Jan 19 14:29:36 openipc-hi3516ev300 kern.err kernel: mmz_userdev:mmz_userdev_release: 
Jan 19 14:29:36 openipc-hi3516ev300 kern.emerg kernel: MMB LEAK(pid=822): 0x4773B000, 16384 bytes, 'AENC(0) CirBuf'
Jan 19 14:29:36 openipc-hi3516ev300 kern.warn kernel: mmz_userdev:mmz_userdev_release: 
Jan 19 14:29:36 openipc-hi3516ev300 kern.emerg kernel: mmb<0x4773b000> mapped to userspace 0xb60f7000 will be unmaped!
Jan 19 14:29:49 openipc-hi3516ev300 cron.info crond[685]: time disparity of 27376709 minutes detected
Jan 19 14:30:00 openipc-hi3516ev300 cron.info crond[685]: USER root pid 829 cmd /usr/bin/logger -p cron.info -t crond WTF ?
Jan 19 14:30:00 openipc-hi3516ev300 cron.info crond: WTF ?
Jan 19 14:35:00 openipc-hi3516ev300 cron.info crond[685]: USER root pid 835 cmd /usr/bin/logger -p cron.info -t crond WTF ?
Jan 19 14:35:00 openipc-hi3516ev300 cron.info crond: WTF ?
Jan 19 14:40:00 openipc-hi3516ev300 cron.info crond[685]: USER root pid 842 cmd /usr/bin/logger -p cron.info -t crond WTF ?
Jan 19 14:40:00 openipc-hi3516ev300 cron.info crond: WTF ?
Jan 19 14:45:00 openipc-hi3516ev300 cron.info crond[685]: USER root pid 848 cmd /usr/bin/logger -p cron.info -t crond WTF ?
Jan 19 14:45:00 openipc-hi3516ev300 cron.info crond: WTF ?
Jan 19 14:50:00 openipc-hi3516ev300 cron.info crond[685]: USER root pid 854 cmd /usr/bin/logger -p cron.info -t crond WTF ?
Jan 19 14:50:00 openipc-hi3516ev300 cron.info crond: WTF ?
Jan 19 14:55:00 openipc-hi3516ev300 cron.info crond[685]: USER root pid 860 cmd /usr/bin/logger -p cron.info -t crond WTF ?
Jan 19 14:55:00 openipc-hi3516ev300 cron.info crond: WTF ?
[... cut ...]
Jan 19 15:00:46 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 15:00:46 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 15:00:46 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 15:00:46 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan 19 15:00:46 openipc-hi3516ev300 kern.err kernel: hibvt-i2c 12060000.i2c: wait idle abort!, RIS: 0x611

widgetii, 2022-01-19:
Please show `free` output on this system

widgetii, 2022-01-19:
Everything is ok with memory, so UART output is needed

esuslik, 2022-01-19:
я бы мог отправить вам камеру если вы в россии находитесь

esuslik, 2022-01-20:
Обратил внимание что видимо это не камеры перезагржаются полностью, а падает либо поток либо полностью majestic
Я неправильно читал uptime. Камры работают по 15 часов уже без перезагрузок, но на них не работает видеопоток
При включеном watchdog поток начинает работать на несколько секунд и потом падает

dimerr, 2022-01-20:
С такой температурой, скоро и провода расплавятся, что уж о сенсоре говорить

esuslik, 2022-01-20:
ну у нас стоит около 50шт на стоковой прошивке, пока проблем не было. Поэтому и хочется понять где проблема.
На стоковой к сожалению не могу посмотреть температуру, там нет такой информации

widgetii, 2022-01-20:
https://github.com/OpenIPC/camerasrnd/blob/master/docs/FAQ-ru.md#temperature

esuslik, 2022-01-20:
это же надо через uart?

esuslik, 2022-01-20:
сегодня днем без подсветки на камерах mahestic не падает. Аптайм по 4 часа,
Температура у одной камеры 104, у второй в пределах 80
Также утром обновил majestic master+8da5a71, 2022-01-19

esuslik, 2022-01-20:
после того как включилась подсветка, mahjestic начал падать через несколько секунд после запуска.

Запустил через ssh killall majestic; majestic

17:34:45 [     sdk] dump_vb_configuration@2447    VB configuration:
17:34:45 [     sdk] dump_vb_configuration@2457      [0]: 7558272 x 6
17:34:45 [     sdk] dump_vb_configuration@2457      [2]: 608256 x 1
17:34:45 [     sdk] init_sensor@2650              Sensor driver loaded
17:34:45 [    puts]                               linear mode
17:34:45 [    puts]                               binit false IMX335 i2c init
17:34:45 [    puts]                               -------Sony IMX335_init_5M_2592x1944_12bit_linear30 Initial OK!-------
17:34:45 [     sdk] log_venc_chn@1566             H.264 2592x1944 15fps 4096Kbit  15 GOP
17:34:45 [     sdk] create_vpss_chn@1319            new venc: 0   vpss_grp: 0,   vpss_chn: 0
17:34:45 [     sdk] log_venc_chn@1566             H.264 704x576 15fps 1024Kbit  15 GOP
17:34:45 [     sdk] create_vpss_chn@1319            new venc: 1   vpss_grp: 0,   vpss_chn: 1
17:34:45 [     osd] init_osd@92                   OSD initialized
17:34:45 [   audio] init_audio_in@512             Ai(0,0) bind to AencChn:0 ok!
17:34:45 [image_tu] start_image_params_tuning@63  Image tuning task started
17:34:45 [     sdk] start_sdk@827                 HiSilicon SDK started
17:34:45 [   httpd] new_http_server@346           HTTP server started on :::80
17:34:45 [   netip] netip_start@2010              NETIP server started on port 34567
17:34:50 [   netip] send_netip_video@2220         Slow network for NETIP TCP transmit, skipping data
17:34:50 [   netip] send_netip_audio@2199         Slow network for NETIP TCP transmit, skipping data
Segmentation fault

esuslik, 2022-01-20:
без подсветки работал без падений 5 часов