OpenIPC/firmware #159: Планирцется ли реализация функции подключения к vpn?

esuslik, 2022-01-18:
Было бы очень круто иметь возможность подключаться к openvpn серверу для камер которые находятся за nat или если нет статического ip адреса.

gitgayhub, 2022-01-19:
I noticed there's wireguard integrated , haven't figured out how to use yet

```
/etc/wireguard.conf
/lib/modules/3.18.20/extra/wireguard.ko
```

gitgayhub, 2022-01-19:
I have configured wireguard configure file at` /etc/wireguard.con`f, and then run `insmod /lib/modules/3.18.20/extra/wireguard.ko`

wireguard server log error message

```
[5166899.693602] wireguard: wg0: Sending keepalive packet to peer 15 ((einval))
[5166899.693622] wireguard: wg0: Sending keepalive packet to peer 14 ((einval))
[5166899.693752] wireguard: wg0: Sending handshake initiation to peer 15 ((einval))
[5166899.693814] wireguard: wg0: Sending handshake initiation to peer 14 ((einval))
[5166904.813599] wireguard: wg0: Handshake for peer 15 ((einval)) did not complete after 5 seconds, retrying (try 2)
[5166904.813601] wireguard: wg0: Handshake for peer 14 ((einval)) did not complete after 5 seconds, retrying (try 2)
[5166904.813637] wireguard: wg0: Sending handshake initiation to peer 15 ((einval))
[5166904.813748] wireguard: wg0: Sending handshake initiation to peer 14 ((einval))
[5166909.933614] wireguard: wg0: Handshake for peer 15 ((einval)) did not complete after 5 seconds, retrying (try 3)
[5166909.933660] wireguard: wg0: Sending handshake initiation to peer 15 ((einval))
[5166910.189599] wireguard: wg0: Handshake for peer 14 ((einval)) did not complete after 5 seconds, retrying (try 3)
[5166910.189635] wireguard: wg0: Sending handshake initiation to peer 14 ((einval))
[5166915.309601] wireguard: wg0: Handshake for peer 15 ((einval)) did not complete after 5 seconds, retrying (try 4)
[5166915.309640] wireguard: wg0: Sending handshake initiation to peer 15 ((einval))
[5166915.565608] wireguard: wg0: Handshake for peer 14 ((einval)) did not complete after 5 seconds, retrying (try 4)
[5166915.565650] wireguard: wg0: Sending handshake initiation to peer 14 ((einval))
[5166920.685611] wireguard: wg0: Handshake for peer 15 ((einval)) did not complete after 5 seconds, retrying (try 5)
[5166920.685678] wireguard: wg0: Sending handshake initiation to peer 15 ((einval))
[5166920.941625] wireguard: wg0: Handshake for peer 14 ((einval)) did not complete after 5 seconds, retrying (try 5)
[5166920.941712] wireguard: wg0: Sending handshake initiation to peer 14 ((einval))

```

Seems to be a key handshake issue, tried replacing different privtekey and pubkey, still the same error

gitgayhub, 2022-01-19:
Follow [this document](https://www.wireguard.com/quickstart/) to configure wireguard, I have done it successfully

widgetii, 2022-01-19:
@esuslik We have no plans to add OpenVPN to mainstream firmware because of size, but you can add it to a separate profile and build your own custom firmware

esuslik, 2022-01-19:
попробуем настроить на wireguard

closed 2022-01-19 by esuslik