OpenIPC/firmware #160: При ночном режиме картинка сиреневая

esuslik, 2022-01-18:
Камеры 000529E9, 000529B2
Также в настройках прошивки при изменении переключателя nightmode ничего не происходит. Подсветка работает всегда когда темно, нет возможности ее отключить или изменить чувствительность

widgetii, 2022-01-19:
I have no idea what's 000529E9 and 000529B2 mean

esuslik, 2022-01-19:
это модели камер из /OpenIPC/coupler
Hi3516Ev300 +IMX335/SC4236/SC2239
    000529B2
    000529D9
    000529E9
    000589C9

widgetii, 2022-01-19:
Exchange irCutPin1 and irCutPin2 values and try again.
According to backlight behavior, there is an answer in the FAQ https://github.com/OpenIPC/camerasrnd/blob/master/docs/FAQ-ru.md#ir

widgetii, 2022-01-21:
Is the issue resolved?

esuslik, 2022-01-21:
сейчас картинка не синерневая стала, видимо после обновления majestic
Но при работе с посдсветкой majestic падает
И картинка стала плохого качетсва, все в шумах

esuslik, 2022-01-21:
пример картинки скину позже

esuslik, 2022-01-21:
[image]

widgetii, 2022-01-21:
We don't support different day/night profiles currently because of the huge amount of work for each combination of SoC + sensor but you can help the project and use [PQTools](https://openipc.org/wiki/en/transfer-image-quality-tuning#ev300-family) to determine ISP settings which stock software set in different light conditions. We can replicate them as settings or profiles depending on complexity.

This is advanced work and you will need UART adapter incase something goes south

esuslik, 2022-01-21:
к сожалению у меня нет uart адаптера и навыков его использования.
При сиреновом изображении картинка была разборчивая, но просто сиреневая. Такая картинка лучше чем сейчас с шумами. 
Из помощти проекту я бы мог предоставить камеры для тестирования

esuslik, 2022-01-21:
в целом настройка nightmode никак не влияет на картинку.

esuslik, 2022-01-21:
/utils/ittb_ev300_V1.0.1.2# pqtools 
/usr/sbin/pqtools: line 9: ipctool: not found

esuslik, 2022-01-24:
Видео поток работает при свете все ок. Сразу как только выключается свет, падает majestic

esuslik, 2022-01-31:
обратил внимание что на стоковой прошивке когда включается подсветка, то камар щелкает, переключает завтор видимо.
на прошивке openipc щелчков нет

esuslik, 2022-02-11:
> use [PQTools](https://openipc.org/wiki/en/transfer-image-quality-tuning#ev300-family) to determine

ссылка больше не доступна

widgetii, 2022-02-11:
Link is available here https://openipc.org/wiki/en/image-quality-tuning.html

esuslik, 2022-02-11:
при запуске на камере:
root@openipc-hi3516ev300:~# pqtools
ipcinfo: unrecognized option: chip-id

Sorry but the 
OpenIPC is asking for your help to support development cost and long-term maintenance of what we believe will serve a fundamental role in the advancement of a stable, flexible and most importantly, Open IP Network Camera Framework for users worldwide.

Your contribution will help us advance development proposals forward, and interact with the community on a regular basis.

  https://openipc.org/contribution/

available options:
        --chip-name
        --family
        --long-sensor
        --short-sensor
        --temp
        --xm-mac
        --help is not supported at the moment...

widgetii, 2022-02-11:
We just fixed the issue, please do `sysupgrade` for the system and try again

esuslik, 2022-02-11:
все нашел, пробую обновить

esuslik, 2022-02-16:
Падение majestic при выключении освещения 100% воспроизводимая. Каждый день при отклчючении освещения он падает и поднимается по watchdog постоянно. До тех пор пока не  будет достаточного освещения. 
Когда освещения достаточно, то все работает как часы.

Также при недостаточном освещении поднимается сильно битрейт. Исправить можно только уменьшением контарста вручную, когда картинка превращается просто в серое поле одноцветное.

В настройках видеопотока не работает ограничение битрейта. Если стоит например ограничение 2048Кб, то при ночной сьемке битрейт всеравно шкалит.

esuslik, 2022-04-21:
Нужны ли логи падения majestic при недстатке освещения? Падает 100% каждый день когда не достаточно освещения

esuslik, 2022-05-19:
запустил pqtools для попытки получения корректной картинки при слабом освещении(сейчас много шумов и битрейт периодически повышается до 40мбит)
При попытке получсить параметры на стоковой прошивке получаю ошибки и не могу эти настройки перенести на oipc чтобы проверить.
На oipc данный раздел работает корректно и настройки можно править.

Раздел настроек Exposure Attr
Пример ошибок:
<HI_GetMpi>(744)--------------welcome to HI_GetMpi----------------
<OpenDllFuncEx>(150)not open dll file
<HI_HAL_ISP_GetExpsureAuto>(1453)HI_MPI_ISP_GetExposureAttr ERR:0xffffffff
<HI_ReadRegs>(308)HI_GetMpi data failed!
connection close!
connection accepted,client(10.100.0.30)!
pqtools:pq command cmd = 10 add = 0xFF011103 len = 32
pstCmdArgs->IspDev 0
================ *RegType is :1
================ *u32addr is :ff011103
<HI_GetMpi>(744)--------------welcome to HI_GetMpi----------------
<OpenDllFuncEx>(150)not open dll file
<HI_HAL_ISP_GetExpsureManual>(1513)HI_MPI_ISP_GetExposureAttr ERR:0xffffffff
<HI_ReadRegs>(308)HI_GetMpi data failed!
connection close!

widgetii, 2022-05-19:
Совпадает ли версия SDK на обоих системах?
`cat /proc/umap/vi|head -2`

esuslik, 2022-05-19:
на камере Version: [Hi3516EV200_MPP_V1.0.1.2 B030 Release], Build Time[Oct 18 2019, 18:21:00]
в pqtools выбираю такуюже

Еще при открытии pqtools выходит 3 ошибки что не удалось загрузить 
hipqautofocus.dll
hipqavsptools.dll
hipqispcalib.dll

Matlab compiler стоит нужный как по инструкции

esuslik, 2022-05-19:
с библиотеками удалось решить. Поставил все с нуля на другом пк с win 10, ошибок по ним больше нет.

Осталось разобраться как прочитать раздел настроек Exposure Attr

esuslik, 2022-05-21:
судя по экспериментам раздел настроек Exposure Attr это скорее всего то что нужно, чтобы решить проблему с шумами и завышеным битрейтом. Как-то можно всетаки их стянуть с оригинальной прошивки?

esuslik, 2022-06-18:
проблема с сиреневым цветом решена путем установки корректных значений gpio я переключения день ночь.
Шум на картинке и повышеный битрейт сохраняются.

closed 2022-06-18 by esuslik