OpenIPC/firmware #161: Не работает сеть на HI3516CV300

doctor-dark, 2022-01-18:
Во время загрузки в логах вижу ошибку:
`[    0.943271] hisi-femac 10050000.ethernet: connect to PHY failed!`

Соответственно сети нет. Загрузиться по nfs не могу
Полный лог загрузки OpenIPC
`hi3516cv300 System startup

U-Boot 2010.06 (Aug 20 2020 - 10:49:38)

Check Flash Memory Controller v100 ... Found
SPI Nand(cs 0) ID: 0x98 0xc2 Name:"TC58CVG0S3H"
Block:128KB Page:2KB Chip:128MB*1 OOB:64B ECC:8bit/512
ECC provided by Flash Memory Controller
SPI Nand total size: 128MB
MMC:
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !

EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
In:    serial
Out:   serial
Err:   serial
*No SD card found!
USB:   scanning bus for devices... 1 USB Device(s) found
0 Storage Device(s) found
Hit any key to stop autoboot:  0
hi3516cv300-vc #
hi3516cv300-vc # tftp 0x82000000 openwrt
Hisilicon ETH net controler
phyaddr=[3]
MAC:   00-00-23-34-45-66
eth0 : phy status change : LINK=DOWN : DUPLEX=FULL : SPEED=100M
eth0 : phy status change : LINK=UP : DUPLEX=FULL : SPEED=100M
TFTP from server 192.168.1.2; our IP address is 192.168.1.10
Download Filename 'openwrt'.
Download to address: 0x82000000
Downloading: #################################################
done
Bytes transferred = 1901843 (1d0513 hex)
hi3516cv300-vc # bootm 0x82000000
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   OpenIPC.org | Linux-3.18.20
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1901779 Bytes = 1.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.18.20 (runner@fv-az32-73) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 latest-178-g157e03c47e) ) #1 Sat Jul 10 19:45:12 UTC 2021
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Hisilicon Hi3516CV300 DEMO Board
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: =mem=64M console=ttyAMA0,115200 root=/dev/nfs nfsroot=192.168.1.2:/srv/nfs/openipc_rootfs ip=dhcp
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 514440K/524288K available (3595K kernel code, 169K rwdata, 1028K rodata, 156K init, 246K bss, 9848K reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[... 361 lines cut ...]
load vpss.ko for Hi3516CV300...OK!
load vou.ko for Hi3516CV300...OK!
load rc.ko ...OK!
load venc.ko ...OK!
load chnl.ko for Hi3516CV300...OK!
load vedu.ko ...OK!
load h264e.ko ...OK!
load h265e.ko ...OK!
load jpege.ko ...OK!
load ive.ko for Hi3516CV300...OK!
load hi_piris.ko OK!
load ai.ko OK!
load ao.ko OK!
load aenc.ko OK!
load adec.ko OK!
load acodec.ko for Hi3516CV300...OK!
insert audio
load hi_mipi driver successful!
Load hi_cipher.ko success. 
hirtc init ok. ver=Aug  5 2021, 13:02:42.
Hisilicon Watchdog Timer: 0.01 initialized. default_margin=60 sec (nodeamon= 1)
hiwtdg init ok. ver=Aug  5 2021, 13:02:43.
==== Your input Sensor type is imx307 ====
`
В чём может быть проблема? Заранее спасибо!

doctor-dark, 2022-01-27:
Проблема состоит в том что для phy в dts жёстко указан адрес, в моём случае некорректный. Решением послужило удаление строки reg= <1> из /arch/arm/boot/dts/, что позволило ядру автоматически искать корректный адрес phy

dimerr, 2022-02-03:
Fixed. f8d2d2de90ec470d58c100285b0c1c0f11a810a2

closed 2022-02-03 by dimerr