OpenIPC/firmware #167: ./autogen.sh: 18: aclocal: not found when building for XM530

shotreal, 2022-01-25:
First of all thanks for all the work that has gone into this project.

I have two Besder cameras and as the webinterface is shit I opened them to see if i could gain access.
Both are using the XM530 cpu and I gained access to the bootloader via spi.

I thought it might be a good idea to flash a decent operating system and found this project.
After getting familiar with buildroot I modified the building.sh to build openIPC for the xm530 ... unfortunately it crashes with the below error. Anny suggestions?

Since I'm already here, could you give some more details on the status of the development for the xm530? Somewhere I read that video is not working. Is there anything i can do to support? 

[...]
Applying libevent-mbedtls.patch using patch: 
patching file buffer.c
Hunk #1 succeeded at 3080 (offset 10 lines).
patching file bufferevent_mbedtls.c
>>> libevent-openipc 2db55e43cffd2fb6a34b5c997f2b8d043ca0ece5 Updating config.sub and config.guess
for file in config.guess config.sub; do for i in $(find /home/oberon/ipcamhack/firmware/output/build/libevent-openipc-2db55e43cffd2fb6a34b5c997f2b8d043ca0ece5 -name $file); do cp support/gnuconfig/$file $i; done; done
>>> libevent-openipc 2db55e43cffd2fb6a34b5c997f2b8d043ca0ece5 Patching libtool
>>> libevent-openipc 2db55e43cffd2fb6a34b5c997f2b8d043ca0ece5 Configuring
mkdir -p /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/host-pkgconf/host/ /home/oberon/ipcamhack/firmware/output/per-package/host-pkgconf/host/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/host-skeleton/host/ /home/oberon/ipcamhack/firmware/output/per-package/host-skeleton/host/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/mbedtls-openipc/host/ /home/oberon/ipcamhack/firmware/output/per-package/mbedtls-openipc/host/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/skeleton/host/ /home/oberon/ipcamhack/firmware/output/per-package/skeleton/host/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/toolchain/host/ /home/oberon/ipcamhack/firmware/output/per-package/toolchain/host/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host
mkdir -p /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/target
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/host-pkgconf/target/ /home/oberon/ipcamhack/firmware/output/per-package/host-pkgconf/target/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/target
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/host-skeleton/target/ /home/oberon/ipcamhack/firmware/output/per-package/host-skeleton/target/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/target
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/mbedtls-openipc/target/ /home/oberon/ipcamhack/firmware/output/per-package/mbedtls-openipc/target/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/target
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/skeleton/target/ /home/oberon/ipcamhack/firmware/output/per-package/skeleton/target/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/target
rsync -a --link-dest=/home/oberon/ipcamhack/firmware/output/per-package/toolchain/target/ /home/oberon/ipcamhack/firmware/output/per-package/toolchain/target/ /home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/target
cd /home/oberon/ipcamhack/firmware/output/build/libevent-openipc-2db55e43cffd2fb6a34b5c997f2b8d043ca0ece5 && PATH="/home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host/bin:/home/oberon/ipcamhack/firmware/output/per-package/libevent-openipc/host/sbin:/usr/local/cuda-10.1/bin:/home/oberon/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" ./autogen.sh
./autogen.sh: 18: aclocal: not found

shotreal, 2022-01-25:
Well, I just found this https://kerneltalks.com/troubleshooting/how-to-resolve-aclocal-not-found-error-in-ubuntu/
Apparently i am missing autotools-dev and automake.
started a new build ... lets see if this solves my problem.

shotreal, 2022-01-25:
When setting the u boot environment prior to installation i get a too many args err. 

U-Boot> setenv uk 'mw.b 0x80007fc0 ff 1000000; tftp 0x80007fc0 uImage. && sf probe 0; sf erase 0x50000 0x200000' 
** Too many args (max. 16) **

widgetii, 2022-01-27:
These are constraints for specific U-Boot build, please use [Coupler](https://github.com/openipc/coupler) to avoid the issue