OpenIPC/firmware #182: Mi Home Security Camera 1080p SXJ01ZM support (GM8136)

testuser7, 2022-02-21:
Hi,

Is there a support for wireless module used in this camera (88W8977: https://www.nxp.com/products/wireless/wi-fi-plus-bluetooth/2-4-5-ghz-dual-band-1x1-wi-fi-4-802-11n-plus-bluetooth-5-2-solution:88W8977) in your firmware?

ZigFisher, 2022-06-10:
GM8136 now supported in TestMode

testuser7, 2023-03-06:
I'm trying to run OpenIPC firmware on this device but no success.

```
## Booting kernel from Legacy Image at 02000000 ...
   Image Name:   Linux-3.3.0-gm8136
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1411312 Bytes = 1.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Loading Kernel Image ... OK
OK
ft_mode:0
: mem=128M gmmem=90M console=ttyS0,115200 user_debug=31 init=/squashfs_init rootwait root=/dev/mmcblk0p2 rootfstype=squashfs

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0
Linux version 3.3.0 (runner@fv-az180-704) (gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty) ) #2 PREEMPT Sun Mar 5 18:38:12 UTC 2023
CPU: FA6 [66056263] revision 3 (ARMv5TE), cr=0000397f
CPU VIPT aliasing data cache, VIPT aliasing instruction cache
Machine: Grain-Media GM8136 series
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
Kernel command line: mem=128M gmmem=90M console=ttyS0,115200 user_debug=31 init=/squashfs_init rootwait root=/dev/mmcblk0p2 rootfstype=squashfs
PID hash table entries: 512 (order: -1, 2048 bytes)
Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Memory: 128MB = 128MB total
Memory: 126068k/126068k available, 5004k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0x88800000 - 0xff000000   (1896 MB)
    lowmem  : 0x80000000 - 0x88000000   ( 128 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x803759f4   (3511 kB)
      .init : 0x80376000 - 0x80392000   ( 112 kB)
      .data : 0x80392000 - 0x803ae8c0   ( 115 kB)
       .bss : 0x803ae8e4 - 0x803beed0   (  66 kB)
NR_IRQS:64
gm_jiffies_init, system HZ: 100, pClk: 100000000 
console [ttyS0] enabled
Calibrating delay loop... 858.52 BogoMIPS (lpj=4292608)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x2a1488 - 0x2a14d0
devtmpfs: initialized
FMEM: 23040 pages(0x5a00000 bytes) from bank0 are reserved for Frammap. 
FMEM: Logical memory ends up at 0x88000000, init_mm:0x80004000(0x4000), PAGE_OFFSET:0x80000000(0x0), 
FMEM: FA726 Test and Debug Register: 0x0 
NET: Registered protocol family 16
FMEM Idle Process Up. 
PMU: Mapped at 0xfe000000 
IC: GM8136, version: 0x1 
iotable: VA: 0xfe000000, PA: 0x90c00000, Length: 4096 
iotable: VA: 0xfe001000, PA: 0x90700000, Length: 4096 
iotable: VA: 0xfe002000, PA: 0x90800000, Length: 4096 
iotable: VA: 0xfe003000, PA: 0x90900000, Length: 4096 
iotable: VA: 0xfe004000, PA: 0x90d00000, Length: 4096 
iotable: VA: 0xfe005000, PA: 0x96000000, Length: 4096 
bio: create slab <bio-0> at 0
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource fttmr010:1
NET: Registered protocol family 2
IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
TCP established hash table entries: 4096 (order: 3, 32768 bytes)
[... 108 lines cut ...]
grainmedia: Loading of kernel modules and initialization of the video system has started
Cannot parse config file '/etc/fw_env.config': No such file or directory
Cannot parse config file '/etc/fw_env.config': No such file or directory
grainmedia: Get data from ipcinfo and set SENSOR as unknown
Cannot parse config file '/etc/fw_env.config': No such file or directory
Error: environment not initialized
----------------------------------------------------------------
ddr name: frammap0 
base: 0x1800000 
end: 0x7200000 
size: 0x5a00000 bytes
memory allocated: 0x0 bytes 
memory free: 0x5a00000 bytes 
max available slice: 0x5a00000 bytes
memory allocate count: 0 
clear address: 0x1800000 
dirty pages: 0 
clear pages: 23040 
size alignment: 0x1000 
Invalid argument for video frontend: unknown
Starting majestic: OK

Welcome to OpenIPC
openipc-gm8136 login: 
```

FlyRouter, 2023-03-07:
```
setenv ipaddr 192.168.1.51; setenv serverip 192.168.1.72; saveenv
setenv bootargs 'mem=128M gmmem=90M console=ttyS0,115200 user_debug=31 root=/dev/mtdblock3 init=/init rootfstype=squashfs mtdparts=nor-flash:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)'; saveenv

setenv bootcmd 'setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read 0x02000000 0x50000 0x200000; bootm 0x02000000'; saveenv
setenv uk 'mw.b 0x02000000 ff 1000000; tftp 0x02000000 uImage.${soc} && sf probe 0; sf erase 0x50000 0x200000; sf write 0x02000000 0x50000 ${filesize}'; saveenv
setenv ur 'mw.b 0x02000000 ff 1000000; tftp 0x02000000 rootfs.squashfs.${soc} && sf probe 0; sf erase 0x250000 0x500000; sf write 0x02000000 0x250000 ${filesize}'; saveenv

```

testuser7, 2023-03-07:
Thanks for the flashing instructions. But the problem is that there is no WiFi driver (88W8977).