OpenIPC/firmware #200: Wrong color on HI3518ev100

fennec622, 2022-03-12:
Hi 

someone can help me to configure my camera ESCAM QD300 with 

`board:
  vendor: OpenIPC
  version: 2.2.02.16
chip:
  vendor: HiSilicon
  model: 3518EV100
ethernet:
  mac: "00:12:15:4c:d6:41"
  u-mdio-phyaddr: 1
  phy-id: 0x02430c54
  d-mdio-phyaddr: 0
  phy-mode: mii
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 9a7f592c
        contains:
          - name: uboot-env
            offset: 0x30000
      - name: wtf
        size: 0x10000
        sha1: d12e61c5
      - name: kernel
        size: 0x200000
        sha1: af194ffa
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: 0fe0850d
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
    addr-mode: 3-byte
ram:
  total: 64M
  media: 32M
firmware:
  kernel: "3.0.8 (Wed Feb 16 03:51:47 UTC 2022)"
  toolchain: gcc version 7.5.0 (Buildroot -gfe5fb71)
  sdk: "Hi3518_MPP_V1.0.B.0  (Nov 19 2015, 16:37:04)"
  main-app: /usr/bin/majestic
Error: unexpected value for SOI == 0x9711
sensors:
- vendor: OmniVision
  model: OV9712
  control:
    bus: 0
    type: i2c
    addr: 0x60
  data:
    type: DC
  clock: 24MHz`

`muxctrl_reg0 0x200f0000 0x1 GPIO1_0 [SHUTTER_TRIG]
muxctrl_reg1 0x200f0004 0x1 GPIO1_1 [SDIO_CCLK_OUT]
muxctrl_reg2 0x200f0008 0x1 GPIO1_2 [SENSOR_CLK]
muxctrl_reg3 0x200f000c 0 [GPIO1_4] SPI0_SCLK
muxctrl_reg4 0x200f0010 0 [GPIO1_5] SPI0_SDO
muxctrl_reg5 0x200f0014 0 [GPIO1_6] SPI0_SDI
muxctrl_reg6 0x200f0018 0x1 GPIO2_0 [I2C_SDA]
muxctrl_reg7 0x200f001c 0x1 GPIO2_1 [I2C_SCL]
muxctrl_reg8 0x200f0020 0x1 GPIO2_2 [UART1_RTSN]
muxctrl_reg9 0x200f0024 0x1 GPIO2_3 [UART1_RXD]
muxctrl_reg10 0x200f0028 0x1 GPIO2_4 [UART1_CTSN]
muxctrl_reg11 0x200f002c 0x1 GPIO2_5 [UART1_TXD]
muxctrl_reg12 0x200f0030 0x1 GPIO3_0 [MII_CRS] VOU1120_DATA10
muxctrl_reg13 0x200f0034 0x1 GPIO3_1 [MII_COL] VOU1120_DATA9
muxctrl_reg14 0x200f0038 0x1 GPIO4_3 [MII_RXD3] VOU1120_DATA15
muxctrl_reg15 0x200f003c 0x1 GPIO4_2 [MII_RXD2] VOU1120_DATA11
muxctrl_reg16 0x200f0040 0x1 GPIO4_1 [MII_RXD1] VOU1120_DATA8
muxctrl_reg17 0x200f0044 0x1 GPIO4_0 [MII_RXD0] VOU1120_DATA12
muxctrl_reg18 0x200f0048 0x1 GPIO4_7 [MII_TXD3] VOU1120_DATA3
muxctrl_reg19 0x200f004c 0x1 GPIO4_6 [MII_TXD2] VOU1120_DATA13
muxctrl_reg20 0x200f0050 0x1 GPIO4_5 [MII_TXD1] VOU1120_DATA0
muxctrl_reg21 0x200f0054 0x1 GPIO4_4 [MII_TXD0] VOU1120_DATA4
muxctrl_reg22 0x200f0058 0x1 GPIO3_2 [MII_RXCK] VOU1120_CLK
muxctrl_reg23 0x200f005c 0x3 GPIO3_3 MII_TXCK VOU1120_DATA7 [RMII_CLK]
muxctrl_reg24 0x200f0060 0x1 GPIO3_4 [MII_RXDV] VOU1120_DATA1
muxctrl_reg25 0x200f0064 0x1 GPIO3_5 [MII_TXEN] VOU1120_DATA5
muxctrl_reg26 0x200f0068 0x1 GPIO2_6 [MII_TXER]
muxctrl_reg27 0x200f006c 0x1 GPIO2_7 [MII_RXER]
muxctrl_reg28 0x200f0070 0 [GPIO1_3] EPHY_CLK VOU1120_DATA2
muxctrl_reg29 0x200f0074 0x1 GPIO3_6 [MDCK] VOU1120_DATA6 BOOT_SEL
muxctrl_reg30 0x200f0078 0x1 GPIO3_7 [MDIO] VOU1120_DATA14
muxctrl_reg31 0x200f007c 0x1 GPIO1_7 [FLASH_TRIG]
muxctrl_reg32 0x200f0080 0x1 GPIO6_0 [SDIO_CARD_DETECT]
muxctrl_reg33 0x200f0084 0x1 GPIO6_1 [SDIO_CARD_POWER_EN]
muxctrl_reg34 0x200f0088 0x1 GPIO6_2 [SDIO_CWPR]
muxctrl_reg35 0x200f008c 0x1 GPIO6_3 [SDIO_CCMD]
muxctrl_reg36 0x200f0090 0x1 GPIO6_4 [SDIO_CDATA0] CLK_TEST_OUT0 CLK_TEST_OUT1 CLK_TEST_OUT2 CLK_TEST_OUT3
muxctrl_reg37 0x200f0094 0x1 PLL_TEST_OUT0 [SDIO_CDATA1] GPIO6_5 PLL_TEST_OUT1 PLL_TEST_OUT2 PLL_TEST_OUT3 RTC_TEST_CLK
muxctrl_reg38 0x200f0098 0x1 GPIO6_6 [SDIO_CDATA2]
muxctrl_reg39 0x200f009c 0x1 GPIO6_7 [SDIO_CDATA3]
muxctrl_reg40 0x200f00a0 0 [SFC_DIO] GPIO7_0
muxctrl_reg41 0x200f00a4 0 [SFC_WP_IO2] GPIO7_1
muxctrl_reg42 0x200f00a8 0 [SFC_CLK] GPIO7_2 SFC_ADDR_MODE
muxctrl_reg43 0x200f00ac 0 [SFC_DOI] GPIO7_3
muxctrl_reg44 0x200f00b0 0 [SFC_HOLD_IO3] GPIO7_4
muxctrl_reg45 0x200f00b4 0x1 GPIO5_0 [USB_OVRCUR]
muxctrl_reg46 0x200f00b8 0 [GPIO5_1] USB_PWREN
muxctrl_reg47 0x200f00bc 0x1 GPIO5_2 [PWM_OUT0]
muxctrl_reg48 0x200f00c0 0 [GPIO5_3] PWM_OUT1
muxctrl_reg49 0x200f00c4 0x1 IR_IN [GPIO7_5]
muxctrl_reg50 0x200f00c8 0x1 reserved [GPIO9_0]
muxctrl_reg51 0x200f00cc 0x1 reserved [GPIO9_1]
muxctrl_reg52 0x200f00d0 0x1 reserved [GPIO9_2]
muxctrl_reg53 0x200f00d4 0 [reserved] GPIO9_3
muxctrl_reg54 0x200f00d8 0x1 reserved [GPIO9_4]
muxctrl_reg55 0x200f00dc 0x1 reserved [GPIO9_5]
muxctrl_reg56 0x200f00e0 0x1 reserved [GPIO9_6]
muxctrl_reg57 0x200f00e4 0x1 reserved [GPIO9_7]
muxctrl_reg58 0x200f0108 0 [GPIO7_6] UART2_RXD
muxctrl_reg59 0x200f010c 0 [GPIO7_7] UART2_TXD
muxctrl_reg60 0x200f0110 0 [GPIO5_4] SPI1_SCLK
muxctrl_reg61 0x200f0114 0 [GPIO5_5] SPI1_SDO
muxctrl_reg62 0x200f0118 0 [GPIO5_6] SPI1_SDI
muxctrl_reg63 0x200f011c 0 [GPIO5_7] SPI1_CSN
muxctrl_reg64 0x200f0120 0 [GPIO0_0] JTAG_TRSTN TEMPER_DQ
muxctrl_reg65 0x200f0124 0 [GPIO0_1] JTAG_TCK TEMPER_DQ
muxctrl_reg66 0x200f0128 0 [GPIO0_2] JTAG_TMS TEMPER_DQ
muxctrl_reg67 0x200f012c 0 [GPIO0_3] JTAG_TDO TEMPER_DQ
muxctrl_reg68 0x200f0130 0 [GPIO0_4] JTAG_TDI TEMPER_DQ
muxctrl_reg69 0x200f0134 0x1 SVB_PWM [GPIO0_5] TEMPER_DQ
muxctrl_reg70 0x200f0138 0x1 GPIO0_6 [SVB_PWM] TEMPER_DQ
muxctrl_reg71 0x200f013c 0 [SYS_RSTN_OUT] GPIO0_7 TEMPER_DQ
muxctrl_reg72 0x200f0140 0 [VIU_CLK] GPIO11_6
muxctrl_reg73 0x200f0144 0 [VIU_VS] GPIO11_5
muxctrl_reg74 0x200f0148 0 [VIU_HS] GPIO11_4
muxctrl_reg75 0x200f014c 0 [VIU_DAT11] GPIO11_3
muxctrl_reg76 0x200f0150 0 [VIU_DAT10] GPIO11_2
muxctrl_reg77 0x200f0154 0 [VIU_DAT9] GPIO11_1
muxctrl_reg78 0x200f0158 0 [VIU_DAT8] GPIO11_0
muxctrl_reg79 0x200f015c 0 [VIU_DAT7] GPIO10_7
muxctrl_reg80 0x200f0160 0 [VIU_DAT6] GPIO10_6
muxctrl_reg81 0x200f0164 0 [VIU_DAT5] GPIO10_5
muxctrl_reg82 0x200f0168 0 [VIU_DAT4] GPIO10_4
muxctrl_reg83 0x200f016c 0 [VIU_DAT3] GPIO10_3
muxctrl_reg84 0x200f0170 0 [VIU_DAT2] GPIO10_2
muxctrl_reg85 0x200f0174 0 [VIU_DAT1] GPIO10_1
muxctrl_reg86 0x200f0178 0 [VIU_DAT0] GPIO10_0`

`system:
  buffer: 1024
image:
  mirror: false
  flip: false
  rotate: none
  contrast: auto
  hue: 50
  saturation: 50
  luminance: auto
osd:
  enabled: true
  template: "%d/%m/%y %H:%M:%S"
  posX: 0
  posY: 0
nightMode:
  enabled: true
  irSensorPinInvert: false
  irCutPin1: 1
  irCutPin2: 0
  pinSwitchDelayUs: 150
  irSensorPin: 75
records:
  enabled: false
  path: /mnt/mmc/%Y/%m/%d/%H.mp4
  maxUsage: 95
video0:
  enabled: true
  size: 1280x720
  fps: 15
  gopSize: 4
  codec: h264
  bitrate: 1500
  rcMode: avbr
video1:
  enabled: false
jpeg:
  enabled: true
  toProgressive: false
  size: 1280x720
mjpeg:
  size: 640x360
  fps: 5
  bitrate: 1024
audio:
  enabled: false
  volume: auto
  srate: 8000
rtsp:
  enabled: true
  port: 554
hls:
  enabled: true
youtube:
  enabled: false
motionDetect:
  enabled: false
  profile: outdoor
  visualize: true
  debug: true
ipeye:
  enabled: false
onvif:
  enabled: false
watchdog:
  enabled: true
  timeout: 20
isp:
  exposure: auto
  slowShutter: low
  antiFlicker: disabled
  memMode: normal
  rawMode: none
  sensorConfig: /etc/sensors/ov9712_i2c_dc_720p.ini
netip:
  enabled: false
  ignoreSetTime: false
  snapshots: false
cloud:
  enabled: false`

I have always bad color

thanks for your help
[image]

dimerr, 2022-03-13:
swap ircut pins in cfg

fennec622, 2022-03-13:
> swap ircut pins in cfg

Thanks for your help do you know how can i find GPIO number for sensors

thanks

widgetii, 2022-03-13:
Run `ipctool gpio scan` on original firmware

fennec622, 2022-03-13:
> Run `ipctool gpio scan` on original firmware

Thanks for your help

Original firmware

# ./ipctool gpio scan
Gr: 0, Addr:0x201400FC, Data:0x2F = 0bxx101111, Addr:0x20140400, Dir:0x20 = 0bxx100000
Gr: 1, Addr:0x201501E0, Data:0x08 = 0bx0001xxx, Addr:0x20150400, Dir:0x28 = 0bx0101xxx
Gr: 2, Addr:0x201600A0, Data:0x28 = 0bxx1x1xxx, Addr:0x20160400, Dir:0x28 = 0bxx1x1xxx
Gr: 3, Addr:0x20170014, Data:0x00 = 0bxxxxx0x0, Addr:0x20170400, Dir:0x04 = 0bxxxxx1x0
Gr: 4, Addr:0x20180300, Data:0x00 = 0b00xxxxxx, Addr:0x20180400, Dir:0xC0 = 0b11xxxxxx
Gr: 5, Addr:0x201903F8, Data:0x00 = 0b0000000x, Addr:0x20190400, Dir:0x0A = 0b0000101x
Gr: 6, Addr:0x201A000C, Data:0x03 = 0bxxxxxx11, Addr:0x201A0400, Dir:0x00 = 0bxxxxxx00
Gr: 7, Addr:0x201B0380, Data:0x60 = 0b011xxxxx, Addr:0x201B0400, Dir:0x00 = 0b000xxxxx
Gr: 8, Addr:0x201C0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201C0400, Dir:0x00 = 0bxxxxxxxx
Gr: 9, Addr:0x201D03FC, Data:0x00 = 0b00000000, Addr:0x201D0400, Dir:0x08 = 0b00001000
Gr:10, Addr:0x201E0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201E0400, Dir:0x00 = 0bxxxxxxxx
Gr:11, Addr:0x201F0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201F0400, Dir:0x00 = 0bxxxxxxxx
======================================================================================
Waiting for while something changes...
^C
#
And openipc

root@openipc-hi3518ev100:~# ipctool gpio scan
Gr: 0, Addr:0x201400FC, Data:0x2F = 0bxx101111, Addr:0x20140400, Dir:0x20 = 0bxx100000
Gr: 1, Addr:0x201501E0, Data:0x08 = 0bx0001xxx, Addr:0x20150400, Dir:0x00 = 0bx0000xxx
Gr: 2, Addr:0x20160000, Data:0x00 = 0bxxxxxxxx, Addr:0x20160400, Dir:0x00 = 0bxxxxxxxx
Gr: 3, Addr:0x20170000, Data:0x00 = 0bxxxxxxxx, Addr:0x20170400, Dir:0x00 = 0bxxxxxxxx
Gr: 4, Addr:0x20180000, Data:0x00 = 0bxxxxxxxx, Addr:0x20180400, Dir:0x00 = 0bxxxxxxxx
Gr: 5, Addr:0x201903E8, Data:0x00 = 0b00000x0x, Addr:0x20190400, Dir:0x00 = 0b00000x0x
Gr: 6, Addr:0x201A0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201A0400, Dir:0x00 = 0bxxxxxxxx
Gr: 7, Addr:0x201B0380, Data:0x60 = 0b011xxxxx, Addr:0x201B0400, Dir:0x00 = 0b000xxxxx
Gr: 8, Addr:0x201C0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201C0400, Dir:0x03 = 0bxxxxxxxx
Gr: 9, Addr:0x201D03DC, Data:0x00 = 0b0000x000, Addr:0x201D0400, Dir:0x00 = 0b0000x000
Gr:10, Addr:0x201E0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201E0400, Dir:0x00 = 0bxxxxxxxx
Gr:11, Addr:0x201F0000, Data:0x00 = 0bxxxxxxxx, Addr:0x201F0400, Dir:0x00 = 0bxxxxxxxx
======================================================================================
Waiting for while something changes...
^C
root@openipc-hi3518ev100:~#

Now how can i do for change in openipc ? thanks

cronyx, 2022-03-17:
> 

You need to run **./ipctool gpio scan** and simulate the change of day and night by closing and opening the IR sensor, then check what ipctool gives out

cronyx, 2022-03-17:
> GPIO24

irSensorPin: 24
irCutPin1: 38
irCutPin1: 39

try this, you may have to swap ircutpin1 and ircutpin2

fennec622, 2022-03-17:
> > @cronyx
> > thanks a lot !!!! color is good
> > last question the night or day mode is automatique ? if not do you know how can I have value of lux or something else to know the night
> > Again thanks
> 
> As far as I know there are no additional settings, the switching is fully automatic

No automatic 
How can i find this gpio ?

GPIO pin to turn on night mode illumination

[image]

fennec622, 2022-03-19:
thanks for your help all is ok 
auto night mode with this setup 
ESCAM QD300 HI3518EV100

[image]

closed 2022-03-19 by fennec622