OpenIPC/firmware #201: IMX327 support for Hi3516AV200

gitgayhub, 2022-03-14:
Hi, thanks for the great project again

I'm wondering if you can add support for imx327 on hi3516av200 platform. I've got a camera with telnet access to stock firmware so I already got the libsns file but don't know how to write the config file
[imx327.zip](https://github.com/OpenIPC/firmware/files/8246696/imx327.zip)

I can arrange remote telnet access if needed

Thank you very much

ipctool inside stock firmware
```
/mnt/nfs # ./ipctool
---
chip:
  vendor: HiSilicon
  model: 3516AV200
ethernet:
  mac: "xxxxxx"
  phy-mode: rmii
rom:
  - type: nand
    block: 256K
    partitions:
      - name: uboot
        size: 0x80000
        sha1: 83c5691d
      - name: ubootenv
        size: 0x80000
        sha1: e5bc57ab
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel
        size: 0x400000
        sha1: 2a81aa03
      - name: rootfs
        size: 0x2000000
        path: /,yaffs2
        sha1: f7cd3716
      - name: appfs
        size: 0x6400000
        path: /home,yaffs2,rw
      - name: datafs
        size: 0x17700000
        path: /data_fs,yaffs2,rw
    size: 512M
ram:
  total: 512M
  media: 256M
firmware:
  u-boot: "2010.06 (Jan 10 2019 - 12:48:22)"
  kernel: "3.18.20 (SMP Tue Dec 4 17:45:49 CST 2018)"
  toolchain: gcc version 4.9.4 20150629 (prerelease) (Hisilicon_v600_20170615)
  sdk: "Hi3519V101_MPP_V2.0.4.0 B040 Release (Nov 10 2017, 14:28:05)"
  main-app: ./face
sensors:
- vendor: Sony
  model: IMX327
  control:
    bus: 0
    type: spi
  vicap0-state: up
  vicap1-state: down
  data:
    vicap0-input: MIPI0
    mipi0-type: MIPI
```

ipctool inside openipc:

```
root@openipc-hi3516av200:/etc/sensors# ipctool

The ipctool installed as remote GitHub plugin

---
board:
  vendor: OpenIPC
  version: 2.2.03.14
chip:
  vendor: HiSilicon
  model: 3516AV200
ethernet:
  mac: "00:00:23:34:45:66"
  phy-mode: rmii
rom:
  - type: nand
    block: 256K
    partitions:
      - name: boot
        size: 0x80000
        sha1: 83c5691d
      - name: ubootenv
        size: 0x80000
        sha1: ccbef8e6
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel_old
        size: 0x400000
        sha1: 2a81aa03
      - name: rootfs_old
        size: 0x2000000
        sha1: ffe4896d
      - name: appfs
        size: 0x6400000
        sha1: 2c1800fc
      - name: kernel
        size: 0x200000
        sha1: 6645c1c6
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: e7d58ee8
      - name: rootfs_data
        size: 0x500000
        path: /overlay,jffs2,rw
      - name: others
        size: 0x16b00000
        sha1: 3faf6123
    size: 512M
ram:
  total: 256M
firmware:
  u-boot: "2010.06 (Jan 10 2019 - 12:48:22)"
  kernel: "3.18.20 (SMP Mon Mar 14 03:45:33 UTC 2022)"
  toolchain: gcc version 7.5.0 (Buildroot -gaa4bdd6)

```

dimerr, 2022-03-15:
Post contents of 
/proc/umap/vi
/proc/umap/hi_mipi

from stock firmware

gitgayhub, 2022-03-15:
```
~ # cat /proc/umap/vi

[VIU] Version: [Hi3519V101_MPP_V2.0.4.0 B040 Release], Build Time: [Nov 10 2017, 14:28:18]

-----MODULE PARAM---------------------------------------------------------------
detect_err_frame  drop_err_frame   stop_int_level    yuv_skip
              10               0                0           0

-----VI-VPSS ONLINE STATE-------------------------------------------------------
     OnlineState
         offline

-----VI MODULE STATE------------------------------------------------------------
     ModuleState
         Started

-----VI PORT SIZE---------------------------------------------------------------
     Dev   WIDTH  HEIGHT
       0    1920    1080

-----VI DEV ATTR1---------------------------------------------------------------
     Dev   IntfM     WkM     ComMsk0     ComMsk1   ScanM     AD0     AD1     AD2
       0    MIPI    1Mux    fff00000           0       P      -1      -1      -1

-----VI DEV ATTR2---------------------------------------------------------------
     AD3     Seq   DPath   DType    DRev    CapX    CapY    CapW    CapH  BasW  BasH  Comp HReph VReph
      -1     N/A     ISP     RGB       N       0       0    1920    1080  1920  1080     N  NONE  NONE

-----VI HIGH DEV ATTR 1---------------------------------------------------------
     Dev  InputM     WkM     ComMsk0     ComMsk1   ScanM     AD0     AD1     AD2

-----VI HIGH DEV ATTR 2---------------------------------------------------------
     AD3     Seq   CombM   CompM    ClkM     Fix    FldP   DPath   DType    DRev

-----VI HIGH DEV ATTR 3---------------------------------------------------------
    CapX    CapY    CapW    CapH  BasW  BasH  Comp HReph VReph

-----VI PHYCHN ATTR 1-----------------------------------------------------------
  PhyChn    CapX    CapY    CapW    CapH    DstW    DstH  CapSel  Mirror    Flip
       0       0       0    1920    1080    1920    1080    both       N       N

-----VI PHYCHN ATTR 2-----------------------------------------------------------
   IntEn  PixFom  SrcRat  DstRat    Comp
       Y   SP420      -1      -1       N

-----VI PHYCHN STATUS 1---------------------------------------------------------
  PhyChn     Dev      IntCnt  VbFail      LosInt  TopLos  BotLos  BufCnt    IntT
       0       0        1784       0           2       0       2       2     230

-----VI PHYCHN STATUS 2---------------------------------------------------------
   SendT   Field  Stride MaxIntT     IntGapT     MaxGapT LIntCnt  ThrCnt AutoDis
       0     frm    1920    2151       40038       40618       0       1       0

-----VI PHYCHN STATUS 3---------------------------------------------------------
 CasAutD  TmgErr      ccErrN      IntRat
       0       0           0          25

-----VI LDC ATTR----------------------------------------------------------------
 PhyChn  ViewType  Ratio minRatio  COffX  COffY Enable

-----VI Cylind Correction ATTR------------------------------------------------------------
 PhyChn  Enable   Ratio  CYOffX  CYOffY

-----VI SPREAD ATTR------------------------------------------------------------
 PhyChn  Enable    OutW    OutH    Coef

-----VI ROTATE_EX ATTR------------------------------------------------------------
 PhyChn Enable ViewType  Angle  CYOffX  CYOffY    OutW    OutH

[... 273 lines cut ...]
     0  1       0       0
     0  2       0       0
     0  3       0       0

-----fsm timeout and escape info---------------------------------------------
 link clkTOutCnt d0TOutCnt   d1TOutCnt   d2TOutCnt   d3TOutCnt clkEscCnt  d0EscCnt  d1EscCnt  d2EscCnt  d3EscCnt
    0     0           0           0           0           0         0         0         0         0         0
    1     0           0           0           0           0         0         0         0         0         0
    2     0           0           0           0           0         0         0         0         0         0
-----MIPI int error info-----------------------------------------------------------

 Devno vc0CRC vc1CRC vc2CRC vc3CRC vc0OrderErr vc1OrderErr vc2OrderErr vc3OrderErr vc0NMatCnt vc1NMatCnt vc2NMatCnt vc3NMatCnt
     0      0      0      0      0           0           0           0           0          0          0          0          0

 Devno HCntErr vc0HECC vc1HECC vc2HECC vc3HECC vc0DtErr vc1DtErr vc2DtErr vc3DtErr
     0       0       0       0       0       0        0        0        0        0

 Devno CMD_FIFO_RERR DATA_FIFO_RERR CMD_FIFO_WERR DATA_FIFO_WERR
     0             0              0             0              0

-----ALING Err info--------------------------------------
 Devno FIFO_FullErr  Lane0Err  Lane1Err  Lane2Err  Lane3Err  Lane4Err  Lane5Err  Lane6Err  Lane7Err  Lane8Err  Lane9Err Lane10Err Lane11Err
     0            0         0         0         0         0         0         0         0         0         0         0         0         0
~ #
```

dimerr, 2022-03-15:
I'd say - just copy imx385 config and change .so name inside

then edit /usr/bin/load_hisilicon

https://github.com/OpenIPC/firmware/blob/master/general/package/hisilicon-osdrv-hi3519v101/files/script/load_hisilicon#L450

change to ```spi0_4wire_pin_mux```
then
```fw_setenv sensor imx327```

reboot and try

gitgayhub, 2022-03-15:
Thanks, I made the changes above, boot process will stuck at :

```
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0x82000000, mmz_size: 32M
hisilicon: Get data from environment and set SENSOR as imx327
++++++++++++++++++++++++++++++++++++++++++++++
==============vi_vpss_online==============
Segmentation fault
Segmentation fault
r

insmod: can't insert 'hi3519v101_vpss.ko': Resource busy
```
Can't get into shell environment. I logged in via ssh, majestic is not running, manually running "SENSOR=imx327 majestic give the following errors:

```
root@openipc-hi3516av200:~# SENSOR=imx327 majestic
07:22:08 [    main] main@135                      Majestic Lite for HiSilicon, version master+912fa6f, built on 2022-03-08
07:22:08 [app_conf] load_config@345               Using /etc/majestic.yaml as main configuration
07:22:08 [watchdog] watchdog_start@82             Watchdog device (/dev/watchdog) not found
07:22:08 [     sdk] try_sensor_config@3081        SENSOR=imx327
07:22:08 [     sdk] find_sensor_config@3049       matched sensor config: imx327.ini
07:22:08 [     sdk] start_sdk@246                 Using /etc/sensors/imx327.ini sensor configuration file
07:22:08 [     sdk] start_sdk@255                 App was built with MPP version: Hi3519V101_MPP_V1.0.5.0.B040 Release
07:22:08 [     sdk] start_sdk@258                 Current MPP version: HI_VERSION=Hi3519V101_MPP_V1.0.5.0 B040 Release
open sys: No such file or directory
07:22:08 [     sdk] start_sdk@267                 Cannot get chip id
07:22:08 [     sdk] start_sdk@267                 Unknown error code 2
07:22:08 [    main] main@223                      Shutdown main thread

```

gitgayhub, 2022-03-15:
Update:
Sorry I forgot to set osmem in uboot, after setting osmem, `SENSOR=imx327 majestic` without rebooting gave the following error:

```
root@openipc-hi3516av200:/etc/sensors# SENSOR=imx327 majestic
07:49:34 [    main] main@135                      Majestic Lite for HiSilicon, version master+912fa6f, built on 2022-03-08
07:49:34 [app_conf] load_config@345               Using /etc/majestic.yaml as main configuration
07:49:34 [watchdog] watchdog_start@82             Watchdog device (/dev/watchdog) not found
07:49:34 [     sdk] try_sensor_config@3081        SENSOR=imx327
07:49:34 [     sdk] find_sensor_config@3049       matched sensor config: imx327.ini
07:49:34 [     sdk] start_sdk@246                 Using /etc/sensors/imx327.ini sensor configuration file
07:49:34 [     sdk] start_sdk@255                 App was built with MPP version: Hi3519V101_MPP_V1.0.5.0.B040 Release
07:49:34 [     sdk] start_sdk@258                 Current MPP version: HI_VERSION=Hi3519V101_MPP_V1.0.5.0 B040 Release
07:49:34 [     sdk] start_sdk@281                 sensor stSnsObj
07:49:34 [     sdk] start_sdk@286                   input_mode MIPI, WDR NONE
07:49:34 [     sdk] start_sdk@293                   dev [1920x1080]@4x4 25fps, RGGB
07:49:34 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx327.so
07:49:34 [     sdk] dump_vb_configuration@2018    VB configuration:
07:49:34 [     sdk] dump_vb_configuration@2028      [0]: 3110400 x 3
07:49:34 [     sdk] dump_vb_configuration@2028      [1]: 4147200 x 1
07:49:34 [     sdk] init_sensor@2221              Sensor driver loaded
07:49:34 [    puts]                               linear mode
07:49:34 [    puts]                               Dng image static info have not been initialized in Cmos.c!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
07:49:34 [    puts]                               I2C_WRITE error!
[... 84 lines cut ...]
07:50:02 [     hal] take_yuv420sp_video_fram@1088 ERR_VPSS_BUF_EMPTY: The picture queue is empty
07:50:02 [image_co] image_control_compute_his@100 Cannot take yuv420sp VI frame
07:50:02 [image_co] image_control_compute_his@100 ERR_VPSS_BUF_EMPTY: The picture queue is empty
07:50:02 [image_tu] tune_image_params@180         get_histogram errored
07:50:02 [    vpss] VpssUserGetFrame@388          vpss(0, 0) user get frame time out, s32MilliSec=10000
07:50:04 [     sdk] venc_read@814                 Timeout get any data from venc channel 0
[... cut ...]
^C07:50:14 [     log] printf@261                    Interrupt signal received
07:50:14 [    vpss] VPSS_DisableChn@6891          !! Disable VpssChn:0 timeout 120ms!!!
07:50:14 [    venc] VencDestroyChn@5318           Venc: 1 has been start recive picture, not surpport DestroyChn
07:50:14 [     sdk] stop_chn@1738                 Cannot destroy VENC chn 1
07:50:14 [     sdk] stop_chn@1738                 ERR_VENC_NOT_PERM: The operation, for example, modifying static parameters, is forbidden

```

I tried rebooting, but all modification are reverted back

gitgayhub, 2022-03-15:
reflashed openipc and then made modification and rebooted, now it says:

```
hisilicon: Loading of kernel modules and initialization of the video system has started
mmz_start: 0x90000000, mmz_size: 256M
hisilicon: Get data from environment and set SENSOR as imx327
++++++++++++++++++++++++++++++++++++++++++++++
==============vi_vpss_online==============
/usr/bin/load_hisilicon: line 450: spi0_4wire_pin_mux: not found
==== Your input Sensor0 type is imx327 ====
==== Your input Sensor1 type is NULL ====
Starting majestic: OK

Welcome to OpenIPC v2.2
openipc-hi3516av200 login: root

```
```
root@openipc-hi3516av200:~# SENSOR=imx327 majestic
08:12:21 [    main] main@135                      Majestic Lite for HiSilicon, version master+912fa6f, built on 2022-03-08
08:12:21 [app_conf] load_config@345               Using /etc/majestic.yaml as main configuration
08:12:21 [watchdog] watchdog_start@82             Watchdog device (/dev/watchdog) not found
08:12:21 [     sdk] try_sensor_config@3081        SENSOR=imx327
08:12:21 [     sdk] find_sensor_config@3049       matched sensor config: imx327.ini
08:12:21 [     sdk] start_sdk@246                 Using /etc/sensors/imx327.ini sensor configuration file
08:12:21 [     sdk] start_sdk@255                 App was built with MPP version: Hi3519V101_MPP_V1.0.5.0.B040 Release
08:12:21 [     sdk] start_sdk@258                 Current MPP version: HI_VERSION=Hi3519V101_MPP_V1.0.5.0 B040 Release
08:12:21 [     sdk] start_sdk@281                 sensor stSnsObj
08:12:21 [     sdk] start_sdk@286                   input_mode MIPI, WDR NONE
08:12:21 [     sdk] start_sdk@293                   dev [1920x1080]@4x4 25fps, RGGB
08:12:21 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx327.so
08:12:21 [     sdk] dump_vb_configuration@2018    VB configuration:
08:12:21 [     sdk] dump_vb_configuration@2028      [0]: 3110400 x 3
08:12:21 [     sdk] dump_vb_configuration@2028      [1]: 4147200 x 1
08:12:21 [     sdk] init_sensor@2221              Sensor driver loaded
08:12:21 [    puts]                               linear mode
08:12:21 [    puts]                               Dng image static info have not been initialized in Cmos.c!
08:12:21 [    puts]                               I2C_WRITE error!
08:12:21 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
08:12:22 [    puts]                               I2C_WRITE error!
[... 47 lines cut ...]
08:12:22 [     sdk] start_sdk@392                 HiSilicon SDK started
08:12:22 [   httpd] new_http_server@346           HTTP server started on :::80
08:12:22 [    rtsp] rtsp_init@30                  RTSP server started on port 554
08:12:22 [   netip] netip_start@2013              NETIP server started on port 34567
08:12:24 [     sdk] venc_read@814                 Timeout get any data from venc channel 0
08:12:24 [     sdk] venc_read@834                 NOTE: Current frame is NULL!
08:12:26 [     sdk] venc_read@814                 Timeout get any data from venc channel 0
08:12:26 [     sdk] venc_read@834                 NOTE: Current frame is NULL!
^C08:12:27 [     log] printf@261                    Interrupt signal received
08:12:35 [     hal] take_yuv420sp_video_fram@1088 Cannot get chn 0 frame for vpssGrp=0
08:12:35 [     hal] take_yuv420sp_video_fram@1088 ERR_VPSS_BUF_EMPTY: The picture queue is empty
08:12:35 [image_co] image_control_compute_his@100 Cannot take yuv420sp VI frame
[... cut ...]
08:12:35 [     sdk] stop_chn@1741                 Stopped 1 channel
08:12:35 [    vpss] VPSS_DisableChn@6891          !! Disable VpssChn:0 timeout 120ms!!!
08:12:35 [     sdk] stop_chn@1741                 Stopped 0 channel
08:12:35 [     sdk] ISP_thread_proc@160           Shutdown isp_run thread
08:12:36 [     viu] VIU_DRV_DisableChn@1344       !! Disable ViChn:0 timeout 120ms!!!
08:12:36 [     sdk] stop_sdk@2967                 Stop sdk Ok!
08:12:36 [    main] main@223                      Shutdown main thread

```

dimerr, 2022-03-15:
https://github.com/OpenIPC/firmware/blob/master/general/package/hisilicon-osdrv-hi3519v101/files/script/load_hisilicon

take this file, ive added imx327_spi there

```fw_setenv sensor imx327_spi```

and you have to take .so file from stock fw and use it

gitgayhub, 2022-03-15:
I used the libsns file from stock firmware, seems something is wrong with the absolute/relative path
```
root@openipc-hi3516av200:/usr/lib/sensors# SENSOR=imx327 majestic
08:49:56 [    main] main@135                      Majestic Lite for HiSilicon, version master+912fa6f, built on 2022-03-08
08:49:56 [app_conf] load_config@345               Using /etc/majestic.yaml as main configuration
08:49:56 [watchdog] watchdog_start@82             Watchdog device (/dev/watchdog) not found
08:49:56 [     sdk] try_sensor_config@3081        SENSOR=imx327
08:49:56 [     sdk] find_sensor_config@3049       matched sensor config: imx327_spi.ini
08:49:56 [     sdk] start_sdk@246                 Using /etc/sensors/imx327_spi.ini sensor configuration file
08:49:56 [     sdk] start_sdk@255                 App was built with MPP version: Hi3519V101_MPP_V1.0.5.0.B040 Release
08:49:56 [     sdk] start_sdk@258                 Current MPP version: HI_VERSION=Hi3519V101_MPP_V1.0.5.0 B040 Release
08:49:56 [     sdk] start_sdk@281                 sensor stSnsObj
08:49:56 [     sdk] start_sdk@286                   input_mode MIPI, WDR NONE
08:49:56 [     sdk] start_sdk@293                   dev [1920x1080]@4x4 25fps, RGGB
08:49:56 [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx327.so
08:49:56 [  sensor] try_to_load@22                dlopen "/usr/lib/sensors/libsns_imx327.so" error: Error loading shared library ld-linux.so.3: No such file or directory (needed by /usr/lib/sensors/libsns_imx327.so)
08:49:56 [  sensor] try_to_load@19                trying to load .//usr/lib/sensors/libsns_imx327.so
08:49:56 [  sensor] try_to_load@22                dlopen ".//usr/lib/sensors/libsns_imx327.so" error: Error loading shared library .//usr/lib/sensors/libsns_imx327.so: No such file or directory
08:49:56 [  sensor] try_to_load@19                trying to load /usr/lib//usr/lib/sensors/libsns_imx327.so
08:49:56 [  sensor] try_to_load@22                dlopen "/usr/lib//usr/lib/sensors/libsns_imx327.so" error: Error loading shared library /usr/lib//usr/lib/sensors/libsns_imx327.so: No such file or directory
08:49:56 [    main] main@223                      Shutdown main thread

```

dimerr, 2022-03-15:
That file is very weird, look at it's size... over 800k... seems again some reinvented wheel from manufacturer

gitgayhub, 2022-03-15:
Guess I will have to gave up this camera again

Thanks anyway, I will try openipc on more cameras

closed 2022-03-17 by gitgayhub