OpenIPC/firmware #203: SDK toolchain links

oliv3r, 2022-03-23:
Hey Guys,

should we have a page that links (maybe even mirrors) these vendor SDK's? I know the hisilicon one is easily to be obtained I think, e.g. kernel sources, u-boot sources; but when searching for the anyka sources, I only find sites that speak of 'Linux SDK' in text, without URL's. Has anybody tried to obtain these? Should we keep links in the wiki etc?

Btw, Anyka seems to be rebadged hisilicon? So maybe the sdk is the same too?

oliv3r, 2022-03-24:
As I have reached out to a few vendors, the first ot reply, is 'ezviz' which has a few doorbell/PTZ/cameras. Not sure who the real OEM is however. They redirect us to http://opensource.ys7.com/ for their stuff. So this might be the 'central location for all'. Sadly however, the copyright page at the bottom explicitly lists their copyright explicitly, so they might be their own OEM.

gitgayhub, 2022-03-29:
ezviz is part of hikvision, no way they will give you soc SDK. Guess you will have to apply directly from anyka [http://www.anyka.com/en/proApply.aspx](http://www.anyka.com/en/proApply.aspx)

Or I see someone is selling anyka SDK on taobao.com, price is 700CNY, you can buy there if you have methods

oliv3r, 2022-03-31:
> ezviz is part of hikvision, no way they will give you soc SDK. Guess you will have to apply directly from anyka http://www.anyka.com/en/proApply.aspx
I did, but anyka's mailboxes are full, so those signup requests just bounce :p  

as for 'ezviz is part of hikvision', what do you mean, ezvi is a consumer brand, hikvision a chip and camera manufacturer. You suggesting that ezviz is to hikvision as toyota is to lexus? Could make sense.

> 
> Or I see someone is selling anyka SDK on taobao.com, price is 700CNY(110USD), you can buy there if you have methods
redicilous :p

to be fair, we don't really care for the SDK, we REALLY care for the GPL source code, which usually is part of the SDK. I've already started to go down the 'GPL source code request' route; but as you can imagine, that is a quite difficult one.

gitgayhub, 2022-03-31:
ezviz is a brand created and owned by hikvision, ezviz is for lower-end consumer market, while hikvision is mainly for higher-end commercial market, and hikvision doesn't manufacture or design any chip, they just print their own logo on the chips they ordered from other chip companies like hisilicon, for example, the most used HK-2015-1 equals Hi3516AV100, nothing different

gitgayhub, 2022-03-31:
Hisilicon is part of Huawei, a much larger company than hikvision, they design chips but can't manufacture it, they used to own the largest market share of the IP camera SoC, but after the US sanction on Huawei, they can't find any chip foundry that is allowed to manufacture their chips. So then comes the other companies like Anyka, Fullhan, Goke, Ingenic and SigmaStar which are not sanctioned by the US government. They are the cheap replacement of hisilicon's low-end chips

Anyka has nothing to do with Hisilicon, it's just another independent chip company which ezviz happens to used their chips on some of their cameras. Their chips have nothing to do with the ones of hisilicon

khseal, 2022-07-18:
> Anyka has nothing to do with Hisilicon, it's just another independent chip company which ezviz happens to used their chips on some of their cameras.

I think you are wrong. I have a camera with a hisilicon processor. The software in the camera is from the anuka processor.

[root@anyka ~]$ cat /proc/cpuinfo 
Processor       : ARM926EJ-S rev 5 (v5l)
BogoMIPS        : 199.06
Features        : swp half fastmult edsp java 
CPU implementer : 0x41
CPU architecture: 5TEJ
CPU variant     : 0x0
CPU part        : 0x926
CPU revision    : 5

Hardware        : CLOUD39EV3_AK3918EV300_MNBD
Revision        : 0000
Serial          : 0000000000000000

[image]

widgetii, 2022-07-18:
It's a wrong assumption. HiSilicon and Anyka SDK have different APIs

khseal, 2022-07-18:
How does it work then? Perhaps the processor is relabeled?

ZigFisher, 2022-07-18:
It's a vile Chinese knockoff.
The chip marking is specially applied incorrectly to attract attention

ZigFisher, 2022-07-18:
> I think you are wrong. I have a camera with a hisilicon processor. The software in the camera is from the anuka processor.
>
> [root@anyka ~]$ cat /proc/cpuinfo Processor : ARM926EJ-S rev 5 (v5l) [...]

Please send me dump from this device via PM in Telegram messenger (my nick @FlyRouter)

closed 2023-11-19 by viktorxda