OpenIPC/firmware #204: hi3516cv100 with imx291_i2c

Peterka35, 2022-03-26:
No sensor imx291_i2c configuration with hi3516cv100
[image]

dimerr, 2022-06-13:
This is likely to be a false positive detection, or a very specific board, we dont have such driver, you can check for libsns_imx291.so in your stock firmware

khseal, 2022-10-09:
The sensor is identified correctly. 
This sensor is in 00002532
https://www.cctvsp.ru/cctv/modul-dlya-ip-kamery-ipg-54h20pl-s-hi3516cimx291-2mp

mpissarello, 2022-11-07:
I have the same camera: it's marked 54H20PL-S, it's a Hi3516CV100 + IMX291.

[https://t.me/openipc/29720](url)

However, the original firmware boot log shows a 53H20L board with IMX122 sensor.

I setup majestic with the ```imx122_spi_dc_1080p.ini``` config already present, but it shows a lot of ```Open /dev/ssp error!``` errors.

After some investigations I found that the imx291 sensor is missing inside the ```/usr/bin/load_hisilicon``` script, so the ```ssp_sony.ko``` module is not loaded at boot: I added the sensor at the same line of the imx122 (line 98) and the ```Open /dev/ssp error!``` error is gone.

However, majestic now shows errors on the venc_read:

```
15:47:26  <       majestic> [    main] main@125                      Majestic Lite for HiSilicon, version HEAD+1ae6d03, built on 2022-11-07
15:47:26  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
15:47:26  <       majestic> [watchdog] watchdog_start@101            Detected Hisilicon Watchdog
15:47:26  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
15:47:27  <       majestic> [     sdk] start_sdk@403                 Using /etc/sensors/imx122_spi_dc_1080p.ini sensor configuration file
15:47:27  <       majestic> [     sdk] start_sdk@412                 App was built with MPP version: Hi3518_MPP_V1.0.B.0.
15:47:27  <       majestic> [     sdk] start_sdk@415                 Current MPP version: HI_VERSION=Hi3518_MPP_V1.0.B.0 
15:47:27  <       majestic> [     sdk] start_sdk@436                 sensor imx122
15:47:27  <       majestic> [     sdk] start_sdk@448                   dev [1920x1080]@200x20 30fps, RGGB
15:47:27  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx122.so
15:47:27  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 97404KB
15:47:27  <       majestic> [     sdk] dump_vb_configuration@2136    VB configuration:
15:47:27  <       majestic> [     sdk] dump_vb_configuration@2146      [0]: 3133440 x 10
15:47:27  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 66784KB
15:47:27  <       majestic> [     sdk] init_sensor@2332              Sensor driver loaded
15:47:27  <       majestic> [    puts]                               -------Sony IMX122 Sensor Initial OK!-------
15:47:27  <       majestic> [     sdk] log_venc_chn@1245             H.264 1920x1080 30fps 4096Kbit  30 GOP
15:47:27  <     iq_monitor> [     sdk] monitor_thread@332            Loading IQ profile /etc/sensors/iq/default.ini
15:47:27  <       majestic> [     sdk] create_vpss_chn@992             new venc: 0   vpss_grp: 0,   vpss_chn: 0
15:47:27  <       majestic> [     sdk] start_video@1886                venc took 7664KB of MMZ [1518]
15:47:27  <       majestic> [     sdk] start_sdk@548                 HiSilicon SDK started
15:47:27  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 50168KB
15:47:27  <       majestic> [   httpd] new_http_server@370           HTTP server started on :::80
15:47:27  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
15:47:27  <       majestic> [   netip] netip_start@2051              NETIP server started on port 34567
15:47:27  <       majestic> [    gpio] set_gpio@25                   set_gpio(38, 0)
15:47:27  <       majestic> [    gpio] set_gpio@25                   set_gpio(39, 0)
15:47:27  <       majestic> [    gpio] set_gpio@25                   set_gpio(38, 1)
15:47:27  <       majestic> [    gpio] set_gpio@25                   set_gpio(39, 0)
15:47:27  <       majestic> [    gpio] set_gpio@25                   set_gpio(38, 0)
15:47:27  <       majestic> [    gpio] set_gpio@25                   set_gpio(39, 1)
15:47:28  <       majestic> [    gpio] set_gpio@25                   set_gpio(38, 0)
15:47:28  <       majestic> [    gpio] set_gpio@25                   set_gpio(39, 0)
15:47:29  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
15:47:29  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
15:47:31  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
15:47:31  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
15:47:33  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
15:47:33  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
15:47:35  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
15:47:35  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
15:47:37  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
15:47:37  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
15:47:39  <       majestic> [     sdk] venc_read@916                 Timeout get any data from venc channel 0
15:47:39  <       majestic> [     sdk] venc_read@936                 NOTE: Current frame is NULL!
^C
15:47:40  <       majestic> [     log] printf@268                    Interrupt signal received
15:47:40  <       majestic> [     sdk] stop_chn@1927                 Stopped 0 channel
[... cut ...]
[original_venc.txt](https://github.com/OpenIPC/firmware/files/9952856/original_venc.txt)

OPENIPC:
[openipc_first_boot.txt](https://github.com/OpenIPC/firmware/files/9952983/openipc_first_boot.txt)
[openipc_boot_after_mod.txt](https://github.com/OpenIPC/firmware/files/9952985/openipc_boot_after_mod.txt)
[openipc_lsmod.txt](https://github.com/OpenIPC/firmware/files/9952885/openipc_lsmod.txt)
[openipc_vi.txt](https://github.com/OpenIPC/firmware/files/9952886/openipc_vi.txt)
[openipc_venc.txt](https://github.com/OpenIPC/firmware/files/9952888/openipc_venc.txt)

closed 2026-10-16 by openipc-ai