OpenIPC/firmware #205: [Feature Suggestion] Any plan to add Homekit support?

gitgayhub, 2022-03-29:
Hi, first I want to say thanks for this great project again

Just want to know if there's any plan to add Apple Homekit support? I find there are many opensource projects to convert rtsp streams and publish to Homekit, for OpenIPC, this should be easier I guess? Since there's no need to convert rtsp streams using ffmpeg

ZigFisher, 2022-06-28:
Hi
In the near future we plan to add the MQTT protocol

gitgayhub, 2022-10-21:
> Hi In the near future we plan to add the MQTT protocol

Thanks, I noticed the new ```MQTT client``` menu under Services menu.  But there's no option to choose MQTT as method to send notification under the menu ```Motion guard```. Seems I can only send MQTT message and snapshot by manually executing the script ```/usr/sbin/send2mqtt.sh```

I tried to add a line ```motion_send2mqtt="true"``` to ```/etc/webui/motion.conf``` and reboot, can't make it working

gitgayhub, 2022-10-21:
I then looked in ```/usr/sbin/motiondetection.sh``` and  a line ```[ "true" = "$motion_send2mqtt" ] && send2mqtt.sh``` and rebooted, still no luck

Actually, the motion detection function doesn't seem to work at all. I added another line ```touch /tmp/motion_test``` at the beginning of ```/usr/sbin/motiondetection.sh```, and wave my hand before the camera, nothing happened, no ```motion_test``` file was created in ```/tmp``` directory

gitgayhub, 2022-10-21:
Fixed myself and submitted a [pull request](https://github.com/OpenIPC/microbe-web/pull/100) to add the send2mqtt option

The file ```/etc/init.d/S92motion``` is missing in the final firmware, have to create one myself to start ```motiondetection.sh``` automatically on system startup

```
#!/bin/sh
#
# Start motion detection
#

config_file=/etc/webui/motion.conf
[ ! -f "$config_file" ] && echo "Config file not found." && exit 2
source $config_file

start(){
	if [ "true" = "$motion_enabled" ]; then
			echo "Starting motion detection..."
			/usr/sbin/motiondetection.sh
	else
			echo "Motion detection service is disabled in ${config_file}."
			exit 3
	fi
}

stop(){
	echo "Stopping motion detection..."
	kill -9 $(pidof motiondetection.sh)
}

case "$1" in
        start)
				start
                ;;
        stop)
                stop
                ;;
		restart)
				stop
				sleep 1
				start
				;;
        *)
                echo "Usage: $0 {start|stop|restart}"
                exit 1
                ;;
esac
```

closed 2026-09-04 by openipc-ai