OpenIPC/firmware #206: Предложение. Несколько профилей конфигурации для дня и ночи.

vebmaster, 2022-03-29:
Здравствуйте.
Хочу предложить следующий функционал.
Камера (majestic) с помощью датчика освещённости меняет профили/конфигурации в зависимости от освещённости.
Например вариант с 2 профилями день/ночь.
Зачем это надо: я заметил, что ночью при низкой освещённости (или наоборот днём при яркой) некоторые настройки могут негативно сказываться на качестве получаемой картинки с видеокамеры.
Например такие параметры как: Яркость, AGC, DWDR, BLC, Затвор, Битрейт
И если настроить 2 профиля конфигурации для дня и ночи, то можно получать наилучшую картинку для соответствующего режима камеры (день/ночь).
Т.е. не просто смена "цветной" на "ч\б", а смена почти целого профиля со своими настройками картинки.

В телеграмме пишут, что это неактуально уже, но я считаю наоборот это более продвинутой конфигурацией камеры.

vebmaster, 2022-05-28:
@katyo
> У вас прошивка работает нормально при низкой освещенности? Какой процессор?

Да, нормально. Только ваш вопрос немного не по теме обсуждения :) Лучше создать отдельный вопрос, если у вас проблемы какие то или добро пожаловать в чат телеграм: https://t.me/openipc_modding
Было как то что фпс проседал в темноте, когда я slowShutter изменял на любое значение, отличное от "disabled". После установки в "disabled" фпс в темноте перестал проседать.

katyo, 2022-05-30:
Такой же конфиг, как у меня. Но в моём случае бесконечно перезагружается при недостатке освещения из-за того, что стример быстро съедает всю доступную память и система перестаёт сбрасывать watchdog.
Я создал вопрос в треккере majestic.

widgetii, 2022-05-30:
Такую возможность планируется добавить в ближайшем будущем, как только будет получено достаточно отзывов о работе единого профиля в `isp.iqProfile`

vebmaster, 2022-05-30:
> как только будет получено достаточно отзывов о работе единого профиля в `isp.iqProfile`

это в последних прошивках появилось? У меня щас декабрьская стоит, не в курсе такого.

closed 2022-09-12 by widgetii