OpenIPC/firmware #207: Anyka AK3918Ev200

ricardojlrufino, 2022-04-14:
Hi, i try boot this version. but i get no console after kernel load:
```
anyka$setenv loadaddr 81008000

anyka$readcfg; sf probe 0:0 ${sf_hz} 0; sf read ${loadaddr} 678000 15DC60; bootm ${loadaddr}

SF: 1432672 bytes @ 0x678000 Read: OK
## Booting kernel from Legacy Image at 81808000 ...
   Image Name:   Linux-3.4.35-ak3918ev200
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1199088 Bytes = 1.1 MiB
   Load Address: 81008000
   Entry Point:  81008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Anyka Linux Kernel Version: 1.0.05
```

My original boot:
```
U-Boot 2013.10.0-AK_V2.0.04 (Apr 14 2020 - 11:47:31)

DRAM:  64 MiB
set  pin:32 i:5
set  pin:34 i:8
8 MiB
Create flash partition table init OK!
ANYKA SDHC/MMC4.0: 0
Load Env CRC OK!
In:    serial
Out:   serial
Err:   serial
Net:   AKEthernet-0

Hit any key to stop autoboot:  3                                                                                                                         0

SF: 1307448 bytes @ 0x37000 Read: OK
## Booting kernel from Legacy Image at 81808000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1307384 Bytes = 1.2 MiB
   Load Address: 81808000
   Entry Point:  81808040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Anyka Linux Kernel Version: 2.5.02
Booting Linux on physical CPU 0
Linux version 3.4.35 (ma@ma-PC) (gcc version 4.8.5 (anyka (gcc-4.8.5 + binutils-2.24 + ulcibc-0.9.33.2)(20170223)) ) #5 Wed Sep 18 15:36:19 CST 2019
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: Cloud39E_AK3918E+H42_V1.0.2
Memory policy: ECC disabled, Data cache writeback
ANYKA CPU AK3918 (ID 0x20150200)
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 10160
Kernel command line: console=ttySAK0,115200n8 root=/dev/mmcblk0p2 rw rootwait rootfstype=squashfs init=/init mem=64M memsize=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
```

Geotv123, 2022-05-13:
> Hi, i try boot this version. but i get no console after kernel load:
>
> ```
> anyka$setenv loadaddr 81008000
>
> anyka$readcfg; sf probe 0:0 ${sf_hz} 0; sf read ${loadaddr} 678000 15DC60; bootm [...]

Hi @ricardojlrufino 

Did you solve the issue? Let us know how your progress is going.

ricardojlrufino, 2022-06-05:
I haven't been able to run it yet...

I've been experimenting with this camera and document on my blog.
https://ricardojlrufino.wordpress.com/2022/04/03/hack-ipcam-anyka-oled-display-on-cam/
https://ricardojlrufino.wordpress.com/2022/02/14/hack-ipcam-anyka-teardown-and-root-access/

leonelhs, 2022-07-12:
I'm facing exactly the same problem mentioned above , no console after kernel loads.
I'm using this method mentioned at this blog for upgrade my kernel [https://blog.caller.xyz/v380-ipcam-firmware-patching](url). It basically consist in replace MTD partitions, my uImage are placed at partition MTD1. 
The generated openipc output rootfs.squashfs is to big for being flashed in the RFS partition so this has given to me a hint, this ak200 openipc build  its for different hardware variant than mine the V380E2_C2,  my variant has 8 partitions.

> 1.5M mtd1 KER
> 4.0K mtd2 MAC
> 4.0K mtd3 ENV
> 1.5M mtd4 RFS
> 1.6M mtd5 USR
> 1.2M mtd6 MVS
> 460K mtd7 EXT
> 1.5M mtd8 JFS

Does anybody available to clarify our issues @ZigFisher?
Partitions are related problem?
because the OEM u-boot remains, is this capable to load the new kernel? 
[Lamobo-D1 dafault board config ](https://github.com/Lamobo/Lamobo-D1/blob/master/src/kernel/arch/arm/configs/aimer39_ak3918_D1_defconfig) has some kernel features that aren’t included in this configuration like kernel params and Ram init address, shall I include some special address for my board?
It could be that I have no idea what I’m doing?

I have tried this command to match my zImage load_address params whit my old uImage without  succes.
_mkimage -A arm -O linux -T kernel -C none -a 81808000 -e 81808040 -n Linux -d zImage uImage_

I'm using the ch341prog, to flash from the scratch the original firmware so I'm available to try any idea with no fear to brick my camera.

dimerr, 2022-07-12:
We have discovered that sdk version in stock firmwares are significantly higher than  in lamobo-d1, thus we are not sure if that lamobo-d1 is able to start at all.

I have built kernel just from their sources, with defconfig, still same  thing

leonelhs, 2022-07-12:
By the way I'm using the uImage produced by OpenIPC firmware repo: Linux-3.4.35-ak3918ev200 not Lamobo-D1

Piorun2002, 2023-02-20:
The anyka has dedicated /usr (C) partition when the full rootfs can be located  (5M should be enough fir it)
```
Creating 6 MTD partitions on "spi0.0":
0x000000033000-0x0000001b3000 : "KERNEL"
0x0000001b3000-0x0000001b4000 : "MAC"
0x0000001b4000-0x0000001b5000 : "ENV"
0x0000001b5000-0x0000002af000 : "A"
0x0000002af000-0x0000002fa000 : "B"
0x0000002fa000-0x0000007e8000 : "C"

[root@camFR ~]$ df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               896.0K    896.0K         0 100% /
/dev/mtdblock6            4.8M      4.8M         0 100% /usr
/dev/mtdblock5          300.0K    116.0K    184.0K  39% /etc/jffs2
/dev/loop0               95.0K       512     94.5K   1% /tmp/ramdisk
[root@camFR ~]$

```
So I thing we need to create new rootf, load it to /dev/mtdblock6   (can be done from u-boot) and change the bootargs to mount root from mtdblock6 instead of mtd4

`Kernel command line: console=ttySAK0,115200n8 root=/dev/mtdblock4 rootfstype=squashfs init=/sbin/init mem=64M flash=SF device_id=00000000000 ethaddr=00:55:7b:b5:7d:11 devinfo=jcoxxx memsize=64M
`
Full boot log from came Sannce I41V  /  ANNKE I41V 
```
[root@camFR ~]$ dmesg
Booting Linux on physical CPU 0
Linux version 3.4.35 (root@ubuntu) (gcc version 4.8.5 (anyka (gcc-4.8.5 + binutils-2.24 + ulcibc-0.9.33.2)(20170223)) ) #1 Thu Aug 29 20:35:38 CST 2019
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: Cloud39EV3_AK3918E80PIN_MNBD
Memory policy: ECC disabled, Data cache writeback
ANYKA CPU AK3918 (ID 0x20160100)
On node 0 totalpages: 7680
free_area_init_node: node 0, pgdat c03d30b0, node_mem_map c03f9000
  Normal zone: 60 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 7620 pages, LIFO batch:0
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 7620
Kernel command line: console=ttySAK0,115200n8 root=/dev/mtdblock4 rootfstype=squashfs init=/sbin/init mem=64M flash=SF device_id=00000000000 ethaddr=00:55:7b:b5:7d:f7 devinfo=jcoxxx memsize=64M
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 30MB = 30MB total
Memory: 26268k/26268k available, 4452k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc2000000 - 0xff000000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc1e00000   (  30 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0394000   (3632 kB)
      .init : 0xc0394000 - 0xc03ae000   ( 104 kB)
      .data : 0xc03ae000 - 0xc03d3e18   ( 152 kB)
       .bss : 0xc03d3e3c - 0xc03f4694   ( 131 kB)
SLUB: Genslabs=13, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:117
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
AK39 console driver initial
console [ttySAK0] enabled
Calibrating delay loop... 199.06 BogoMIPS (lpj=995328)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x824cb9c8 - 0x824cba20
devtmpfs: initialized
NET: Registered protocol family 16
-- Checked Platform AK3919Ev300(V400) ---
[... 232 lines cut ...]
exp_time=2988,4294939810
uio_vcodec_ioctl AKUIO_VIDEO_RESET
isp_>blc_mode=0exp_time=3745
exp_time=3745,4294939866
ak_isp_set_user_params_do 1458, set main osd mem attr
ak_isp_set_user_params_do 1473 chn:0, paddr:814ff540 size:49152
ak_isp_set_user_params_do 1458, set main osd mem attr
ak_isp_set_user_params_do 1473 chn:1, paddr:81511540 size:49152
ak_isp_set_user_params_do 1458, set main osd mem attr
ak_isp_set_user_params_do 1473 chn:2, paddr:81523540 size:49152
ak_isp_set_user_params_do 1509 paddr:81530940 size:10240
ak_isp_set_user_params_do 1509 paddr:81534540 size:10240
ak_isp_set_user_params_do 1509 paddr:81538140 size:10240
MAINCHN0 iomap f1731540 814f9540 size:24576
SUBCHN0 iomap f1767540 8152f540 size:5120
HiconnDiscovery uses obsolete (PF_INET,SOCK_PACKET)
RTL871X: indicate disassoc
RTL871X: set bssid:00:00:00:00:00:00
RTL871X: set ssid [CCCCCCC] fw_state=0x00000008
RTL871X: set bssid:9c:d6:43:11:11:11
RTL871X: start auth
RTL871X: auth success, start assoc
RTL871X: assoc success

```

closed 2026-11-09 by openipc-ai