OpenIPC/firmware #216: gk7205v300 зависает в ночью (нехватка памяти)

fadermix, 2022-05-10:
Две камеры, обе на gk7205v300 imx335. Зависают в одно и то же время, ночью во время дождя. На улице присутствует фонарь, кадр не совсем темный.

Одна камера бесконечно перезагружается, отказываясь отдавать поток и медленно загружая странички, другая отдаёт побитый поток (траффик в плеере идёт, но декодировать не получается). Первая камера работает по tcp, вторая по udp. Включенный или отключенный hls - не влияет.

Конфиг камер практически дефолтный:
```
system:
  buffer: 1024
image:
  mirror: false
  flip: false
  rotate: none
  hue: 50
  saturation: 50
  luminance: auto
  contrast: 45
osd:
  enabled: true
  template: "%d.%m.%y %H:%M:%S WIDE"
  posX: 0
  posY: 0
nightMode:
  enabled: true
  pinSwitchDelayUs: 150
  irSensorPinInvert: false
  irCutPin1: 11
  irCutPin2: 10
  irSensorPin: 66
records:
  enabled: false
  path: /mnt/mmc/%Y/%m/%d/%H.mp4
  maxUsage: 95
video0:
  enabled: true
  codec: h264
  size: 2592x1944
  fps: 12
  gopMode: normal
  rcMode: avbr
video1:
  enabled: false
  codec: h265
jpeg:
  enabled: true
  toProgressive: false
mjpeg:
  size: 640x360
  fps: 5
  bitrate: 1024
audio:
  enabled: false
  volume: auto
  srate: 16000
  codec: opus
  device: hw:2
rtsp:
  enabled: true
  port: 554
hls:
  enabled: true
youtube:
  enabled: false
motionDetect:
  enabled: true
  profile: outdoor
  visualize: false
  debug: false
ipeye:
  enabled: false
onvif:
[... 93 lines cut ...]
Jan  1 00:01:46 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:46 openipc-gk7205v300 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  1 00:01:48 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Down
Jan  1 00:01:51 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Jan  1 00:01:56 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Down
Jan  1 00:01:57 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Jan  1 00:02:00 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Down
Jan  1 00:02:01 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Jan  1 00:02:15 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Down
Jan  1 00:02:16 openipc-gk7205v300 kern.info kernel: femac 10040000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
```

fadermix, 2022-05-23:
Upd: судя по всему - проблемы с платной, отваливается ethernet контроллер при понижении температуры ночью

katyo, 2022-05-27:
@fadermix У меня в majestic начинает быстро течь память при низкой освещенности, после чего прошивка перезагружается с watchdog.
Проблема присутствует давно. Исходников этой программы нет.

carbofos, 2022-07-18:
> Upd: судя по всему - проблемы с платной, отваливается ethernet контроллер при понижении температуры ночью

Нет, это баг.
Подтверждаю, что баг все еще есть.
У меня в темноте зависает вся система в аналогичной конфигурации, по сети не отвечает. При включении освещения сразу одупляется.

katyo, 2022-07-19:
@carbofos Какой у вас кодек используется?

У меня память течёт при использовании H264.

carbofos, 2022-07-19:
> @carbofos Какой у вас кодек используется?
> 
> У меня память течёт при использовании H264.

h.265
Думаю что это связано с расчетом экспозиции кадра в majestic.
При обычной работе он занимает 17200. Через пару минут темноты:
```
  835     1 root     S    24768  91%   3% /usr/bin/majestic -s
```

Еще через десять:
```
  835     1 root     R    27244 100%  33% /usr/bin/majestic -s
client_loop: send disconnect: Broken pipe
user@pc ~ $ 
```

katyo, 2022-07-19:
В моём случае с h264 память утекает буквально за секунды. Далее собака перезагружает его.

widgetii, 2022-07-19:
Проблема неоднократно обсуждалась в Telegram канале. Используйте PQTools, чтобы сделать профиль для камеры

carbofos, 2022-07-20:
> Проблема неоднократно обсуждалась в Telegram канале. Используйте PQTools, чтобы сделать профиль для камеры

Не могу найти pqtools для платформы goke. Единственное описание этой софтины нашел, но там нет ни слова про goke:

https://openipc.github.io/wiki/en/image-quality-tuning

В телеграме тоже не нашел. Где брать pqtools для goke?

widgetii, 2022-07-21:
https://t.me/openipc_modding/152091

carbofos, 2022-07-22:
Перекапывая сообщения в телеге нашел еще один костыль для решения этой проблемы: отключить HLS.
Для этого в командной строке камеры через ssh:

```
yaml-cli -i /etc/majestic.yaml -s .isp.blkCnt 6
yaml-cli -i /etc/majestic.yaml -s .image.contrast 50
yaml-cli -i /etc/majestic.yaml -s .image.luminance 50
yaml-cli -i /etc/majestic.yaml -s .video0.codec h264
yaml-cli -i /etc/majestic.yaml -s .hls.enabled false
reboot
```

carbofos, 2022-07-22:
Еще рецепт из группы телеги:

Когда на камере запустите pqtools там же будет ссылка откуда взять клиентскую часть:

```
root@openipc-gk7205v300:~# pqtools

A supported processor with unknown MPP was found...

Download PQTools from https://github.com/widgetii/pqtools/raw/master/gk7205v200/PQTools.tgz

Downloading the latest tools, please wait...

ittb_V1.0.2.0/
ittb_V1.0.2.0/start.sh
ittb_V1.0.2.0/config.cfg
ittb_V1.0.2.0/libs/
ittb_V1.0.2.0/dev/
ittb_V1.0.2.0/lib/
ittb_V1.0.2.0/lib/libdehaze.so
ittb_V1.0.2.0/lib/libvoice_engine.so
ittb_V1.0.2.0/lib/libpthread-0.9.33.2.so
ittb_V1.0.2.0/lib/libisp.so
ittb_V1.0.2.0/lib/libive.so
ittb_V1.0.2.0/lib/libuClibc-0.9.33.2.so
ittb_V1.0.2.0/lib/libldci.so
ittb_V1.0.2.0/lib/libsecurec.so
ittb_V1.0.2.0/lib/libawb_natura.so
ittb_V1.0.2.0/lib/ld-uClibc-0.9.33.2.so
ittb_V1.0.2.0/lib/ld-uClibc.so.0
ittb_V1.0.2.0/lib/libupvqe.so
ittb_V1.0.2.0/lib/libm-0.9.33.2.so
ittb_V1.0.2.0/lib/libae.so
ittb_V1.0.2.0/lib/libmpi.so
ittb_V1.0.2.0/lib/libc.so.0
ittb_V1.0.2.0/lib/libresolv-0.9.33.2.so
ittb_V1.0.2.0/lib/libawb.so
ittb_V1.0.2.0/lib/libresolv.so.0
ittb_V1.0.2.0/lib/libdl-0.9.33.2.so
ittb_V1.0.2.0/lib/libdnvqe.so
ittb_V1.0.2.0/lib/libivp.so
ittb_V1.0.2.0/lib/libpthread.so.0
ittb_V1.0.2.0/lib/libir_auto.so
ittb_V1.0.2.0/lib/libm.so.0
ittb_V1.0.2.0/lib/libtde.so
ittb_V1.0.2.0/lib/libdl.so.0
ittb_V1.0.2.0/lib/libdrc.so
ittb_V1.0.2.0/ittb_control

We run the utilities for sensor calibration...

mounting dev
mounting libs
pqtools:begin to run
The result is 1
g_bUseSpecAWB is 0
CJIANG_PQT_Realpath failed![libs/libae.so]port is : 4321

<PQT_CreatServerSocket>(1811)bind ok!
pqtools:Waiting for connection from client!

<CJIANG_PQT_CommuInit>(2029)Not support uart communication!
```

Сообщается что клиентская часть pqtools хоть и под винду, но работает под wine.
У меня не запустилось.

katyo, 2022-07-22:
Можно ли просто извлечь профиль из заводской прошивки?
Где его следует там искать?

katyo, 2022-07-22:
pqtools как есть запустить не удалось:
```
root@cam1:~# pqtools

A supported processor with [Hi3516EV200_MPP_V1.0.1.2 B030] was found...

Downloading the latest tools, please wait...

gzip: invalid magic
tar: short read
```
В этом скрипте URL: https://raw.githubusercontent.com/widgetii/pqtools/master/hi3516ev300/ittb_ev300_V1.0.1.2.tgz
По нему отдают 404.
По всей видимости следует исправить на https://github.com/widgetii/pqtools/blob/master/hi3516ev300/ittb_V1.0.1.2.tgz

katyo, 2022-07-22:
@carbofos,
Клиентская часть HiPQTools под wine у меня запускается, от Goke по вашей ссылке тоже.
wine-7.8 (Staging)

widgetii, 2022-07-28:
В свежей версии прошивки появилась работа с профилями, которая в условиях низкой освещенности гасит шумы и уменьшает битрейт (и также расход памяти)

katyo, 2022-07-28:
Подтверждаю заметные улучшения съёмки в условиях низкой освещённости с выбранным профилем. Тестировал на hi3516ev200 с imx307. В тех же условиях раньше majestic с h264 просто съедал всю доступную память, захлёбываясь шумом. Система переставала отвечать, и watchdog делал ребут. Теперь в глубоких сумерках без искусственного освещения ещё возможно что-то осмысленное наблюдать.

closed 2022-08-17 by widgetii