OpenIPC/firmware #221: HI3518EV100 / cannot capture yuv420 image

TheChatty, 2022-05-13:
Loading of yuv420 image fails, jpg directly afterwards succeeds:

```
00:07:00 [     hal] take_yuv420sp_video_fram@1098 Cannot get chn 0 frame for vpssGrp=0
00:07:00 [     hal] take_yuv420sp_video_fram@1098 ERR_VENC_NOBUF: The buffer fails to be allocated due to some causes such as oversize of the data buffer applied for
00:07:00 [     sdk] sdk_take_yuv@3213             Cannot take yuv420sp VI frame
00:07:00 [     sdk] sdk_take_yuv@3213             ERR_VENC_NOBUF: The buffer fails to be allocated due to some causes such as oversize of the data buffer applied for
00:07:26 [     sdk] sdk_take_jpeg@1777            take jpeg venc_chn(1)   1280x720  q:50
00:07:26 [      vi] ViuDrvProcPrepFrm@2931        [    vi] s32PhyChn 0 get vb fail,w:1280,h:720
00:07:26 [     sdk] VENC_SaveJpeg@701             JPEG snapshot was received from hardware encoder
00:07:26 [  on_jpg] send_regular_jpeg@76          JPEG was sent to client
Segmentation fault (core dumped)
```
A seg fault is following soon after.

widgetii, 2022-05-13:
According to logs there is no enough memory for do that on camera

closed 2022-05-13 by widgetii

TheChatty, 2022-05-13:
```
MemTotal:          28416 kB
MemFree:            7684 kB
```
Shouldn't that be enough? Even if not, this case should be handled gracefully unless you want to allow DoS attacks.

widgetii, 2022-05-13:
Unfortunately, I don't have such hardware to debug the issue