OpenIPC/firmware #230: cannot find Majestic sourcecode

sen-h, 2022-05-22:
Hi, 
it appears that all the majestic makefiles simply point at pre-compiled binary versions hosted on an AWS instance.
Here is the majestic-xm530.mk file as an example:

`################################################################################
#
# majestic-xm530
#
################################################################################

MAJESTIC_XM530_VERSION = current
MAJESTIC_XM530_SOURCE = majestic.xm530.lite.master.tar.bz2
MAJESTIC_XM530_SITE = https://openipc.s3-eu-west-1.amazonaws.com
MAJESTIC_XM530_LICENSE = MIT
MAJESTIC_XM530_LICENSE_FILES = LICENSE

define MAJESTIC_XM530_INSTALL_TARGET_CMDS
        $(INSTALL) -m 755 -d $(TARGET_DIR)/etc
        $(INSTALL) -m 644 $(@D)/majestic-mini.yaml $(TARGET_DIR)/etc/majestic.yaml
        $(INSTALL) -m 644 $(@D)/majestic.yaml $(TARGET_DIR)/etc/majestic.full

        $(INSTALL) -m 755 -d $(TARGET_DIR)/usr/bin
        $(INSTALL) -m 755 -t $(TARGET_DIR)/usr/bin $(@D)/majestic
endef

$(eval $(generic-package))
` 

I understand the usefullness of this for speeding up builds, but is the **actual** source code for Majestic hosted anywhere publicly available?

widgetii, 2022-05-23:
Majestic is supplied only in precompiled binary form

closed 2022-05-23 by widgetii

brainstorm, 2022-09-29:
@widgetii Could you elaborate as to why is it closed source?

widgetii, 2022-09-30:
Because it's a commercial product
You might take a look at [mini](https://github.com/openipc/mini) as open source alternative

xfarrow, 2024-09-25:
But if it's a commercial product, who are you selling it to? The open source community would love and appreciate it to be opened. People want to install OpenIPC to drive away from proprietary software only to end with another proprietary software. Privacy-focused people are the very target of this project.

widgetii, 2024-09-27:
There are plenty of open source streamers, you're free to choose