OpenIPC/firmware #238: Детекция движения с отправкой на email

esuslik, 2022-06-18:
Возможно реализовать детекцию движения с отправкой на email как на стоковых прощивках?
Есть облачные сервисы которые для экономии на ресурсах серверов перекладывают функцию детекции на камеру и принимают сигнализацию на email

widgetii, 2022-06-18:
Если есть какой-то протокол, который поддерживается существующими серверами, то нужно его описание и методику, как (и на чем) тестировать его корректность

esuslik, 2022-06-19:
на сколько я понял это стандартная реализация почтового клиента.
для примера скрины с двух камер. Это просто настройка с какого почтового сервера, куда отправить письмо и скакой темой и содержимым.
Тестировать возможно на обычном почтовом сервере, например на яндексе. 
[image]
[image]

ZigFisher, 2022-06-28:
Отправку можно делать через имеющийся у нас в прошивке curl, он понимает почтовые протоколы и умеет SSL.
Это следующий этап для создания плагина от @themactep

ZigFisher, 2022-07-24:
Добавлен плагин отправки снапшота на Email

esuslik, 2022-07-25:
Плагин увидел, настроил. Но не понятно как к нему прикрутить детекцию, чтобы при движении в кадре, шла отправка на email

themactep, 2022-07-25:
Маджестик умер или mjpeg отключен?

esuslik, 2022-07-25:
отключен mjpeg был

esuslik, 2022-07-25:
они у меня все отключены кроме rtsp, т.к. на них нет авторизации

esuslik, 2022-07-25:
при нажатии на кнопку send to email что-то должно происходить?
Я вулючил mjpeg, картинка появилась. При нажатии send to email ничего не происходит

Также есть поозрение что фуекция send to mail не должна зависеть от mjpeg и т.п.
При детекции движения я так понммаю должно просто уходить пустое письмо с текстом что замечено движение.

themactep, 2022-07-25:
для отсылки снэпа должен быть включен jpeg и настроена отсылка например на почту. проверить работоспособность настроек можно запустив /usr/sbin/send2email.sh из консоли. если есть ошибка, она там вылезет.

esuslik, 2022-07-26:
Проверил на почту яндекс приходит письмо со снапшотом.
Есть возможность убрать вложение снапшота? или возможность отключать его отправку.
Кажется плагин send2email это должен быть только транспорт. А вот что уже отправлять мы бы выбирали в других плагинах типа детектор движения и внем уже ставить галку надо ли прикреплять вложение и какой текст сообщения.

Плагин motion detect ничего не отправляет. 
Запустил в консоли logread -f | grep "Motion detected in 3 regions" и там тишина, возможно детекто не не пишет события в лог?

themactep, 2022-07-26:
Какая камера? Есть ли в ней аппаратная детекция движения?

esuslik, 2022-07-26:
hi3516ev300 (hi3516ev300 family), imx335, 16 MB Flash, 2.2.07.25-lite

На стоковой прошивке работает. Не знаю аппаратная или нет. Но на картинке при включении детекции, видно квадратиками что детектит движения

themactep, 2022-07-26:
Использование send2email для motion guard -- это частный случай. Это вполне самостоятельный плагин для отправки снэпшота на преднастроенную почту. Я не вижу смысла в том, чтобы убирать из письма картинку. И что тогда останется? Разверни мысль и приведи алгоритм использвования без картинки.

themactep, 2022-07-26:
Запусти маджестик из консоли и посмотри, что будет в выводе при движении перед камерой. Может там синтаксис другой. Естественно, в настройках маждестика должна быть включена поддержка motion detection.

esuslik, 2022-07-26:
Есть видеорегистратор который пишет архив. На нем можно подключить детекцию движения чтобы на ленте архива показывать участки где есть движение. 
Узнает он о том что было движение путем получения письма на email. И ему достаточно просто сообщения без вложений.
То есть если без картинки будет экономится место.
Если в камере не включен jpeg то на почту приходит битая картинка.

Если смотреть аналогию с камерами и стоковыми прошивками, то там много событий который можно детектить и отсылать по разным каналам связи.

Например детектор проблем с сд картой, при обнаружении отправить на email. Зачем тут картинка?
Детоктор проблемы с сетью, отправить после восстановления уведомление на email. Зачем тут картинка?
Детектор вообще любого события может быть, и не всегда нужно чтобы была картинка.

Понятно что такгого функционала пока нет, но  на будущее он может появиться.

esuslik, 2022-07-26:
Но даже с картинкой сейчас при движении ничего не приходит на email. Где-то может можно логи посмотреть, почем не уходит или может нечему уходить?

esuslik, 2022-07-26:
[image]

themactep, 2022-07-26:
Запусти `/usr/sbin/send2email.sh` из консоли камеры. Если картинка приходит, значит нет вызова из парсера логов. Надо стартовать маждестик руками в консоли и отследить, шлет ли камера сигнал, что замечено движение. Вот пример сообщений с камеры:

```
Jul 25 18:41:25 openipc-hi3518ev200 user.debug majestic[958]: Motion detected in 1 regions, 1872x188 -> [1912x432]
Jul 25 18:41:25 openipc-hi3518ev200 user.debug majestic[958]: 1832,48  80,590
Jul 25 18:41:25 openipc-hi3518ev200 user.debug majestic[958]: 1896,702  16,172
Jul 25 18:41:25 openipc-hi3518ev200 user.debug majestic[958]: Motion detected in 2 regions, 1832x48 -> [1912x874]
Jul 25 18:41:25 openipc-hi3518ev200 user.debug majestic[958]: 1808,0  104,890
Jul 25 18:41:25 openipc-hi3518ev200 user.debug majestic[958]: 1616,998  72,40
Jul 25 18:41:25 openipc-hi3518ev200 user.debug majestic[958]: Motion detected in 2 regions, 1616x0 -> [1912x1038]
Jul 25 18:41:26 openipc-hi3518ev200 user.debug majestic[958]: 1776,0  136,882
Jul 25 18:41:26 openipc-hi3518ev200 user.debug majestic[958]: 1816,908  96,162
Jul 25 18:41:26 openipc-hi3518ev200 user.debug majestic[958]: 1616,998  40,40
Jul 25 18:41:26 openipc-hi3518ev200 user.debug majestic[958]: Motion detected in 3 regions, 1616x0 -> [1912x1070]
```

themactep, 2022-07-26:
Детект есть. Осталось понять, почему не шлет. Какой содержимое файла, что создается при настройке детектора? Он там выводится рядом с формой.

themactep, 2022-07-26:
После перезагрузки камеры `# ps | grep motion`, чтобы проверить, что созданные файл запущен. Вывод должен содержать строку типа `585 root     {S92motion} /bin/sh /etc/init.d/S92motion start`.

esuslik, 2022-07-26:
[image]

esuslik, 2022-07-26:
[image]

esuslik, 2022-07-26:
Кстати если выставлять Sensitivity то осле применения всегда ползунок показывает значение 3, хотя в скрипте в конфиг файле стоит то которое применил.
То есть не понять какое значение sensivity стоит не понимая и расшифровывая конфиг файл

esuslik, 2022-07-26:
Удалось протестировать отправку на yandex через yandex
Включил cli -s .system.logLevel debug

Сообщения пошли на ящик. Теперь осталоcm разобраться почему не корректно приходят на регистратор. Есть подозрение что где-то не подставляется поле  email_from_name или email_from_address

themactep, 2022-07-27:
> Кстати если выставлять Sensitivity то осле применения всегда ползунок показывает значение 3, хотя в скрипте в конфиг файле стоит то которое применил. То есть не понять какое значение sensivity стоит не понимая и расшифровывая конфиг файл

в свежей версии должно выставлятся нужное значение на полозке.

esuslik, 2022-07-27:
ползунок теперь работает кака надо.

Сообщения пошли на ящик. Теперь осталоcь разобраться почему не корректно приходят на регистратор. Есть подозрение что где-то не подставляется поле email_from_name или email_from_address

themactep, 2022-07-27:
я добавил логи в сервисы отправки. будет легче обнаружить ошибку.

esuslik, 2022-07-27:
# cat /tmp/webui/email.log
* response reading failed

Может где-то еще есть логи поподробнее?

themactep, 2022-07-27:
Это ошибка SSL/TLS сертификата. Принимающая сторона точно умеет SSL? Сертификат валидный?

esuslik, 2022-07-27:
не умеет, и в настройках плагина ssl отключен

themactep, 2022-07-27:
тяни новый апдейт, там запись в лог команды, которая отправляет почту. будем разбирать ее.

esuslik, 2022-07-28:
curl --verbose --silent --connect-timeout 100 --max-time 100 --user alert1@video.ru:1234321 --url smtp://app.3it.ru:1025 --mail-from CHE_6_001@video.ru --mail-rcpt alert2@video.ru -F '=(;type=multipart/mixed' -F "=Thu Jul 28 08:36:50 +05 2022;type=text/plain" -F "file=@/tmp/email_snap.jpg;type=image/jpeg;encoder=base64" -F '=)' -H "Subject: Snapshot from openipc-hi3516ev300" -H "From: \"CHE_6_001@video.ru\" <CHE_6_001@video.ru>" -H "To: \"alert2@video.ru\" <alert2@video.ru>"

* response reading failed

themactep, 2022-07-28:
Во From и To имя в ковычках, которые должны быть экранированы. Это форматирование гитхаба съело слэши или вывод в консоли такой?

themactep, 2022-07-28:
[image]

esuslik, 2022-07-28:
гитхаб съел. Кстати при включении плагина motionguard начинает падать маджестик.  Если перезапустить маджестик с консоли то не падает. Есть подозрение что дело в том что плагин начинает парсить логи и от этого что-то ломается.

[image]

themactep, 2022-07-28:
А если зайти телнетом на app.3it.ru, порт 1025 и сэмулировать отправку письма, ошибки будут?
```
EHLO app.3it.ru
MAIL FROM: CHE_6_001@video.ru
RCPT TO: alert2@video.ru
QUIT
```

esuslik, 2022-07-28:
не дает ничего сделать 
[image]

esuslik, 2022-07-28:
а нет, дает
[image]

themactep, 2022-07-28:
команды в телнете надо вводить руками построчно, получая после каждой ответ от сервера.

esuslik, 2022-07-28:
не успею тогда, через несколько секунд связь обрывает

themactep, 2022-07-29:
в шелле линукса:
`echo -e "EHLO app.3it.ru\nMAIL FROM: CHE_6_001@video.ru\nRCPT TO: alert2@video.ru\nQUIT" | nc app.3it.ru 1025`

esuslik, 2022-07-29:
[image]

themactep, 2022-07-29:
```
root@openipc-hi3518ev100:~# echo -e "EHLO app.3it.ru\nMAIL FROM: CHE_6_001@video.ru\nRCPT TO: alert2@video.ru\nQUIT" | nc mail.themactep.com 25
220 micro.themactep.com ESMTP Exim 4.96 Thu, 28 Jul 2022 23:49:23 -0400
250-micro.themactep.com Hello d192-186-103-98.static.comm.cgocable.net [192.186.103.98]
250-SIZE 52428800
250-8BITMIME
250-PIPELINING
250-PIPECONNECT
250-CHUNKING
250-STARTTLS
250-PRDR
250 HELP
250 OK
550 relay not permitted
221 micro.themactep.com closing connection
root@openipc-hi3518ev100:~#
```
Это пример того, как отвечает нормальный SMTP сервер.
Смотри логи на принимающей стороне.

esuslik, 2022-07-29:
там нет логов, там закрытый регистратор. Суть в том что с камеры с ориг прошивкой работает как положено и с камеры dahua с ориг прошивкой тоже работает.
Разработчикам рега я отписался, они тоже будут смотреть

esuslik, 2022-07-29:
Еще хотелось бы разобраться с падением маджестика, не знаю сюда это или нет. Вроде чюда т.к. падает при включении плагинов, а вроде и маджестик падает.
При включении плагина motionguard начинает падать маджестик. Если перезапустить маджестик с консоли то не падает. 
Причем когда падает маджести то он не отправляет корку на фтп и не пишет ее на саму камеру. И watchdog не всегда срабатывает.

themactep, 2022-07-29:
> там нет логов, там закрытый регистратор. Суть в том что с камеры с ориг прошивкой работает как положено и с камеры dahua с ориг прошивкой тоже работает. Разработчикам рега я отписался, они тоже будут смотреть

Это Yucca что ли? Ну, они вчера форкнули репо, вроде начали что-то пилить для совместимости. Может чего родят.

esuslik, 2022-07-29:
нет это другой

themactep, 2022-07-29:
Маджестик вроде бы @widgetii пофиксил сегодня. В новой сборке должен приползти.

esuslik, 2022-07-29:
обновился час назад, проблема с падением сохраняется

widgetii, 2022-07-29:
1. На каждую проблему нужно открывать отдельный тикет, не нужно все мешать в кучу
2. В проблеме должно описываться как ее воспроизвести (шаг за шагом)

esuslik, 2022-07-29:
так это понятно, я  об этом писал что проблема только при в ключении модуля в вебе который парсит логи. Поэтому пока не заводил отдельный тикет в реп маджестика. Ждал ответа по вебу может проблема тут и кроется, а не в маджестике. 
Тикет в маджестик сделаю, только ни логов ни корок не прикреплю, т.к. они не создаются.

closed 2023-06-25 by esuslik