OpenIPC/firmware #243: Переписать телеграм бота

p0i5k, 2022-06-20:
Текущая реализация телеграм бота, написанная на шелле, использует jsonfilter, непоставляемый в прошивке.
Переписать бота на языке C исполозованнием libjson-c

ZigFisher, 2022-06-28:
Под эту задачу, для поддержки и подогрева интереса, возможен небольшой финансовый донат или видеокамеры.

ZigFisher, 2022-07-06:
The current implementation of the telegram bot, written on the shell, uses jsonfilter, which is not supplied in the firmware.
Rewrite bot in C language using libjson-c

ZigFisher, 2022-07-06:
For this task, to support and warm up interest, a small financial donation or video cameras are possible.

themactep, 2022-07-07:
Helpful links for prospective developers:
https://core.telegram.org/bots
https://core.telegram.org/bots/samples

closed 2026-09-04 by openipc-ai