OpenIPC/firmware #247: ffmpeg не может сохранить rtsp в ts (pts?)

orlovsn, 2022-07-01:
```
ffmpeg -fflags +genpts -stimeout 5000000 -i rtsp://admin:111111@10.42.0.176 -vsync 0 -copyts -map 0 -c:v copy -an tmp.ts
ffmpeg version 4.2.4 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection' --extra-ldflags='-Wl,-z,relro -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld ' --extra-cflags=' ' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --disable-crystalhd --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-libaom --enable-libdav1d --enable-libass --enable-libbluray --enable-libcdio --enable-libdrm --enable-libjack --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libmp3lame --enable-nvenc --enable-openal --enable-opencl --enable-opengl --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librsvg --enable-libsrt --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libvorbis --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-version3 --enable-vapoursynth --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxvid --enable-libzimg --enable-libzvbi --enable-avfilter --enable-avresample --enable-libmodplug --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --shlibdir=/usr/lib64 --enable-libmfx --enable-runtime-cpudetect
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Input #0, rtsp, from 'rtsp://admin:111111@10.42.0.176':
  Metadata:
    title           : RTSP Session
  Duration: N/A, start: 0.066689, bitrate: N/A
    Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 2560x1920, 15 fps, 15 tbr, 90k tbn, 15 tbc
File 'tmp.ts' already exists. Overwrite ? [y/N] y
Output #0, mpegts, to 'tmp.ts':
  Metadata:
    title           : RTSP Session
    encoder         : Lavf58.29.100
    Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 2560x1920, q=2-31, 15 fps, 15 tbr, 90k tbn, 90k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
[mpegts @ 0x561027658040] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[mpegts @ 0x561027658040] first pts value must be set
av_interleaved_write_frame(): Invalid data found when processing input
frame=    1 fps=0.0 q=-1.0 Lsize=       0kB time=-00:00:00.06 bitrate=  -0.0kbits/s speed=N/A
video:218kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
Conversion failed!
```

в mp4 сохраняется нормально этой-же командой:

```
ffmpeg -fflags +genpts -stimeout 5000000 -i rtsp://admin:111111@10.42.0.176 -vsync 0 -copyts -map 0 -c:v copy -an tmp.ts
```

флаги  -fflags +genpts  ничего не меняют

п.с. из hls потока ffmpeg может сформировать hls без проблем:

```
ffmpeg -i "http://admin:111111@10.42.0.176/video.m3u8" -vsync 0 -copyts -map 0 -c:v copy -c:a aac -movflags frag_keyframe+empty_moov -f hls -hls_list_size 3 -hls_flags delete_segments -segment_list_flags +live -strftime 1 -hls_time 3  -hls_segment_filename "/mnt/raid/topol/ext/ramfs/49f83d34-05c3-48db-9aef-a956f06402fd/%y%m%d-%H%M%S.ts" "/mnt/raid/topol/ext/ramfs/49f83d34-05c3-48db-9aef-a956f06402fd/live.m3u8"
but, certainly, with double delay
```

orlovsn, 2022-07-02:
> Что пишет при проигрывании первого файла `ffplay tmp.ts` ?

tmp.ts нулевой длины  
если попробовать его проиграть то: 

```
ffplay -nodisp tmp.ts
ffplay version 4.2.4 Copyright (c) 2003-2020 the FFmpeg developers
  built with gcc 8 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection' --extra-ldflags='-Wl,-z,relro -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld ' --extra-cflags=' ' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --disable-crystalhd --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-libaom --enable-libdav1d --enable-libass --enable-libbluray --enable-libcdio --enable-libdrm --enable-libjack --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libmp3lame --enable-nvenc --enable-openal --enable-opencl --enable-opengl --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librsvg --enable-libsrt --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libvorbis --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-version3 --enable-vapoursynth --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxvid --enable-libzimg --enable-libzvbi --enable-avfilter --enable-avresample --enable-libmodplug --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --shlibdir=/usr/lib64 --enable-libmfx --enable-runtime-cpudetect
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
tmp.ts: Invalid data found when processing inputsq=    0B f=0/0
```

widgetii, 2022-07-02:
Issue only reproduces on T31, for EV300 everything is ok