OpenIPC/firmware #248: Cannot save Majestic setting changes (xm510)

schneemaier, 2022-07-02:
Hi, i just updated my XM510 based camera to the latest version. 

Majestic config changes cannot be saved from the WebUI, it always ends with "No changes found." message, regardless which setting i change on the UI. This is not a new issue, i had it with the previous versions too.

Thanks

schneemaier, 2022-07-06:
Did the upgrade:
```root@openipc-xm510:~# sysupgrade -k -r
OpenIPC System Updater

Vendor	xiongmai
SoC	xm510
Kernel	0x62be6f87
RootFS	master+954ae70, 2022-07-01

Stop services, sync and fresh memory
Killing majestic .. OK
Killing crond ...... OK
Killing klogd ....... OK
Killing ntpd ...... OK
Killing rngd ...... OK
Killing syslogd ....... OK

Firmware
Download from https://github.com/OpenIPC/firmware/releases/download/latest/openipc.xm510-br.tgz
Received and unpacked

Kernel
Update kernel from /tmp/uImage.xm510
SoC OK
New version, going to update
Erasing block: 22/22 (100%) 
Writing kb: 1401/1401 (100%) 
Verifying kb: 1401/1401 (100%) 
Kernel updated to 0x62c5068d

RootFS
Update rootfs from /tmp/rootfs.squashfs.xm510
SoC OK
New version, going to update
Erasing block: 40/40 (100%) 
Writing kb: 2528/2528 (100%) 
Verifying kb: 2528/2528 (100%) 
RootFS updated to master+954ae70, 2022-07-01

Unconditional reboot
```

After reboot:

```
root@openipc-xm510:~# firstboot
OpenIPC System Updater

Vendor	xiongmai
SoC	xm510
Kernel	0x62c5068d
RootFS	master+0b23ebd, 2022-07-06

Stop services, sync and fresh memory
Killing majestic .. OK
Killing crond ...... OK
Killing klogd ...... OK
Killing ntpd ...... OK
Killing rngd ....... OK
Killing syslogd ...... OK

OverlayFS
Erase overlay partition
Erasing 64 Kibyte @ b0000 - 100% complete.Cleanmarker written at a0000.

Unconditional reboot
```

Setup new password and tried top change some settings from majestic settings, but still receiving error on the UI. Looked into dmesg and every time i try to save a change i get the following line: `overlayfs: ERROR - failed to whiteout 'majestic.yaml'`

When i looked at /overlay/etc directory i did not find the majestic.yaml file there. Issuing `touch /overlay/etc/majestic.yaml` fixed the issue in a way that i don't see any error message on the UI. But the majestic.yaml file is not updated at /overlay/etc/majestic.yaml after saving the changes on the UI.

ZigFisher, 2022-07-09:
Good afternoon
You have some strange problem with your flash drive
Try to connect to the bootloader via UART and execute the command `sf probe 0; sf lock 0`

schneemaier, 2022-07-09:
It seems that my u/boot does not have the sf lock command:
```
U-Boot> sf probe 0
U-Boot> sf lock 0
sf - SPI flash sub-system

Usage:
sf probe [[bus:]cs] [hz] [mode] - init flash device on given SPI bus
                                  and chip select
sf read addr offset len - read `len' bytes starting at
                                  `offset' to memory at `addr'
sf write addr offset len        - write `len' bytes from memory
                                  at `addr' to flash at `offset'
sf erase offset [+]len          - erase `len' bytes from `offset'
                                  `+len' round up `len' to block size
sf update addr offset len       - erase and write `len' bytes from memory
                                  at `addr' to flash at `offset'
U-Boot>
```

I did some more tests and found that when I press save on the Majestic config page it actually whips out the config file from the overlay fs:

Before pressing save:
```
root@openipc-xm510:~# ls -l /overlay/etc/majestic.yaml 
-rw-r--r--    1 root     root           703 Jul  9 14:34 /overlay/etc/majestic.yaml
```
After save:
```
root@openipc-xm510-sch:~# ls -l /overlay/etc/majestic.yaml 
ls: /overlay/etc/majestic.yaml: No such file or directory
```
So the erase operation works, but for some reason it fails to write the config

dimerr, 2022-07-10:
are you using dev version of webui? this is a known problem with overlay dir structure on different kernels and i think it was fixed already...

schneemaier, 2022-07-10:
I am not sure which webUI is installed (the installed line is empty), I guess it is the stable version as i installed it with the whole firmware package.

Installed 
Stable [master+0ac2ff6, 2022-06-29](https://github.com/OpenIPC/microbe-web/commits/master)
Unstable [dev+e8113f3, 2022-07-09](https://github.com/OpenIPC/microbe-web/commits/dev)

I tried to switch to the development version, but it is failing via the UI:

```
Archive:  /tmp/microbe.zip
   creating: microbe-web-dev/
   creating: microbe-web-dev/dev/
   creating: microbe-web-dev/dev/css/
  inflating: microbe-web-dev/dev/css/main.scss
   creating: microbe-web-dev/files/
   creating: microbe-web-dev/files/usr/
   creating: microbe-web-dev/files/usr/sbin/
  inflating: microbe-web-dev/files/usr/sbin/send2telegram.sh
  inflating: microbe-web-dev/files/usr/sbin/send2yadisk.sh
  inflating: microbe-web-dev/files/usr/sbin/sendcoredump.sh
  inflating: microbe-web-dev/files/usr/sbin/updatewebui.sh
   creating: microbe-web-dev/files/var/
   creating: microbe-web-dev/files/var/www/
  inflating: microbe-web-dev/files/var/www/401.html
   creating: microbe-web-dev/files/var/www/a/
  inflating: microbe-web-dev/files/var/www/a/bootstrap.css.gz
  inflating: microbe-web-dev/files/var/www/a/bootstrap.js.gz
  inflating: microbe-web-dev/files/var/www/a/bootstrap.override.css
  inflating: microbe-web-dev/files/var/www/a/favicon.png
  inflating: microbe-web-dev/files/var/www/a/gear.svg.gz
  inflating: microbe-web-dev/files/var/www/a/github.svg.gz
  inflating: microbe-web-dev/files/var/www/a/light-off.svg.gz
  inflating: microbe-web-dev/files/var/www/a/light-on.svg.gz
  inflating: microbe-web-dev/files/var/www/a/logo-ipeye.webp
  inflating: microbe-web-dev/files/var/www/a/logo.svg.gz
  inflating: microbe-web-dev/files/var/www/a/main.js
  inflating: microbe-web-dev/files/var/www/a/telegram.svg.gz
  inflating: microbe-web-dev/files/var/www/a/tz.js.gz
   creating: microbe-web-dev/files/var/www/cgi-bin/
  inflating: microbe-web-dev/files/var/www/cgi-bin/_mj.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/about.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/admin.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/config.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/console.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/debugging.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/endpoints.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/firmware-reset.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/firmware-update.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/firmware-upload-parts.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/firmware.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/info-cron.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/info-dmesg.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/info-httpd.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/info-log.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/info-majestic.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/info-modules.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/info-overlay.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/jrun.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/majestic-config-actions.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/majestic-config-compare.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/majestic-settings.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/majestic.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/network-ntp.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/network-socks5.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/network.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/night.cgi
   creating: microbe-web-dev/files/var/www/cgi-bin/p/
  inflating: microbe-web-dev/files/var/www/cgi-bin/p/common.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/p/footer.cgi
  inflating: microbe-web-dev/files/var/www/cgi-bin/p/header.cgi
[... 288 lines cut ...]
rm: can't remove '/var/www/cgi-bin/majestic-config-reset.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/majestic-config-restore.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/majestic-debug.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/majestic-github.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/majestic-settings-update.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/network-ntp-reset.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/network-ntp-update.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/network-tz-update.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/network-update.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/ntp-update.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/preview-help.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/preview-mjpeg.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/preview-video.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/progress.cgi': Not supported
rm: can't remove '/var/www/cgi-bin/team.txt': Not supported
rm: can't remove '/var/www/cgi-bin/webui-settings-update.cgi': Not supported
rm: can't remove '/var/www/login.html': Not supported
Delete bundle
Delete temp directory
root@openipc-xm510:~# 
```

I was able to copy all files if i added the /overlay/ prefix to the files going into the usr and www the directories. It upgraded the UI, but did not change the behavior of the updates

I tried updating Mejestic parameters via the new UI, now I don't get an error. It simply fails silently

themactep, 2022-07-10:
No version of the installed web interface means you have a bundled one, that comes with the firmware.

themactep, 2022-07-10:
There is something wrong with your overlay partition that causes errors during copying/deleting files.

schneemaier, 2022-07-10:
Here is the filesystem inforamtion:
```
root@openipc-xm510:/# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.5M      2.5M         0 100% /rom
devtmpfs                  7.5M         0      7.5M   0% /dev
/dev/mtdblock4          704.0K    364.0K    340.0K  52% /overlay
overlayfs               704.0K    364.0K    340.0K  52% /
tmpfs                     7.5M         0      7.5M   0% /dev/shm
tmpfs                     7.5M     12.0K      7.5M   0% /tmp
tmpfs                     7.5M     44.0K      7.5M   1% /run
root@openipc-xm510:/# mount
rootfs on / type rootfs (rw)
/dev/root on /rom type squashfs (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=7664k,nr_inodes=1916,mode=755)
/dev/mtdblock4 on /overlay type jffs2 (rw,relatime)
overlayfs on / type overlayfs (rw,relatime,lowerdir=/,upperdir=/overlay)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
sysfs on /sys type sysfs (rw,relatime)
root@openipc-xm510:/# cat /etc/fstab
# <file system>	<mount pt>	<type>	<options>	<dump>	<pass>
/dev/root	/		ext2	rw,noauto	0	1
proc		/proc		proc	defaults	0	0
devpts		/dev/pts	devpts	defaults,gid=5,mode=620,ptmxmode=0666	0	0
tmpfs		/dev/shm	tmpfs	mode=0777	0	0
tmpfs		/tmp		tmpfs	mode=1777	0	0
tmpfs		/run		tmpfs	mode=0755,nosuid,nodev	0	0
sysfs		/sys		sysfs	defaults	0	0
```

If the problem is with the overlay partition on `/dev/mtdblock4` how can i fix it? Note that it was created by the firmware. 

Or the overlay fs you are referring to is the `overlayfs on / type overlayfs (rw,relatime,lowerdir=/,upperdir=/overlay)`? Maybe the problem is that the upper and lover directories need to be swapped? How can i try that?

schneemaier, 2022-07-11:
After several tries i was able to update the WebUI, now it shows the latest development version
```
Installed dev+e8113f3, 2022-07-09
Stable [master+0ac2ff6, 2022-06-29](https://github.com/OpenIPC/microbe-web/commits/master)
Unstable [dev+e8113f3, 2022-07-09](https://github.com/OpenIPC/microbe-web/commits/dev)
```

I can change, update every setting i tried, except for majestic. Those settings are not saved with the error message dmesg: `overlayfs: ERROR - failed to whiteout 'majestic.yaml'`

schneemaier, 2022-07-18:
I read a bit more about overlayfs and it seems that the config is correct. The upperdir should be writable. 

Can it be kernel related?

schneemaier, 2022-09-09:
After some testing it seems to be a overlayfs issue wiht white out failing. Looking at the kernel repo I found this patch related to whiteout issue in overlayfs: https://git.kernel.org/pub/scm/linux/kernel/git/mszeredi/vfs.git/commit/?id=3e01cee3b980f96463cb6f378ab05303a99903d9

schneemaier, 2023-01-17:
I created a workaround in the webUI package. Replaced cp with cat in several instances and now i can save the settings without errors. I also fixed the the same problem in the webUI updated.

https://github.com/OpenIPC/microbe-web/pull/116

themactep, 2023-01-18:
I believe we had that overlay thing fixed like a year ago. If it still affects some platform, I think it should be fixed there, too. We heen to fix the core of the issue without writing more workarounds. Otherwise it will surface again somewhere down the road in another script which will be using a legit `cp` instead of `cat`.

schneemaier, 2023-01-18:
Can you point me to the fix? 

I agree that this is a workaround, which may fail after some updates.

widgetii, 2023-01-18:
I believe it's better to fix root cause, for example back port changes for overlayfs from recent kernel version to XM's one (let's find proper patches to fix the issue in Linux kernel mailing lists)

dimerr, 2023-01-18:
[image]
I would start from syncing these with xm530 or any other working board that has similar kernel ver

widgetii, 2026-10-15:
I re-tested this on a lab XM510 (JXH62 sensor) and found the underlying cause.

**The symptom as reported no longer reproduces.** On current firmware a settings save from the WebUI (`POST /api/v1/config`) writes `/etc/majestic.yaml`, majestic applies it, and it survives a power cycle.

**The cause was the overlay, and it was still there.** On XM510 a file that comes from the read-only squashfs can't be deleted through the overlay, and so it can't be replaced with `cp -f` either:

```
# rm /etc/openipc_donors
rm: can't remove '/etc/openipc_donors': Not supported
# cp -f /tmp/issue.bak /etc/issue
cp: can't create '/etc/issue': File exists
```

XM510's 3.0 kernel uses the old overlayfs, which records a deleted lower file as a "whiteout" xattr in the writable jffs2 partition. The XM510 kernel config never enabled jffs2 xattrs (`CONFIG_JFFS2_FS_XATTR`), so the whiteout couldn't be written. Every other board on this overlayfs (hi3516cv100, hi3518ev100, xm530) has it.

The WebUI of 2022 saved settings with `cp -f /tmp/majestic.yaml /etc/majestic.yaml`, and the image then shipped `/etc/majestic.yaml` in the squashfs. That `cp -f` failed every time, the file never changed, and the WebUI reported nothing changed. That's this issue. Today's WebUI saves through majestic, which writes the file itself, so the settings page stopped hitting it. Anything that `rm`s or `cp -f`s a stock file on XM510 still did, though.

**Fix: #2606**, which enables jffs2 xattrs in the XM510 kernel config. With it, on the same board, all of these work, and they persist across a power cycle:
- `rm` of a stock file;
- `cp -f` over one;
- the exact 2022 save sequence.

schneemaier, 2026-11-02:
Yes, with the latest versions I don't see this issue anymore.

closed 2026-11-02 by schneemaier