OpenIPC/firmware #250: Bootloader flahing failed at 98%

KanG00, 2022-07-02:
I have done BOOT loader flashing as per instructions for my  HI3518EV200 and it failed at 98%.
Now it is bricked.
Could you please tell me what I did wrong and support in recovery.

Logs below:
`hisilicon # sf probe 0 ; sf lock 0
16384 KiB hi_fmc at 0:0 is now current device
unlock all block.
hisilicon # mw.b 0x82000000 ff 0x50000
hisilicon # tftp 0x82000000 openwrt-hi35xx-18ev200-u-boot.bin
Hisilicon ETH net controler
MAC:   00-00-23-34-45-66
eth0 : phy status change : LINK=UP : DUPLEX=FULL : SPEED=100M
TFTP from server 192.168.1.7; our IP address is 192.168.1.9
Download Filename 'openwrt-hi35xx-18ev200-u-boot.bin'.
Download to address: 0x82000000
Downloading: #################################################
done
Bytes transferred = 123532 (1e28c hex)
hisilicon # sf erase 0x0 0x50000
Erasing at 0x50000 -- 100% complete.
hisilicon # sf write ${0x820000000} 0x0 ${123532}
Usage: sf write addr offset len
hisilicon # sf write 0x820000000 0x0 123532
Writing at 0x120000 --  98% complete.
`

Now it stuck and there is no fedback on UART. Only blank messages

p0i5k, 2022-07-02:
Should have rounded 0x1e28c to 0x20000.
Do you have programmer? You can flash it and try again

KanG00, 2022-07-02:
I have Serial to UART converter connected to board only. What kind of programmer would I need to recover it ?

widgetii, 2022-07-03:
There is a good choice https://aliexpress.com/item/32902635911.html

KanG00, 2022-07-03:
I’ll check this promissing tool first for sure, but could you please advice the exact command that I should use for my chip?
Don’t wan’t to get it worse :)
Can I use directly OpenWRT U-boot, or shall I look for some special one?

widgetii, 2022-07-03:
`./hi35xx-tool --chip hi3518ev200 --file=u-boot-hi3518ev200-universal.bin`

Download [u-boot-hi3518ev200-universal.bin](https://github.com/OpenIPC/firmware/releases/download/latest/u-boot-hi3518ev200-universal.bin) file first

KanG00, 2022-07-03:
Any preconditions for tool usage ?

> MacBook-Air-Piotr:burn kang$ ./hi35xx-tool --chip hi3518ev200 --file=u-boot-hi3518ev200-universal.bin
Traceback (most recent call last):
  File "/Users/kang/.ssh/burn/./hi35xx-tool", line 5, in <module>
    import progressbar
ModuleNotFoundError: No module named 'progressbar'

KanG00, 2022-07-03:
ok, I've managed to install universal u-boot and stop the boot process.
How can I define IP address now to download and flash u-boot again?

> /burn $ screen /dev/ttyUSB0 115200
Hisilicon ETH net controler
MAC:   00-00-23-34-45-66
PHY not link.

p0i5k, 2022-07-03:
Your first steps was right. Except this command:
`sf write 0x820000000 0x0 123532`
Use this:
`sf write 0x820000000 0x0 0x20000`

KanG00, 2022-07-03:
I know but with current bootloader I do not have TFTP available as boot loader is not able to connect to Ethernet PHY adapter.
I can copy on SDCARD the files but I don't know the commands do access this directory.

> tftp 0x82000000 openwrt-hi35xx-18ev200-u-boot.bin
Hisilicon ETH net controler
MAC: 00-00-23-34-45-66
**PHY not link.**

KanG00, 2022-07-05:
I've managed to recover the chip finally by using openwrt-hi35xx-18ev200-u-boot.bin u-boot supporting SD card read as universal u-boot was not able to make use of ethernet port in my device and TFTP solution was not working.
After flashing from SD I had to change phyaddru and phyaddrd env's to 0 so that ethernet port started working.
Now I struggle with making Wifi adapter working and recognizing what sensor does my camera use, but this is the topic for different post.
Thank you for th support as "burn" recovery tool was the one that really helped!

dimerr, 2022-07-05:
Latest build has mmc support. Also, unlike old boot, new has ability to change various network settings, you just need to see what mode was on original bootloader, MII or RMII

ZigFisher, 2022-07-06:
> I used u-boot given by @widgetii couple posts above.
> Where is the latest u-boot and firmware to download? I’m having huge problems with self compilation

https://github.com/OpenIPC/firmware/releases/tag/latest

KanG00, 2022-07-06:
Sorry for this lame question, but how to flash my camera using this firmware compressed files?
I was able to program using SERIAL-UART converter but using root and kernel image files.
I've tried to update this using web gui System/Backup-Flash Firmware section, but after pressing update it gets stuck. 
SSH is not responding and Web GUI as well.

ZigFisher, 2022-07-09:
Good afternoon
You need to know which version you currently have installed.
Maybe you have the problematic version installed now.
You need to run the following command in Linux on the camera
`curl -s https://raw.githubusercontent.com/OpenIPC/firmware/master/general/overlay/usr/sbin/sysupgrade | sh -s -- -r -k -n`

KanG00, 2022-07-09:
I get wrong SoC error:

> Kernel
Update kernel from /tmp/uImage.hi3518ev200
sh: od: not found
Wrong SoC! Aborting.

but SoC is correct and defined as env.
below ipctool output:

> ./ipctool
---
board:
  vendor: OpenWrt
  version: 21.04.10.2
chip:
  vendor: HiSilicon
  model: 3518EV200
ethernet:
  mac: "00:00:23:34:45:66"
  u-mdio-phyaddr: 1
  phy-id: 0xffffffff
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 591894b4
      - name: env
        size: 0x10000
        sha1: ead232ae
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel
        size: 0x200000
        sha1: 02ee79e7
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: bf02aa1e
      - name: rootfs_data
        size: 0x8b0000
        path: /overlay,jffs2,rw
    size: 16M
    addr-mode: 3-byte
ram:
  total: 64M
  media: 32M
firmware:
  u-boot: "2010.06 (Jul 10 2021 - 19:37:22)"
  kernel: "3.4.35 (Sat Jul 10 19:35:56 UTC 2021)"
  toolchain: gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 latest-178-g157e03c47e)
  libc: uClibc 0.9.33.2
  sdk: "Hi3518EV200_MPP_V1.0.4.1 B030 Release (Jul  3 2017, 17:50:38)"
  main-app: ./ipctool

KanG00, 2022-07-10:
ok, I solved it but it was a tricky one.
Universal u-boot flashing required different steps order to make it working. Not sure why but it worked.
After that I have read/write access finally.
Now maybe last question or two before closing this thread.

How to add motor support as current status is "Motors not initialized."?
And second one.
Hot to add wifi support as I have only wired interface available from the command line and my camera does have wifi module?

dimerr, 2022-07-10:
oh my... where did you get all that from? starting from old bootloader, now openwrt... jeez.

https://github.com/OpenIPC/firmware/releases/tag/latest

dont use openwrt or anything connected with it

KanG00, 2022-07-10:
Where? From OpenIPC webpage (https://openipc.org) :)
https://github.com/OpenIPC/chaos_calmer

Thanks for the link, it is all in there. Universal u-boot is 1kb different one which I've uploaded, but I'm afraid to touch it again if it's finally workin:)

Now I need to figure out how to configure rotation motor and wifi and my camera is good to go...

widgetii, 2022-07-10:
chaos calmer is deprecated, it's from 2015 (there is a warning in the repo), don't use it anymore

KanG00, 2022-07-10:
So maybe it is good to remove it from https://openipc.org/.
I've used it as I'm a bit familiar with OpenWRT so that sounded like a good choice for me.

dimerr, 2022-07-10:
> Where? From OpenIPC webpage (https://openipc.org) :) https://github.com/OpenIPC/chaos_calmer
> 
> Thanks for the link, it is all in there. Universal u-boot is 1kb different one which I've uploaded, but I'm afraid to touch it again if it's finally working :)
> 
> Now I need to figure out how to configure rotation motor and wifi and my camera is good to go...

i recommend flashing universal bootloader first of all, then install openipc-2.2 and continue with your experiments

KanG00, 2022-07-10:
That is the status as per the moment.
I have universal bootloader and opeipc-2.2 installed.
now I'm trying to figure out how to enable wifi and motor.

ZigFisher, 2022-07-10:
Run the `firstboot` command, it will prepare the partition for saving the changes.

ZigFisher, 2022-07-10:
Thanks for the feedback
Now we will remove the information about the old development branch under openwrt. 
Indeed, it has not been supported for more than two years and is misleading.

KanG00, 2022-07-10:
> Run the `firstboot` command, it will prepare the partition for saving the changes.

that causes reboot, and yes, soc is defined in variable:

> root@openipc-hi3518ev200:~# firstboot 
> Cannot parse config file '/etc/fw_env.config': No such file or directory
> SoC is not defined in U-Boot environment Aborting.
> 
> Unconditional reboot
>

KanG00, 2022-07-10:
this looks like an issue (boot log):

> jffs2: Incompatible feature node (0xe008) found at offset 0x006f2488
> mount: mounting /dev/mtdblock4 on /overlay failed: I/O error
> mount: mounting ubi0:rootfs_data on /overlay failed: No such device

KanG00, 2022-07-10:
ok, found it.
Now SOC variable needs to be in capital leters. with soc it didn't work. After changing soc to SOC /etc/fw_env.config showed up and I was able to make firstboot command :)

KanG00, 2022-07-10:
I've tried to add wifi support based on this instructions (https://github.com/OpenIPC/wiki/blob/master/en/wifi-xm530.md), but first of all I'm not sure if it will suite to my WIFI chip and what is more, I don't have required drivers inside, so it causes this output during boot proces:

> insmod: can't insert '/lib/modules/3.10.103+/xiongmai/dwc_common_port_lib.ko': No such file or directory
> insmod: can't insert '/lib/modules/3.10.103+/xiongmai/dwc_otg.ko': No such file or directory
> insmod: can't insert '/lib/modules/3.10.103+/xiongmai/wifi_pdn.ko': No such file or directory
> insmod: can't insert '/lib/modules/3.10.103+/xiongmai/compat.ko': No such file or directory
> insmod: can't insert '/lib/modules/3.10.103+/xiongmai/xm711.ko': No such file or directory
> FAIL

Could someone advice how to enable wifi in my device ?

dimerr, 2022-07-10:
> ok, found it. Now SOC variable needs to be in capital leters. with soc it didn't work. After changing soc to SOC /etc/fw_env.config showed up and I was able to make firstboot command :)

what capital letters? now where have you found that? after flashing our bootloader you dont need to set anything at all

KanG00, 2022-07-10:
Please look what I've written three posts above.
After running firstboot command I got an error "SoC is not defined in U-Boot environment Aborting."
Than I rebooted, stopped u-boot process and added SOC=hi3518ev200 as previously it was soc=hi3518ev200 and it apparently didn't work. Maybe if I remove both env entries it will also work.

dimerr, 2022-07-11:
you just did saveenv, thats why "soc' started working, along with whole env.

after installing uboot you need to pick partition layout with
```run setnor8m```
which will invoke saveenv also

closed 2022-08-24 by dimerr