OpenIPC/firmware #267: Add mavlink-router package to fpv profile

widgetii, 2022-07-18:
Please consider adding [mavlink router](https://github.com/mavlink-router/mavlink-router) to fpv profile.

This software uses Meson+Ninja build system (can be installed on build machine by `sudo apt install git meson ninja-build` 

Then use [build instructions](https://github.com/mavlink-router/mavlink-router#build), adjusted for cross-compilation (as example of EV300 system):

```
meson setup -Dsystemdsystemunitdir=/usr/lib/systemd/system --buildtype=release --cross-file meson-cross-arm-openipc-linux-musleabi.ini  build .
meson compile -C build
```

Target binary for IPC will be located at `./build/src/mavlink-routerd`. Make sure that `libstdc++.so.6` dependency is also copied to the target.
Consumed size data:
```
280656   mavlink-routerd
1490244  libstdc++.so.6
```
[meson-cross-arm-openipc-linux-musleabi.ini.zip](https://github.com/OpenIPC/firmware/files/9130833/meson-cross-arm-openipc-linux-musleabi.ini.zip)

widgetii, 2022-07-18:
Previous recipe required `libstdc++` as dependency and will take of 1729Kb of space.  If there is no other application in the system which require `libstdc++` one can build `mavlink-routerd` with static-linked `libstc++`

```patch
diff --git a/src/meson.build b/src/meson.build
index 086ce9a..f9cb411 100644
--- a/src/meson.build
+++ b/src/meson.build
@@ -23,6 +23,7 @@ exe_mavlink_router = executable('mavlink-routerd',
         ],
         link_with: libcommon_private,
         install: true,
+        link_args : '-static-libstdc++'
 )
 
 if dep_gtest.found()
```

Final `mavlink-routerd` size is 1006Kb

cronyx, 2022-07-22:
introduced in #269

closed 2022-07-28 by widgetii