OpenIPC/firmware #271: Should apply for ntppool "vendor" status e.g. openipc.pool.ntp.org

linuxtim, 2022-07-22:
Minor thing, but I notice that OpenIPC defaults to using `0.pool.ntp.org`, `1.pool.ntp.org` etc. as its default ntp server list.  The NTP Pool volunteers say don't do this, but instead request that projects apply for a "vendor zone".  As an open source project, this should be free for OpenIPC, e.g. Debian uses by default `0.debian.pool.ntp.org` etc. More details in the link below...

 "Open Source projects are of course particularly welcome to use the pool in their default setup, but we ask that you [get a vendor zone](https://www.ntppool.org/en/vendors.html#vendor-zone) when using the pool as a default configuration."

themactep, 2022-07-22:
Good point. I think we need to set up our own NTP server and offer it as a default server for synchronization leaving other options blank for end user consideration.

Maybe we could even make our own pool and donate its bandwidth back to the NTP Pool. Setting two-three entry level VPS in different datacenters for better stability/accessibility won't cost much.

linuxtim, 2022-07-24:
If you have the resources, then adding a server to pool.ntp.org would be good, but for the default NTP settings, I think it would be better to apply for openipc.pool.ntp.org, and change the default for new builds to use those instead.

I think having the 4 separate servers (like the current default config) and letting the ntp daemon operate is multi-server mode is the optimal config, and will give a much more reliable service than just filling a single server.  Routing problems or network congestion are much more likely to cause problems with a single-server configuration I think.

Apart from that, I'm sure ntp.org have internal performance monitoring set up, and because they have so many NTP servers worldwide (and I think they use location-based dns), their infrastructure will be the best for giving a good ntp service to OpenIPC users everywhere.

linuxtim, 2022-07-24:
If you'd like me to do the coordination with the pool.ntp.org organisers, I'll be happy to...

themactep, 2022-07-24:
Thanks for offering, Tim. We have already set up 4 NTP servers in different datacenters. We're testing them internally, now. They will be supplied as default servers for OpenIPC firmware quite soon.

ystinia, 2022-07-30:
0.time.openipc.org
1.time.openipc.org
2.time.openipc.org
3.time.openipc.org

Servers are activated and registered by default in the firmware config.

closed 2023-11-30 by viktorxda